sinara-hw/sinara #330: Allaki rev1 tests

gkasprow, 2017-10-06:
I started testing Allaki board.
So far no smoke detected.
ADC chain works as expected, at least on the scope with generator attached. Gain control also works.

There is a lot of power dissipation on series resistor in RF amplifier. It could be easily supplied from 6V instead of 11.5V - the 100R resistor dissipates 7.1V. I know the datasheet suggests such value but I don't see much sense in supplying it from 11.5V

hartytp, 2017-10-06:
> There is a lot of power dissipation on series resistor in RF amplifier. It could be easily supplied from 6V instead of 11.5V - the 100R resistor dissipates 7.1V. I know the datasheet suggests such value but I don't see much sense in supplying it from 11.5V

What does that do to the linearity of the amplifier?

One option might be to bias it from a current source off the 6V rail.

gkasprow, 2017-10-06:
I'd add larger inductor in series with small one to have some gain at lower frequencies.
This resistor simply sets bias and may be needed at low frequencies, in order of a few MHz.

hartytp, 2017-10-06:
> This resistor simply sets bias and may be needed at low frequencies, order of a few MHz.

Yes, but if the voltage drop across the resistor is low, it's not a very stiff current source, so I'd expect worse linearity/temperature dependence of gain.

gkasprow, 2017-10-06:
@hartytp that's true. We can add current source instead :)

hartytp, 2017-10-06:
@gkasprow For the time being, my preference would be to just do basic functionality testing now. We can do more detailed testing once Allaki has shipped to the labs.

In the long run, there is a very long list of things I can imagine wanting to characterise, including:
- Output chain phase noise
- Output chain S parameters v frequency
- Switch/attenuator isolation
- Cross-talk
- etc etc etc

gkasprow, 2017-10-06:
@hartytp OK, so I will functionally test all remaining Allaki boards and once Saymas arrive, will ship them to you. So far I tested it outside Sayma since I don't have a code yet to play with attenuator.

jordens, 2017-10-06:
I agree with @hartytp . Basic functionality is all we should verify now. Later we can do the athletics

* [ ] IIP3 vs power and frequency (2D), on input and output
* [ ] NF on input vs frequency
* [ ] in-out, channel-channel, mezzanine-mezzanine crosstalk
* [ ] power supply rejection
* [ ] AM noise

dhslichter, 2017-10-06:
@gkasprow for future iterations we could run from a lower rail with a different resistor.  The resistor serves only to set the bias current (as pointed out by @hartytp ), but if you look at the values on the datasheet [page 3](https://www.minicircuits.com/pdfs/ERA-4SM+.pdf) you start talking about some pretty small resistors to do the 6V bias.  The amplifier voltage is affected by the current through this resistor, so fluctuations in the current enter directly into fluctuations in the output signal.  On the datasheet it specifies the sensitivity to current at the operating voltage, which gives an effective resistance of 10 ohms at the correct bias point.  Thus the larger the bias resistor is, the smaller the effect of a voltage rail fluctuation is on the current fluctuation through the amp, and therefore the better your PSRR.  For a 6V rail, you would need a 21.5 Ohm resistor, and so you are basically 5x more sensitive to noise on the power supply rail than if you use a ~12V rail with 100 ohms.  

If you were to make a stiff 65 mA current source instead, that would probably work, but keep in mind that this current source still needs to look like a current source, and not be compromised by parasitics, out at higher frequencies (at least until the RF choke impedance becomes large).

dhslichter, 2017-10-06:
I would add to the list for future tests:

* [ ] switch settling time
* [ ] attenuator settling time

gkasprow, 2017-10-06:
@dhslichter Thanks for explanation. This power is anyway negligible if you compare with Sayma RTM power. So we can leave it as it is.

dhslichter, 2017-10-06:
Dropping from a 12 V rail to a 6V rail would reduce power consumption by ~400 mW per amplifier, or ~3.2 W for the entire RTM card.  How does that stack up with the rest of the power budget?

gkasprow, 2017-10-06:
At the moment Sayma set consume roughly 24W with one DAC working. No ADC, clock module and RTM modules.

gkasprow, 2017-10-09:
I plugged Allaki to Sayma board and it works. At least I some sinewave at the SMA output.
Register access also works, I can control RF switch.

hartytp, 2017-10-09:
RF out of Sayma: that's a BIG milestone! Congrats.

gkasprow, 2017-10-09:
Now I asked my student to write Python script to control attenuator. So hopefully today would be able to test it. I'd like to measure attenuator response.

gkasprow, 2017-10-09:
Here is screenshot during attenuation transition from 0x2 to 0xff
[image]

gkasprow, 2017-10-09:
And the signal during RF switch transition:
[image]

gkasprow, 2017-10-09:
@jbqubit @jordens @hartytp anything else to quickly measure on Sayma + Allaki setup?
Sayma RTM is being assembled right now.

dhslichter, 2017-10-09:
I think you've done all the necessary quick measurements -- the other precision measurement stuff is not urgent and can be done later.  Also since Allaki is a removable daughtercard, any detailed issues with it can be fixed later without holding up Sayma.  I'd proceed to the other, more pressing items on the list.

gkasprow, 2017-10-09:
@dhslichter I'm thinking about making some adapter that would let us examine Allaki. So far there is no access to RF part, so one cannot even make measurements on parts of the signal chain.
I have SMP to SMA cables, so would be nice to extend the control/supply connectors from Sayma to Allaki by let's say 10cm. What do you think about it? I don't know what access do you need to do full diagnostics of the board.

dhslichter, 2017-10-09:
A hacked-together standalone test setup for Allaki, as you mention, should be capable of performing all of the measurements mentioned above by @hartytp and @jordens, aside from the mezzanine-mezzanine crosstalk (really, all of the crosstalk, as that will be necessarily tied up with physical configuration as plugged in to the RTM card).  

However, I would say let's not worry about putting together such a setup right now -- instead, focusing on getting Kasli and Urukul designs complete and starting the manufacturing seems to be the most pressing priority.

gkasprow, 2017-10-09:
@enjoy-digital I have short Python script that controls RF switch and attenuator. do you think this might be useful for community?

jordens, 2017-10-09:
@gkasprow How do they test the mezzanine at the board house? I'd guess that's where such a characterization board would be required. For someone developing and debugging a new mezzanine, I'd guess a bunch of SMP-SMP (or SMA-SMP) jumpers and a jumper for the control signals (two PCBs, a ribbon cable and two of those hermaphroditic connectors) would be nice. But those who want to develop a mezzanine would need to speak up.

jordens, 2017-10-09:
@gkasprow I agree with the others. If the switch switches, the attenuator attenuates, and the rf amplifier amplifies, we can make more of those, send them out, and characterize later.

gkasprow, 2017-10-09:
@jordens The boards are tested to make sure they won't harm Saymas. So power consumption is verified and ADC outputs. Rest of tests is performed with Saymas.

closed 2017-10-09 by gkasprow

dhslichter, 2017-10-10:
@gkasprow @hartytp @jordens  regarding the RF amplifier bias, I have heard back from Mini-Circuits with the following:

> Using a current source instead of a bias resistor to set the current is a great solution to the problem as this will minimize current fluctuations caused by the lower rail voltage. The large impedance and potentially reactive impedance will not cause issues in your design and will not bring up stability issues in your design. Overall I think it is a good ideal for you if you get to save some power consumption in your system and get to drop your rail voltage with no significant downside.  

So it seems that we should consider using a current source run from the 6 V rail to bias the ERA-4SM+ in the next iteration of Allaki.  We will have to see if there is enough real estate on the board for this!  

The Mini-Circuits folks also directed me to an application note on biasing of the ERA series amplifiers, which contains a derivation of their sensitivity to bias rails and confirms that when using just a bias resistor, running at lower rail voltages leads to greater sensitivity to noise/fluctuations on the rails, as well as temperature fluctuations: https://www.minicircuits.com/pages/pdfs/an60010.pdf