sinara-hw/sinara #331: Kasli: put a 125 MHz clock to MGTREFCLK1

jordens, 2017-10-11:
For Ethernet (125 MHz) and DRTIO (100, 125, 150 MHz) on the same quad we need more clocking flexibility. Sorry for noticing late.

You could e.g. take the KC705 setup or anything else that you have experience with.
[image].

jordens, 2017-10-17:
@marmeladapk @gkasprow ping. How is the routing going? If we need to simplify this by e.g. reducing the EEM connector number, we should discuss this it now.

marmeladapk, 2017-10-17:
@jordens Space is not really a problem and removing anything from your list is not necessary unless you feel something is redundant and won't ever be used. I'm fine with leaving everything as is and adding 125 MHz clock source (you want it on MGTREFCLK1, right?).

jordens, 2017-10-18:
@marmeladapk yes. On the currently unused mgtrefclk input. The list of items to remove is just there to speed up and simplify the layout job. If you say removing them does not help with that, then feel free to keep them ;) you are the expert.

marmeladapk, 2017-10-21:
I added [510FBA125M000AAG](https://www.silabs.com/documents/public/data-sheets/si510-11.pdf) Are you fine with it?

jordens, 2017-10-21:
Sure. That's fine. I don't know whether it makes sense to harmonize that one oscillator with the one from the RTM-SATA adapter.

jordens, 2017-10-21:
Oops. Wrong button. Feel free to close this when you think this is resolved.

hartytp, 2017-10-25:
@marmeladapk what's the status of Kasli? When will it be ready for manufacture?

closed 2017-10-26 by marmeladapk

marmeladapk, 2017-10-26:
Should be this week.

marmeladapk, 2017-11-02:
@hartytp After consultation with Greg it'll be ready on Monday.

hartytp, 2017-11-02:
Great!

gkasprow, 2017-11-02:
At the moment there are just cosmetic things to fix - polygons, additional vias, copper balancing.
PI simulations look good.

hartytp, 2017-11-02:
How did the thermal sim look?

gkasprow, 2017-11-02:
Not done yet.

commit 1e9e545f (2017-10-26): PCB routing
Fixed #331
Closed #296