sinara-hw/sinara #332: Urukul clock buffer output standard

hartytp, 2017-10-11:
@gkasprow Why is the Si53312-B-GM configured to use LVPECL outputs, rather than LVDS outputs? AFACIT, using LVDS would give the same phase noise and frequency performance, but would save some power and allow us to remove the biasing resistors.

Sorry for the late spot. This can be left for the second revision to avoid delays.

gkasprow, 2017-10-11:
I've chosen LVPECL because it has better noise margin than LVDS. This can be easily tested by pulling SFOUT lines low and removing output resistors. Actually, I can add 0R resistor to make it easier.

hartytp, 2017-10-11:
> I've chosen LVPECL because it has better noise margin than LVDS.

You mean lower output phase noise? That wasn't clear to me from the data sheet, but maybe I missed a figure?

hartytp, 2017-10-11:
Aaah. Okay. I wouldn't have expected that to be an issue if the clock traces are kept short and the grounding is good. But, happy to take your advice.

Unless you or @jordens object, can we try LVDS as our default option (so, make the 200R resistors DNP) and only switch to LVPECL if we find any issues during testing?

gkasprow, 2017-10-11:
OK, let's do that. The board is hot enough.

hartytp, 2017-10-11:
Thanks! Power budget will need updating once this decision is finalised.

jordens, 2017-10-11:
Yes. Thanks for spotting. LVDS sounds good to me.

jordens, 2017-10-17:
* [ ] As the bubble says, you want to DNP R42... and R34... since the outputs are LVDS and not LVPECL.

closed 2017-10-17 by gkasprow