sinara-hw/sinara #333: Urukul amplifier choke

hartytp, 2017-10-11:
Sorry for another really late question, but what minimum frequency are we aiming for @jordens ?

One generally wants the choke to have an impedance of something like 500Ohms at the lowest frequency of interest. Current choke is 0.5uH, so it won't be great below around 50MHz. Is that okay for you?

hartytp, 2017-10-11:
NB we're using the lowest biasing resistor/voltage supply recommended by the data sheet, which is probably not great from a linearity/stability point of view. Let's keep this for the current prototype round. But, may be worth considering replacing this in a later revision (@gkasprow any ideas for nice compact and reasonably stable/low-noise current sources?).

gkasprow, 2017-10-11:
@hartytp It seem to work even at 30MHz. I'd use current mirror on dual PNP supplied from low noise LDO.

hartytp, 2017-10-11:
> @hartytp It seem to work even at 30MHz.

It will work, the gain will just be a bit off, as the amplifier sees a load of 50R||100R, rather than 50R.

> I'd use current mirror on dual PNP supplied from low noise LDO.

Okay, sounds good. There are so many ways of doing this, but I don't have a good feeling for the performance implications of them.

Worth considering this for a future revision, as:
- We could eliminate the 7V5 LOD/power rail (run the current source from the 5V rail)
- It would reduce the power consumption a bit
- Should give improved amplifier stability

gkasprow, 2017-10-11:
Current mirror is bad idea - we don't have discrete transistors with current ratio different than 1. So it would double the current consumption. So probably opamp + PNP is best idea in terms of stability. I can add place holder on PCB to test this idea.

hartytp, 2017-10-11:
> So probably opamp + PNP is best idea in terms of stability. I can add place holder on PCB to test this idea.

Yes, but that probably takes up too much PCB area and could need different supplies.

I'm a bit rusty about this, but IIRC one can build multiple-output current mirrors? That way the bias for the current mirror isn't so important as only one is needed for the 4-channel PCB.

Anyway, I'll leave this as something to brood on for a future design revision...

dhslichter, 2017-10-11:
If we can find a clean way of implementing this current source with a lower voltage rail, that's great, but if we can tolerate the increased dissipation (400 mW per channel) of using a bias resistor from 11.5V rail, it is much simpler.  

@hartytp the effect of the choke is basically that one gets a graceful gain degradation at lower and lower frequencies, which seems not unreasonable in case one wants lower-frequency outputs than ~50 MHz.  If there is high-impedance current source instead of a resistor, then this is no longer an issue.

hartytp, 2017-10-11:
> If we can find a clean way of implementing this current source with a lower voltage rail, that's great, but if we can tolerate the increased dissipation (400 mW per channel) of using a bias resistor from 11.5V rail, it is much simpler.

I'd rather avoid this: this board will already get hot enough without adding 4*400mW extra dissipation for the larger bias voltage.

> @hartytp the effect of the choke is basically that one gets a graceful gain degradation at lower and lower frequencies, which seems not unreasonable in case one wants lower-frequency outputs than [...]

All true, but if there is a larger inductor in the same package that we can easily use, it's worth considering it.

dhslichter, 2017-10-11:
Also @hartytp the choke has a current-dependent inductance.  If you look at the datasheet [here](https://www.minicircuits.com/pdfs/TCCH-80+.pdf) you can see we have an inductance probably closer to 1.5-2 uH at 35 mA bias current.  Their measured performance shows that you get about 0.5 dB of loss due to the finite impedance (not accounting for the additional 100 ohm resistor, assuming you go straight to the supply) at 30 MHz, similar to higher frequencies.  

> I'd rather avoid this: this board will already get hot enough without adding 4*400mW extra dissipation for the larger bias voltage.

This was my mistake because of confusing Allaki and Urukul bias parameters and amp chips.  With the current design on Urukul it is only 120 mW per channel = 480 mW for the whole board.  

For a current source, how about something simple like an [LT3092](http://www.linear.com/product/LT3092)?  The impedance drops off substantially at a few MHz, so we still have the same issue with low-frequency gains, and one would need to put some series resistance in as well to prevent it from oscillating, but you could get a small footprint and the dropout is suitable for running from the 5V rail.

gkasprow, 2017-10-11:
I sketched some idea. Reference would be common for 4 channels. This would need only 0.4V..0.5V of dropout
[image]

hartytp, 2017-10-11:
> For a current source, how about something simple like an LT3092? The impedance drops off substantially at a few MHz, so we still have the same issue with low-frequency gains, and one would need to [...]

That's a nice IC. Lots of ways of doing this, just about finding the cheapest/most compact/most power efficient.

> Also @hartytp the choke has a current-dependent inductance. If you look at the datasheet here you can see we have an inductance probably closer to 1.5-2 uH at 35 mA bias current. Their measured [...]

Okay, not worth changing then. Closing this issue.

closed 2017-10-11 by hartytp

hartytp, 2017-10-11:
> I sketched some idea. Reference would be common for 4 channels. This would need only 0.4V..0.5V of dropout

- If we really want to save power, the reference can be powered from 1V8 (and obviously needs heavy bypassing/filtering)
- The OpAmp and get could be powered from +5V to save power.

Not sure what's more compact/cheaper: the quad OpAmp and FETs or @dhslichter's 4x current source IC. Do we have room on the PCB for these current source solutions (bearing in mind that we can scrap a voltage regulator if we do go for this...).

Anyway, this isn't something that I'd want to hold up this prototyping round for...

jordens, 2017-10-12:
* the power loss due to the choke at low f is ok for us.
* I'd like to go with the present resistive current source for now.

jordens, 2017-10-12:
@gkasprow please no changes to the biasing on the current PCB. That would likely lead to a few rounds of discussion, reviews, fixes, changes, and thus delay. Any delay hurts us.

hartytp, 2017-10-12:
> @gkasprow please no changes to the biasing on the current PCB. That would likely lead to a few rounds of discussion, reviews, fixes, changes, and thus delay. Any delay hurts us.

ACK. I don't think anyone is suggesting making this change for the first prototype round (which is basically frozen).

----

Apologies for getting a bit off topic, but:
- Do we really need the P3V3 LDO? AFAICS this only powers the LVDS interface, so couldn't we just run that directly from the 3V6 rail instead? That would save come components and free up some space.
- On the power efficiency front, the only other place in the power budget that I can see where we could save something is the LVDS interface. Is it worth considering running it from 2V5 instead? That would save a bit. But, maybe not worth the hassle. (We might also want diodes to protect from a user accidentally connecting it to 3V3 LVCMOS).

Again, these aren't requests for the current revision, but more comments for future design revisions.

jordens, 2017-10-12:
Yes. Revisiting the power supplies for the next revision is a good idea. We can gather some experience and hack wires on the prototypes. Particularly the current sources, and noise/stability/power/temperature questions that correlate.

We are mixing 2.5 V LVDS (Kasli), and 3.3 V LVDS already. AFAICT that's fine from a compatibility perspective.

hartytp, 2017-10-12:
@jordens Okay. Sounds good.

Moved to a new issue.