sinara-hw/sinara #334: Urukul supplies etc

hartytp, 2017-10-12:
Thoughts on Urukul power supplies for v1.1

**Amplifier bias**
- Bias amplifiers from a low-cost/small current source
- Run current source from P5V0 rail
- Remove P7V0 rail
- P7V5 rail becomes P5V5

**LVDS interface**
- Doesn't need an LDO
- Supply from 2V5 (1V8?)

**Clock buffers**
- If we don't need LVPECL, these can run from P1V8A. Otherwise, they can run from a 2V5 rail (this should probably have an LDO or heavy passive filtering though!).

dhslichter, 2017-10-12:
For current supplies to bias the amplifier -- if we go with an LT3092, we should be careful to field-test the solution on a dummy board, as there will need to be some damping resistors in series with the input and output to prevent oscillation due to the impedances of the filter capacitor and rf choke.  See datasheet for details.

hartytp, 2017-11-20:
@gkasprow Thinking about this a bit more, as the power budget for a 16-channel DDS+ADC+Kasli system looks pretty tight.

> Supply [LVDS interface] from 2V5

Based on our experience with Novo v2.0, it seems that this isn't really possible. So, shall we scrap that idea?

AFAICT, these ICs don't need an LDO and are happy running at 3V6. So, would it make sense to scrap the  regulator IC11 and run these ICs directly from the 3V6 SMPS? That won't save any power, but would save some BOM items.

jordens, 2017-11-20:
Do the 200 mW per board make a difference?

hartytp, 2017-11-20:
> Do the 200 mW per board make a difference?

It's a bit more than that, right?

Currently we are supplying 35mA of bias current per amplifier from a 7V5 SMPS (via a 7V LDO). Plan is to switch to running from a 5V5 SMPS (again, via an LDO). So, 2V * 35mA * 4=280mW. On top of that by suplying the 5V LDO from a 5V5 SMPS, rather than a 7V5 SMPS, we save another 25.6mA*2V. So, the total power saving is 330mW. That's not a huge amount, admittedly, but if we can chip away in a few places we can prob save about 1W which is worth it IMO.

On top of that, we should expect somewhat better performance from the amplifiers in terms of linearity stability and noise, since we are currently not biasing them from a very stiff current source (only 100R).

Plus, we get to remove a power rail, which is always nice.

Admittedly, these gains aren't be huge, so if you want to veto the current source on the grounds that it's  not worth the time then I can live with it. But, IMO, it's enough to be worth thinking about seriously.

jordens, 2017-11-20:
330mW is ~3% (maybe 5% as the current estimate is likely high). I am all for testing this (and a proper current source is appealing) but we should look at the numbers first before changing the design. Actual cost delta from board house (I am not certain that some "at quantity" numbers are applicable, see SMP bullets and BNC connectors) and before/after stability/noise/linearity.

hartytp, 2017-11-20:
> 330mW is ~3% (maybe 5% as the current estimate is likely high). I am all for testing this (and a proper current source is appealing) but we should look at the numbers first before changing the [...]

All agreed.

FWIW: we'll never get a factor of two reduction in the power consumption, since we're dominated by the DDS. But with a few of these small reductions, we can aim for 15% or so, which does seem worth it IMO. Note that it's not just the relative power dissipation that matters, but more absolute questions as well like: can we get a 16-channel DAC/ADC system to run under 5A. Currently, that looks marginal at best (depending on how pessimistic we are with our budgeting). Sometimes 10% makes the difference between being able to do things or not, and that can be significant for users!

hartytp, 2017-11-20:
Si53312 calculation:

Core power consumption: 100mA (maximum, 65mA typical)
VLDS output buffer: 4*20mA
LVPECL output buffers: 4*35mA

LVPECL 3V3: (4 * 35mA+100mA) * 3.3V=240mA * 3.3V = 792mW
LVDS 1V8: (4 * 20mA + 100mA) * 1V8 = 324mW

gkasprow, 2017-11-20:
@hartytp we switched to the LVDS mode some time ago. We also disable not used channels.
At the moment the power consumption is 470mA@12V.

hartytp, 2017-11-20:
> @hartytp we switched to the LVDS mode some time ago.

Okay, I'd forgotten that! Thanks for the reminder. So, we are using 3V3 LVDS, right? Is there any advantage of that over 1V8 LVDS?

hartytp, 2017-11-20:
Switching to 1V8 LVDS saves us 360mW.

hartytp, 2017-11-20:
> At the moment the power consumption is 470mA@12V.

Under what conditions? With all DDS at 1GSPS?

gkasprow, 2017-11-20:
@hartytp At the moment DDS channels are not configured, so this is idle power.

hartytp, 2017-11-20:
@gkasprow is there a mistake in the power budget for the Si53304 line? Should be 100mA (max) + 20mA*number of outputs = 200mA (*not* 100mA as currently listed).

hartytp, 2017-11-20:
> @hartytp we switched to the LVDS mode some time ago

In that case, the power budget is wrong, since LVDS uses 20mA per output.

hartytp, 2017-11-20:
Okay, those are all the areas I can see where we can save power in this design.

gkasprow, 2017-11-20:
@hartytp Correct. There is an error in case of Si53304.
I can try to make modification to see how much we save. Just unsolder the ferrite bead and connect 1.8V instead.  Logic inputs need to be driven from open drain output to prevent latchup.

hartytp, 2017-11-20:
> I can try to make modification to see how much we save.

In my view, this is worth doing, but not as a high priority. Let's finish basic testing first!

AFAICT, there is no major downside to using 1V8 in terms of noise or bandwidth. I think the only reason was that if we use 1V8 we can't use LVPCL outputs. There was a concern that LVDS would provide worse noise performance (in particular w.r.t. EMI) than LVPECL. However, if this turns out not to be a problem then there is no reason not to switch to 1V8 AFAICT.

hartytp, 2018-01-07:
Based on the measurements in #354, there is probably no need for the pre-amp current sources in Urukul: they don't save enough current to be worth the extra complexity, and the linearity of the design seems fine as it is. Top post edited accordingly.

We can reconsider this for BaseMod, where we have much more current thirsty pre-amps and more stringent performance demands.

Running the clock buffers from the 1V8A rail does still seem like a good idea, but needs testing first.

gkasprow, 2018-01-10:
@hartytp I added 0603 footprint as a place to move ferrite bead to test 1.8V supply for SIlabs chips.
I also added 470R resistors in series with input lines to limit the current when driving 1.8V chips from 3.3V CPLD.

hartytp, 2018-01-11:
@gkasprow Thanks! Can we make the 1V8 option the default for the next hardware revision?

jordens, 2018-01-12:
Only if that was actually tested and phase noise measured.

jordens, 2018-01-12:
Let's close this here and move the decision whether to change the default or/and whether to reduce the options to v1.2 or later.

closed 2018-01-12 by jordens

hartytp, 2018-01-14:
Wouldn't it be better to keep it open then and just change the milestone?

hartytp, 2018-01-14:
Okay, just seen you opened a new issue. Fine.

jordens, 2018-01-14:
Since this issue had drifted a lot from the original "power savings" general topic, I decided to draw a line and isolate to the only remaining item.

hartytp, 2018-01-14:
Sure, thanks for doing that. I just didn't see that you'd done it.