sinara-hw/sinara #340: Urukul Panel

gkasprow, 2017-10-22:
@jordens How do you want to have LEDs marked on the panel?
[image]

jordens, 2017-10-22:
No Labels for the LEDs (other than pwr and temp). put the v1.0 on there. Use the same location for the logo, the board name, description, and version as for the other boards like the sma dio. Otherwise very good!

gkasprow, 2017-10-22:
@jordens location for name and logo is same.

gkasprow, 2017-10-22:
[Here](https://github.com/m-labs/sinara/tree/master/ARTIQ_ALTIUM/PANELS/Project%20Outputs%20for%20PANELS) are producction files

closed 2017-10-22 by gkasprow

hartytp, 2017-10-22:
Looks really nice @gkasprow!

When are the prototypes due in?

gkasprow, 2017-10-22:
@hartytp I expect them within 2 weeks.

hartytp, 2017-10-22:
:)