sinara-hw/sinara #341: Urukul pinouts

jordens, 2017-10-25:
In light of #328 , @gkasprow , could you 

- [ ] point out the exact commit that the manufacturing data is based on, and
- [ ] ould you give me the final PDFs and 
- [ ] CPLD pinouts so that we can start writing the CPLD code?

gkasprow, 2017-10-25:
It is tagged production_v1
pdf are [here](https://github.com/m-labs/sinara/tree/master/ARTIQ_ALTIUM/Kasli/3U/PCB_3U_DDS/Project%20Outputs%20for%20PCB_3U_DDS)
CPLD pinout is [here](https://github.com/m-labs/sinara/blob/master/ARTIQ_ALTIUM/Kasli/3U/PCB_3U_DDS/CPLD_code/CPLD_pinout.xlsx)

jordens, 2017-10-26:
@gkasprow 
* You added three tags to commit f8354065 
* I removed all of them because that commit is from May.
* Which commit corresponds to the altium files that the production was based on?

jordens, 2017-10-26:
Is f2c83b2 correct?

gkasprow, 2017-10-26:
I added by chance, then removed but it failed to sync after that.
Yes f2c83b29697ec44a3b97be62d8c359dcdbd4cefd is production version

jordens, 2017-10-26:
* I created https://github.com/m-labs/sinara/releases/tag/Urukul%2Fv1.0 at that commit
* and uploaded the PDFs and pinout xls to that release.
* Could you upload Gerbers, XY and BOM (preferrably all in a zip or tar.gz file) to https://github.com/m-labs/sinara/releases/edit/Urukul%2Fv1.0

gkasprow, 2017-10-26:
done

jordens, 2017-10-27:
@gkasprow could you please clean up https://github.com/m-labs/sinara/releases?
Delete your tags, also from your working tree and from @marmeladapk

closed 2017-11-01 by jordens