sinara-hw/sinara #342: Kasli power supplies

hartytp, 2017-10-27:
Targeting Kasli v1.1 (i.e. *not* for the initial prototype round):

We should consider whether we can save some power by shifting things on to lower power rails. e.g. can we power the clock buffers from the 2V5 rail rather than the 3V3 rail? Can we run the LVDS from 1V8? etc...

marmeladapk, 2017-10-27:
LVDS has to stay 2V5 unless we move from Artix to Kintex (1V8 LVDS is available only in HP banks which appear only in Kintex or higher families).

marmeladapk, 2017-10-27:
One thing we could do is to switch RAM from 1,5 V to 1,35 V.

hartytp, 2017-10-27:
> LVDS has to stay 2V5 unless we move from Artix to Kintex (1V8 LVDS is available only in HP banks which appear only in Kintex or higher families).

Good point.

Anyway, the reason I bring this up is that IIRC, the thermal simulations suggest that even with decent cooling Urukul could be a bit marginal. Kasli dissipates quite a bit more power than Urukul (16W v 9W). 

Things probably won't be too bad in practice, since IIRC the power budget is a real worst case scenario (e.g. it assumes we add populate clock biasing resistors which are currently DNP). But, it might be worth looking for any low-hanging fruit/places where we can shave a bit off the power budget.

hartytp, 2018-01-19:
We plan to use a 2V5 LDO for the clock buffers, so no power savings possible there.

@marmeladapk @gkasprow can you see any other places we can easily save some power on Kasli?

AFAICT, the answer is "no" so I'm closing this issue.

closed 2018-01-19 by hartytp

marmeladapk, 2018-01-19:
@hartytp Switching RAM to 1,35 V (it's not implemented).

hartytp, 2018-01-19:
If that's the only power saving option then IMO we should leave things as they are.

IIRC there were downsides to switching the RAM voltage and the power savings weren't that great.