sinara-hw/sinara #344: Urukul CPLD code for testing

jordens, 2017-11-01:
@gkasprow how do plan to test Urukul, the DDS, and the attenuators? Does the code I wrote to implement the current Urukul design specification work for you (https://github.com/m-labs/urukul)? Are you going to roll your own testing code that just connects one DDS at a time so you can use the ADI eval board?

(from #292)

gkasprow, 2017-11-01:
@jordens I planned to use ADI tools + usb->SPI converter to talk to attenuators in a way I did with Novo board. CPLD is another story, I will try to use your code.

jordens, 2017-11-01:
Does the converter work with an CS configuration that is not "one-hot" but a chip address?

gkasprow, 2017-11-01:
I can generate any state because it is based on FPGA with FORTH processor.

jordens, 2017-11-01:
Ok.

closed 2017-11-01 by jordens

gkasprow, 2017-11-19:
@jordens could you please briefly describe functionality of the code you wrote? I'd like to just burn FLASH and start testing without analysing it.

gkasprow, 2017-11-19:
I've just noticed that Python code is self-explanatory :)

jordens, 2017-11-19:
Yes. Should be well documented. If you hit issues, please file them over there.

gkasprow, 2017-11-20:
@jordens could you build this code? I just need binary file.

jordens, 2017-11-20:
It's all in the releases.

jordens, 2017-11-20:
And please also file questions like that one over there.

gkasprow, 2017-11-20:
@jordens I don't see CPLD binaries in releases...

jordens, 2017-11-20:
There are .jed, .isc and .vm6 files. Those are the "binaries".

gkasprow, 2017-11-20:
@jordens I don't see them. Point me please.

jordens, 2017-11-20:
* https://github.com/quartiq/urukul -> release -> v1.0rc5 ->
* https://github.com/quartiq/urukul/releases/download/v1.0rc5/urukul_v1.0rc5.tar.gz

gkasprow, 2017-11-20:
thanks, I looked at wrong file then.