sinara-hw/sinara #347: Kasli v1.1 cost savings

jordens, 2017-11-02:
# Do not act on this now. Wait for v1.1.

* [ ] consider either splitting all 47 µF into 22 µF or merging/upgrading all 22 µF to 47 µF to save a BOM line

marmeladapk, 2017-11-06:
- [ ] Improve signal integrity for backplane LVDS lines (remove reference plane cut)

hartytp, 2017-11-06:
> determine cost contribution of likely unused parts: MMCX connectors, clock fanouts, make them optional BOM items.

FWIW, unless there is a noise issue with this clock source, we will definitely make use of these connectors for our DDS clocks. I imagine that other groups using the DDS will as well.

While I think the aim of trying to reduce cost/complexity/power is laudable, @gkasprow has already commented that the MMCX connectors cost $1 each, so IMO the extra time/cost/hassle associated with maintaining different population options far outweighs the cost savings from removing these components.

NB the clock fanout is also required as a mux to select the clock source fed to the backplane, so removing it will cause additional problems.

jordens, 2017-11-07:
I'd like to know how much the composition of the cost per board and how much these actually contribute to the price. From the board house.
AFAIK you would also be happy with only 4 instead of 7 mmcx.

hartytp, 2017-11-07:
> AFAIK you would also be happy with only 4 instead of 7 mmcx

Sure, 4+ is fine IMO if you want to save a few $

hartytp, 2018-01-19:
Are there any cost savings we want to make to Kasli?

AFAICT, there are no easy places to save money on the design (but correct me if I'm wrong about that @gkasprow @marmeladapk) so I'm closing this.

closed 2018-01-19 by hartytp