sinara-hw/sinara #349: Kasli rev 1.0 errata

marmeladapk, 2017-11-06:
To be changed for rev1.1
- [ ] Improve signal integrity for backplane LVDS lines (remove reference plane cut)

jordens, 2017-12-05:
@marmeladapk @gkasprow Is Kasli back from the board house?

gkasprow, 2017-12-05:
Components are delivered, the PCBs were shipped yesterday. They will start assembly on Monday. It takes roughly 2 days.

hartytp, 2017-12-05:
Wonderful! Really looking forward to seeing these!

gkasprow, 2017-12-15:
Kasli photo:
[image]

gkasprow, 2017-12-15:
Power supplies work well. It consumes 0.2A@12V
USB works as well. At least I see 4 serial ports under Windows.
[image]
That's all what I can do at home.

hartytp, 2017-12-15:
Great!

jordens, 2017-12-15:
Awesome!

gkasprow, 2017-12-18:
Progress;
JTAG works,
FLASH programming and booting works. There is litle bug, R57must be3 00R
I loaded design which causes 2.5V rail to switch off. With little help of external PSU it wakes up but the board takes 600mA and FPGA gets hreally hot.
Then we realised that it must be split termination that is switched on on all EEM LVDS lines because there are just ordinary registers attached so far.
And indeed, the construct below instantiates uncallibrated split impedance
```

-- LVDS bidir buffers
	eem0 : for I in 0 to 7 generate
		lvds_buf : IOBUFDS_INTERMDISABLE
			GENERIC MAP (
				DIFF_TERM         => "true",
				IBUF_LOW_PWR      => "true",
				IOSTANDARD        => "LVDS_25",
				USE_IBUFDISABLE   => "false")
			PORT MAP (
				O                 => eem0_I(I),
				IO                => eem0_p(I),
				IOB               => eem0_n(I),
				INTERMDISABLE     => eem0_term_dis(I),
				I                 => eem0_O(I),
				IBUFDISABLE       => '0',
				T                 => eem0_S(I) -- 3-state enable input, high=input, low=output
			);
```

gkasprow, 2017-12-18:
clock distribution works as well

jordens, 2017-12-18:
Great job @gkasprow and @marmeladapk.

gkasprow, 2017-12-18:
@jordens The only remaining thing we need to test is Silabs, I2C distribution and EEMs itself. Hopefully we will ship it in Wednesday. We can ship it as it is tomorrow and play with remaining boards that will be delivered on Wednesday.

jordens, 2017-12-18:
No need to rush it then. I'll be gone Friday until the 28th. You can time it so that it arrives at the 28th.

gkasprow, 2017-12-19:
I2C and silabs chip also work.

gkasprow, 2017-12-19:
UART works as well. I talk with FORTH CPU

marmeladapk, 2017-12-26:
@jordens I removed arcs remark, because they were added on purpose - reducing spikes of current density. Same as with you last remark about polygons (believe me, Greg bugged me about this).

We need to switch UART with I2C on FTDI chip, because channel C doesn't have MPSSE. I made bitbang I2C using pyftdi which works (I can program Si5324 and I2C switches), but its timings are horrible. 

SFP0 and SFP2 seem to work on 6,25 Gbps. There is a problem with SFP1 TX line.

[image]
[image]

marmeladapk, 2017-12-26:
@jordens Also FPGA tends to get quite hot with termination enabled so please cool it with fans to avoid damage. Also, there is some kind of a problem with XADC, sometimes it shows all voltages = 0 V and temperature = 0 K.

hartytp, 2017-12-27:
> @jordens Also FPGA tends to get quite hot with termination enabled so please cool it with fans to avoid damage. Also, there is some kind of a problem with XADC, sometimes it shows all voltages = 0 V and temperature = 0 K.

Should we add a heat sink to the FPGA to help to avoid this? Maybe a nice PCB-mount one?

jordens, 2017-12-29:
@marmeladapk Current density at a bend diverges with sharpness/mesh size in many Maxwell approximations. If you think you need an arc, that's most likely a simulation/algorithm artefact which does not correspond to reality. I'd be very surprised if there was a case on Kasli where an arc solves a problem that pair of 45 degree bends does not solve. Both on polygons and on traces.

Ack the UART/I2C switch. Could you post the bitbang i2c code somewhere?

Don't worry about the XADC. Did you test the DDR RAM?

On the GTP pairs you need to be a bit more careful with your design rules. You have some lines coupled closer to unrelated signals than to their differential pair partner!

jordens, 2017-12-29:
@marmeladapk can't get jtag over usb to work. are DIR1 and DIR2 of IC19 wrong?

jordens, 2017-12-29:
[image]
3.5 mil gap near IC18. Is that ok? Did they fix that in the fab?

sbourdeauducq, 2018-01-03:
@marmeladapk @gkasprow ping

gkasprow, 2018-01-03:
@jordens in IC17, the DIR is H so it means A -> B direction which is correct
But on IC19 channel 1 direction is B->A which is wrong.
Channel 2 direction is A -> B which is also wrong. This explains why it does not work
You can simply short pins 8, 5 and 9,4, we will fix it in the right way on remaining boards.
[image]

jordens, 2018-01-03:
Yes, that's what I said.
Could you implement your fix and show me (photo) what you mean? Just shorting still has the same contention issue. You know better where to cut and scratch the lines.

jordens, 2018-01-03:
I have MiSoC running (spi flash XIP, boot from spi flash, SDRAM working, uart working).

gkasprow, 2018-01-04:
Shorting of the input with output via 220R resistors should do the job. Resistor is to limit current that will flow from FTDI to FPGA.
Spartan devices are known to react with latchup when 2.5V JTAG signal is treated with 3.3V logic output :)

jordens, 2018-01-04:
Still unclear what you are suggesting. Desoldering the level converter? Cutting, scratching the traces on both sides and bridging over the converter?

jordens, 2018-01-04:
I desoldered the level converter and added a short on TDO and 1k on TMS. JTAG over FTDI also works with 25 MHz. That seems to have saved nearly 500 mW of contention current. 3.5 W now.

jordens, 2018-01-04:
No. That was the SFP transceiver. 4 W with, 3.5 W without.

gkasprow, 2018-01-04:
This is exactly what I meant :)

jordens, 2018-01-08:
On my board (number 2, thanks for putting the sticker on there) port 5 of IC14 has SDA (EEM1:SDA) shorted to ground.

marmeladapk, 2018-01-08:
On board 3 EEM1 SDA is pulled to 3V3. Could you check if R157 is 2k2 and is connected to 3V3?

[image]

jordens, 2018-01-08:
The 2k2 are there but they don't matter. The resistance to ground is 300 Ohm (310 in one polarity and 270 in the other). No EEM connected. Either the chip is broken or there is some conducting dust ;).

marmeladapk, 2018-01-08:
@jordens Could you share your openocd setup?

jordens, 2018-01-08:
kasli.cfg: 

```tcl
interface ftdi
ftdi_device_desc "Quad RS232-HS"
ftdi_vid_pid 0x0403 0x6011
ftdi_channel 0
ftdi_layout_init 0x0008 0x000b
ftdi_location 1:8

reset_config none
transport select jtag
adapter_khz 25000

source [find cpld/xilinx-xc7.cfg]
source [find cpld/jtagspi.cfg]
```

* The `ftdi_location` is only needed to distinguish multiple FT4232Hs on a machine. It needs to match the bus topology.
* To load a bitstream: `openocd -f kasli.cfg -c 'init; pld load 0 misoc_minisoc_kasli/gateware/top.bit; exit'` 
* To write bitstream and bios to flash: `openocd -f kasli.cfg -c 'init; jtagspi_init 0 /home/rj/src/conda/pkgs/bscan-spi-bitstreams-0.10.0+git-1/share/bscan-spi-bitstreams/bscan_spi_xc7a100t.bit; jtagspi_program misoc_minisoc_kasli/gateware/top.bin 0; jtagspi_program misoc_minisoc_kasli/software/bios/bios.bin 0x400000; exit'`

The spi proxy bitstreams are either in our openocd conda packages or at https://github.com/jordens/bscan_spi_bitstreams

And if you want to play I2C, I pushed my toy code to https://github.com/quartiq/kasli-i2c

jordens, 2018-01-15:
As @sbourdeauducq reported elsewhere, Kasli Ethernet is working. On the board here ("#2") SFP1 also works fine.

marmeladapk, 2018-01-15:
@jordens To build Kasli target I pulled latest Artiq repo and I wanted to update conda environment to it using `conda env update -f conda/artiq-dev.yaml`. However this didn't pull latest migen and misoc with changes for Kasli, so I had to do this manually to be able to build gateware. Is this expected behaviour?

Also, build failed with errors:  `Rule violation (IOSTDTYPE-1) IOStandard Type - I/O port eem0_n is Single-Ended but has an IOStandard of LVDS_25 which can only support Differential`. I used Vivado 2016.4. Is this known or is there some mistake on my side?

jordens, 2018-01-15:
@marmeladapk I didn't know about `conda update` but `-f` is force and I don't think you can use it in this case. Just install the latest with `conda install -c m-labs/label/dev misoc migen openocd`. I don't know about the rule violation. I don't see those. We use `2017.2` or `2017.4`. The [buildbot](http://buildbot.m-labs.hk/waterfall) will generate binary packages for Kasli soon.

marmeladapk, 2018-01-15:
I'll check with 2017.4, but generally I got those errors when in constraints eem lines were set as LVDS, but they didn't have IOBUFDS instantiated in code.

jordens, 2018-01-15:
@marmeladapk Make sure that the misoc/migen that are being used are actually the ones installed through conda and not some packages that you `git clone`-ed yourself and `setup.py install` or `pip install`-ed. But I can see unused inputs causing that warning. Pretty sure it's benign.

marmeladapk, 2018-01-15:
Huh on board 4 and 6 SFP1 also works. Seems like I didn't do the test properly.

I tested last link using @gkasprow SATA-SFP converter. Below is a screenshot of all links working simultaneously at 6,25 Gbps. 
[image]

jordens, 2018-01-17:
Si5324 clock distribution noise and spurs [running from on-board XO](https://github.com/quartiq/kasli-i2c/blob/6e1f137c0faa55276fb8bd6c1e24958200a9c0e5/i2c.py#L392-L402). Looking at J1, noise for the low-frequency part mostly limited by the SA. The one spur at ~449 kHz seems to be the SMPS. It wanders if I touch the SMPS components.
[image]
[image]

hartytp, 2018-01-17:
So, something like -100dBc/Hz at 100kHz offset? We expect something like -130dBc/Hz based on the datasheet. Wonder what limits that.

jordens, 2018-01-17:
The [SA](http://www.siglentamerica.com/USA_website_2014/Documents/DataSheet/SSA3000X_DataSheet.pdf) is the limit on noise density everywhere but not on spurs.

gkasprow, 2018-01-17:
The Silabs supply was not designed to be very low noise. We can add LDO.

jordens, 2018-01-17:
I'd do that but also remove the clock fanout in favor of a single MMCX output form the Si5324 and then using Clocker or/and the backplane. Kasli is too dense and overloaded. The additional space would help with figuring out a good heatsinking solution, better SI, cleaner routing, higher yield, and lower manufacturing risk.

Also I would like to free up six pins on the FPGA by having MOD_DEF1 and MOD_DEF2 routed through the I2C switch only and not additionally directly to the FPGA. I am not aware of an SFP module that uses those pins for non-I2C applications.

jordens, 2018-01-17:
But let's get v1.1 into the fab first. We can't wait. Or is it already there? @gkasprow

gkasprow, 2018-01-17:
@jordens do you need these FPGA pins for something else? Or just test point?
The SFP+ modules have I2C, while some SFP ones use these pins for bandwidth control.

gkasprow, 2018-01-17:
I've just prepared new revisions for: Urukul, BNC, SMA, TTL tester, Sampler.
Now I'm working on Zotino, @marmeladapk will update Kasli ASAP.

hartytp, 2018-01-17:
> The Silabs supply was not designed to be very low noise. We can add LDO.

If we can do this without adding too much additional complexity then I think it would be a good idea. Would it need an extra SMPS on Kasli though to produce the voltage that drives the LDO (or, can we use a 2V5 LDO driven from 3V3 for the Si chip?).

> I'd do that but also remove the clock fanout in favor of a single MMCX output form the Si5324 and then using Clocker or/and the backplane. Kasli is too dense and overloaded. The additional space [...]

If it really does help then I'd be okay with this (although, it's definitely not something I'd like to see happen). However, before we do it I'd like @gkasprow to confirm that removing that clock buffer will really make a significant (at least noticeable) difference to any of the things you've said. e.g. I'm actually not convinced that removing that buffer will improve Kasli's SI or yield. Also, don't we still need a clock buffer to drive the BP clock lines as well as that MMCX?

jordens, 2018-01-17:
@gkasprow AFAICT all modules use the rate select pins (RS0 and RS1) for bandwidth control. The MSA mandates I2C on MOD_DEF1/2.

@hartytp We have an unused diff output and one unused single ended on IC1 and an unused diff on IC2. Have the diff IC2 output go to the backplane, the single ended IC1 to an MMCX.
I am pretty sure the effects on routability, SI and cleanliness will be there. If the 7x MMCX array and the ADCLK948 are gone, that's a significant amount of space on all layers.

@marmeladapk And please finally remove the wire bridge W1. Even if it is DNP, it pierces through all layers and takes up lots of space.

jordens, 2018-01-17:
@gkasprow I'd rather like to use these pins for:

1. RATE_SELECT1 (pin 9 of the SFP). It also gets rid of IC10{ABC}.
2. Reset of the I2C routers. If there lines shorted, on the I2C output ports (as there is on my board), there is no way to get out of that situation and reconfigure the I2C routers other than power cycling. That's bad.
3. Add test points and one more USER_LED.

jordens, 2018-01-17:
You can put the EEPROM on the same I2C port as the SI5324. But I guess you have figured that out as well.

jordens, 2018-01-17:
And I have changed my mind on the second power connector that is pointing towards the backplane. That can go (removed entirely). If someone wants to power internally, then just use the backplane connector, that EMI filter, and a connector on the backplane. In turn please populate the backplane connector.

hartytp, 2018-01-17:
> I am pretty sure the effects on routability, SI and cleanliness will be there. If the 7x MMCX array and the ADCLK948 are gone, that's a significant amount of space on all layers.

Well, it's not a deal breaker for me, but I'd still prefer not to remove these unless @gkasprow and @marmeladapk really think it will help (they are the ones best placed to make that call). I would have no objection to cutting down to 4-5MMCX connectors if that helped (although, we discussed this before and, IIRC, @gkasprow and @marmeladapk said there was no point removing them since the cost is negligible and removing them doesn't really help with the design).

> And I have changed my mind on the second power connector that is pointing towards the backplane. That can go (removed entirely). If someone wants to power internally, then just use the backplane [...]

Great! I completely agree. Let's just make the BP adapter with a power jack on it.

gkasprow, 2018-01-17:
@jordens true, I met some SFP modules that used these pines in other way, but they were custom CERN designs that also used both RD and TD as outputs :)

gkasprow, 2018-01-17:
@jordens @marmeladapk On two kasli boards we noticed failed I2C switch. This may be due to ESD. The I2C are open drain with no protection diode. So we should add TVS to each EEM connector I2C lines.

hartytp, 2018-01-18:
@jordens re the clock buffer: this is still a high value feature for us.
- I don't anticipate any noise issues with the 125MHz clock from the Si chip (assuming we add an LDO for it or at least some better PSU filtering). If we get close to matching the datasheet performance then it will as good as the AD9910 PPLs. Also, most of our Urukul applications aren't that noise sensitive anyway, so a few spurs won't hurt us.
- For Kasli DRTIO slaves, it's really convenient not to have to supply a clock via coax. This makes wiring much nicer. So, it's really nice to have a clock source output at the RTIO_COARSE frequency. That way, Kasli alone can supply everything  (power, DIO and clocks) needed to run EEMs.
- We generally have 100MHz references but not 125MHz/150MHz references, so it's really useful to have Kasli produce the 125MHz/150MHz we need to clock our DDSs from. It's also nice that we don't need to buy new reference oscillators for our DDSs if we change the RTIO frequency.
- We can use an external clock buffer such as clocker if we loose the fanout buffer on Kasli (so long as we keep at least an MMCX clock output). But then we have to source power for that (NB we plan to use all EEM connectors on Kasli, so there won't be one spare to power clocker in many cases). It also increases the cost of the overall system, takes up more rack space etc.
- The clock fanout isn't just a fanout but also a mux for the EEM clock fed to the backplane connector. If we loose the mux then we loose the ability to feed an external clock to the BP connector (we then *only* have the Si chip).
- While I understand your concerns, we have so far seen no evidence that the clock fanout chip on Kasli causes any SI issues etc on the prototype, so there is no actual evidence to back your concerns up. Also, @gkasprow and @marmeladapk -- the experts doing the design -- haven't yet agreed that the concerns are well founded.

hartytp, 2018-01-18:
But, if you're really set on removing it then a possible compromise would be:
- Route the RTIO_coarse clock to the BP
- Keep a single MMCX on Kasli which also outputs the RTIO_coarse clock from the Si chip
- Put a 4x fanout + MMCXs on the Kasli BP adapter. Then we can use that for our applications.

jordens, 2018-01-18:
@hartytp I am OK with exposing the clock at a few places (the backplane, an Kasli MMCX and the panel SMA). But I have the following observations with the 8x clock fanout:

* We definitely need additional space for an LDO for IC2, IC1.
* Kasli is extremely dense. I have been looking at the layout for the last couple weeks and I really don't like the fact that the density has forced Pawel to place all kinds of unrelated logic groups inside of each other. That makes debugging hard, makes reading the layout hard, and is just bad style. The fact that you can cram that much logic onto a board does not mean that you should do it. There are enormous downsides. It is also bad when changing things in the design as there are unnecessary knock-on changes and large parts have to be rerouted. The attitude of making future changes harder just because there is minute short term benefit is not a good one.
* The 7 or 8 coax and MMCX connectors make assembly more tricky and lead to a worsening of the cooling situation. It obstructs cooling (they run straight over the FPGA where the heatsink should be). Moving the 8x buffer of Kasli somewhere else would save a significant amount of power (and temperature) especially in the high density areas of IC1 and IC2.
* You can still do your 125 MHz fanout. You'll get that in three places: the SMA at the front, the backplane and the MMCX on Kasli. Just use Clocker. Otherwise, what purpose does Clocker serve since you would be getting 7 MMCX from Kasli? Using Clocker (or a fan-out on the backplane) makes wiring nicer as it reduces contention in the area where things are worst.
* You have to decide what's important for you. You say that the 125 MHz are "rare", but then you still want the option to not use them but feed the backplane with something else. This weakens both arguments.
* How would you use the MMCX input to the mux on Kasli if not from Clocker? That makes the mux on Kasli pretty useless as Clocker already gives you those outputs?
* Have the 2x8 fanout/mux on the backplane. You get the same mux options as before. And 8 outputs is sufficient to drive even setups with 8 Urukul/Mirny. That's unlikely to be standard.
* Sure there is evidence for SI issues. As Pawel and Greg stated, the routing is extremely dense and that lead to SI issues in the EEM connectors.

hartytp, 2018-01-18:
> The 7 or 8 coax and MMCX connectors make assembly more tricky and lead to a worsening of the cooling situation. 

I would be totally happy with reducing that to 4.

> You have to decide what's important for you. You say that the 125 MHz are "rare", but then you still want the option to not use them but feed the backplane with something else. This weakens both arguments.

IIRC, you were the one who wanted that. I'd be fine with loosing the external input and only supporting backplane clocks from the Si chip. Just pointing out that we loose the muxing capacity if we remove the fanout. But, ...

> Have the 2x8 fanout/mux on the backplane. You get the same mux options as before. And 8 outputs is sufficient to drive even setups with 8 Urukul/Mirny. That's unlikely to be standard.

That's true. I hadn't considered that. The only downside of this is that there is no convenient way of feeding this mux via the front panel.

> Otherwise, what purpose does Clocker serve since you would be getting 7 MMCX from Kasli?

That was more targeting ultra-low noise cases. I'd prefer not to have to use if for simple fanout unless really necessary.

[133 more replies]

closed 2018-02-28 by marmeladapk