sinara-hw/sinara #350: SAYMA AMC SPI Flash

a-shafir, 2017-11-06:
On SAYMA_AMC both SPI bitstream (QSPI0) and config (QSPI1) does not work as expected.

According the schematic  there is something unusual about  how the SPI flash connected. It is shared between the FPGA and LPC ARM MCU using IC42 switch.

According
https://www.xilinx.com/support/documentation/user_guides/ug570-ultrascale-configuration.pdf 
CCLK is a critical signals and shall be routed carefully.

Looks like the design leads to uneven load on the QSPI signals, different traces length, traces splitting (signal reflections) etc. Also there is no termination resistors.

SPI flash is working via JTAG bit-proxy. So i think it is possible to try following:
1. reduce CCLK rate (BITSTREAM.CONFIG.CONFIGRATE) the default is 33MHz that can be too high.
2. try BITSTREAM.CONFIG.SPI_BUSWIDTH = 1
3. try BITSTREAM.CONFIG.SPI_FALL_EDGE = YES/NO
(see https://www.xilinx.com/support/documentation/application_notes/xapp1233-spi-config-ultrascale.pdf )

Also need to make sure the EN_FLASH_UPDATE (driven by LPC MCU) is inactive by default otherwise the load on the FPGA_CCLK etc is even higher.

gkasprow, 2017-11-06:
@a-shafir On my boards QSPI [works perfectly](https://user-images.githubusercontent.com/4325054/27648159-ee6557fa-5c2d-11e7-98ac-7dea3401fa27.png). FPGA boots below one second. There are two resistor options. One is dedicated to direct Kintex QSPI configuration where Artix FPGA is configured using Kintex IO pins.
Second relies on combined configuration where Kintex is configured from QSPI and later on it configures Artix chip.
That's why it is so hard to terminate CClK. On can add 50R resistor to have some kind of source termination.
IC42 is foreseen as update path where user can access FLASH via MCU over Ethernet. It is placed very close to memory chip so does not cause SI issues.
Such trace differences do not matter at such speed.
So if there are some issues, please make sure that resistors responsible for master-slave configuration (R176--R180, R253) are not soldered.
Second FLASH has wrong CS connection which is already clarified here:
https://github.com/m-labs/sinara/issues/209
There is also noted that CCLK needs termination.

hartytp, 2017-12-01:
Anything still need doing here, or can we close this?

jordens, 2017-12-04:
Tracked at https://github.com/m-labs/artiq/issues/846 as we can probably find at least a software/gateware workaround.

closed 2017-12-04 by jordens

hartytp, 2017-12-04:
> Tracked at m-labs/artiq#847 and m-labs/artiq#846 as we can probably find at least a software/gateware workaround.

Good to know!

To double check: there is nothing about this that you'd like to change for Sayma v2.0?

jordens, 2017-12-04:
There are a couple of changes we'd like to see. Rtm should get a flash, slave serial should be properly implemented, the second amc flash should be properly wired.from the top of my head.

hartytp, 2017-12-04:
Should we reopen this and tag with Sayma v2.0?

jordens, 2017-12-04:
No. Iirc those were filed elsewhere already. Or they should be. I don't want to mix two different issue areas with at least as many different responses.

hartytp, 2017-12-04:
> No. Iirc those were filed elsewhere already. Or they should be. I don't want to mix two different issue areas with at least as many different responses.

ACK. I'd forgotten they were filed elsewhere. Sorry for the noise.