sinara-hw/sinara #351: Kasli BP adapter

hartytp, 2017-11-06:
@gkasprow @marmeladapk To test (and use) the IO on the Kasli BP connector, we need a simple breakout board as discussed previously. This should be:

- Small board with 96-pin din, which mates with Kasli and has mounting holes compatible with standard Schroff racks
- 4 EEM connectors outpus on
- Should break out the clock line to an MMCX (via balun). Should also have a 100R resistor that can be used to terminate the clock line if unwanted. (DPN one of the MMCX and the termination resistor).
- If we really want to remove the MMCX connectors from Kasli, we could put a fanout buffer on this board and add a few more MMCX connectors. However, I'd like to have a play with this first to see how easy it is to route these cables.

gkasprow, 2017-11-07:
@marmeladapk In the lab there is example 3U backplane for which I have Altium project. Treat it as template. Check if mounting holes fit the rack holes.
On Kasli there is 125mA fuse on 3.3V supply. This may be not sufficient for clock buffer so either additional buck regulator is needed or we need to replace the fuse on Kasli.
@hartytp do you need the DC jack on this BP adapter?

gkasprow, 2017-11-07:
@hartytp do you think we should mount DIN connectors on proto Kasli series?
How many of such BP shall we produce?

hartytp, 2017-11-07:
> @hartytp do you need the DC jack on this BP adapter?

No, I do not need a DC jack on the BP adapter. I'm not against you adding one if you want, so long as it's easy to access it in standard standard racks.

> @hartytp do you think we should mount DIN connectors on proto Kasli series?

If possible, I would like the DIN connector mounted on all the Kasli prototypes that you order for us. IIRC, @jordens doesn't want the DIN connector fitted on his Kaslis.

hartytp, 2017-11-07:
> How many of such BP shall we produce?

Can you produce one BP adapter for each Kasli you ship to us, please?

Thanks!

hartytp, 2017-12-18:
@marmeladapk Did you make this adapter in the end? It will be useful for testing (and using!) the BP lines on Kasli

marmeladapk, 2017-12-18:
Sorry, forgot about that, but I can do this after testing Kasli.

Currently, the one Kasli board we have doesn't have a backplane connector. Even with this backplane in v1.0 may be of limited use because of reference plane split under some of LVDS lines. We'll measure how much it affects signals.

hartytp, 2017-12-18:
Thanks for the update. No huge rush for this -- do it when you have time.

hartytp, 2017-12-18:
> Even with this backplane in v1.0 may be of limited use because of reference plane split under some of LVDS lines.

Out of curiosity, what do you mean by this? Do you mean that some of the LVDS lines are routed on a layer with no adjacent ground layer? So, you're worried that this affects the impedances of the traces etc and will cause SI issues?

> We'll measure how much it affects signals.

Great, will look forward to seeing the data.

marmeladapk, 2017-12-19:
@hartytp No, it just means that this reference plane is not continuous, so I'm worried about return currents going through unexpected places.

It's on a list of fixes for the next revision though.

gkasprow, 2017-12-19:
@hartytp I wouldn't worry that much about it. It is not 10Gbit. This will give some additional reflections but should not be an issue since LVDS is balanced. Anyway we will fix it in next rev.

hartytp, 2017-12-19:
> @hartytp I wouldn't worry that much about it. It is not 10Gbit. This will give some additional reflections but should not be an issue since LVDS is balanced. Anyway we will fix it in next rev.

@gkasprow I'm not too worried about it for exactly that reason. I suspect this won't be a problem, but it's always nice to have the data to be sure ;)

hartytp, 2018-01-11:
@marmeladapk Did you produce the Kasli BP adapter yet?

hartytp, 2018-01-16:
@marmeladapk ping. Did you produce this adapter yet?

marmeladapk, 2018-01-16:
@hartytp Not yet, will do this this week. Do you want some kind of clock output?

jordens, 2018-01-16:
@marmeladapk Are the designs for the next batch of Kasli boards already at the factory (like Urukul and DIO_BNC/SMA)? If not, could you please do that first?

hartytp, 2018-01-16:
> @marmeladapk Are the designs for the next batch of Kasli boards already at the factory (like Urukul and DIO_BNC/SMA)? If not, could you please do that first?

Agreed, that should take priority.

> @hartytp Not yet, will do this this week. Do you want some kind of clock output?

What I would do is route the clock to a MMCX connector on the board (via a balun), which we can use the MMCX to test signal integrity of the BP clock.

I'd also add a 100R termination resistor (DNPd for now) that can be used in case we decide not to populate the balun + MMCX later on.

hartytp, 2018-02-05:
@marmeladapk Did you have a chance to do this yet?

marmeladapk, 2018-02-06:
[KASLI.PDF](https://github.com/m-labs/sinara/files/1698723/KASLI.PDF)

@hartytp Yes, take a look at schematics if everything is ok.

hartytp, 2018-02-06:
@marmeladapk Thanks! Looks good to me. Few minor comments:

- Add hardware version number
- Label connectors J4-J7 EEM8-EEM11
- Annotation on power connector 5A max. Maybe also annotate with the mating part number?
- Does this have the correct power connector? If so can we remove the "Change to 721A" annotation?
- Are two mounting holes enough, or should this have 4 (can't picture how big it is).

marmeladapk, 2018-02-06:
@hartytp [datasheet of BP connector](http://datasheet.octopart.com/09032966824-HARTING-Elektronik-datasheet-647123.pdf)

hartytp, 2018-02-06:
> @hartytp datasheet of BP connector

What's that for?

hartytp, 2018-02-06:
Are you mounting this board directly off the din 96? I would have though we should provide screw holes to allow the BP adapter to mount in a normal Schroff rack on the backplane rail.

jordens, 2018-02-06:
@marmeladapk Looks good.

* To make the power distribution we could do a few things. (1) Add another barrel connector. AFAICT paralleling those power supplies is not an issue. And IME daisy chaining those Kasli/VHDCI boards is really convenient. (2) Add one or two (dummy EEM?) connectors that only carry the 12V (e.g. for Clocker or Thermostat-like applications)
* To minimize cabling issues, the width of the backplane should be 40 HP (so that two can fit into a crate), the DIN connector should be at the left, and the 4 EEM connectors should be at the far end (right) spaced by 4 HP.
* DIN and EEM connectors facing one side and (at least one of) the power connector facing the other side. Use the straight version of the power connectors (as you noted in the schematic).
* Please use the crate mounting option where the board is mounted to the rails and not the connector (*(**don't** use that "Z-rail" scheme).
* Tom may want another Balun for that
* I think you want to DNP that 100R (R1).

marmeladapk, 2018-02-06:
@hartytp Oh, I thought you were wondering if two mounting holes are enough for this connector. Those are holes for connector.

jordens, 2018-02-06:
Do we want to bother with making room for one or two more DIN connectors to the right of J1 and 4 or 8 more EEM connectors left of J4-J7 so that we can implement the Kasli-EEM0-7-to-2xDIN adapter idea for free? I'd vote no.

hartytp, 2018-02-06:
> To minimize cabling issues, the width of the backplane should be 40 HP (so that two can fit into a crate), the DIN connector should be at the left, and the 4 EEM connectors should be at the far end (right) spaced by 4 HP (Kasli at 0 HP, EEM8-11 at 24, 28, 32, 36 HP).

I'm envisaging this as a small 8HP or so board, not a full-fledged backplane.

hartytp, 2018-02-06:
> Do we want to bother with making room for one or two more DIN connectors to the right of J1 and 4 or 8 more EEM connectors left of J4-J7 so that we can implement the Kasli-EEM0-7-to-2xDIN adapter idea for free (on the backplane: DIN8-11, DIN4-8, DIN0-3, EEM0-3, EEM4-7, EEM8-11) ? I'd vote no.

I'd vote no on this for now. This PCB is trivial and we can respin it later if needed, so let's keep it as simple and small as possible for now.

marmeladapk, 2018-02-06:
I also thought this would be a small board, same size as DIO modules.

jordens, 2018-02-06:
It doesn't make sense to keep this a small board (<40 HP). I don't see any benefit apart from a few quid in PCB cost. It's a very bad idea as it crowds more ribbon cables (and coax) into a corner where there is high density. And it's also mechanically imposssible: a stack of 12 ribbon cables on top of the EEM connector on the backplane will prevent any of the 160 mm long boards from being inserted there. You have to move that out of the way. Have you made a model or tried this yourself?

jordens, 2018-02-06:
If you absolutely want to keep this a small board, then you should (1) use the angled connector and not the one from your datasheet, (2) mount this board in-line with Kasli, (3) use deeper crates, and (4a) use the Z-rail scheme to mount the connector to the real, instead of the board or (4b) mount this board to Kasli fixed at the DIN connectors and get by without a Z-rail

hartytp, 2018-02-06:
@jordens I don't understand you here. You want to have a 40HP board which only has 4 IDCs on it? Remember,  we're not trying to build an actual backplane here, just a quick way of breaking out a few extra IDCs.

If you make the BP adapter 40HP wide then you block the space between the back of the rack and the EEM cards, which is where we plan to route our cabling. Having a small BP connector leaves this space open, giving plenty of room to route cables.

> It's a very bad idea as it crowds more ribbon cables (and coax) into a corner where there is high density.

The BP adapter doesn't really affect this, since it only has 4 EEM connectors on it. The vast bulk of the cabling still comes from Kasli unless you plan to build a full BP which, again, is not what we're doing here.

hartytp, 2018-02-06:
> To make the power distribution we could do a few things. (1) Add another barrel connector. AFAICT paralleling those power supplies is not an issue. And IME daisy chaining those Kasli/VHDCI boards is [...]

If there is space, adding an extra power jack to allow Kaslis to be daisy chained could be a nice feature (although, users have to be very careful about watching the max current consumption).

Personally, I wouldn't bother with dummy EEM connectors for clocker etc. IMHO, that's more likely to cause confusion than to actually help.

> Tom may want another Balun for that

Let's go for the same TCM2-43X+ that we've used elsewhere; it gives a slightly lower amplitude of signal at 100MHz than other baluns, but the edge rate is very good, which is what generally matters most.

hartytp, 2018-02-06:
> If you absolutely want to keep this a small board, then you should (1) use the angled connector and not the one from your datasheet, (2) mount this board in-line with Kasli, (3) use deeper crates, [...]

At least for the first version, I'd like to keep this mounted orthogonal to Kasli. If we make the board 4HP then that still leaves 4HP between the edge of the board and the next board to feed cables through, which should be plenty AFAICT.

jordens, 2018-02-06:
If you just want to break out the IDCs, then the other solution I proposed above (in-line with Kasli) is the easiest.

If you have a narrow horizontal backplane and any 160 mm long board next to Kasli, you'll still have a big problem threading the 12 ribbon cables through the gap. Even with the 8 HP wide Kasli, and especially if the backplane has 4 IDC connectors right where you cables are coming through.

I don't see how you are going to make this 4 HP wide.

* [ ] Ok. Let's put two barrel connectors with two of those EMC chokes onto this.

jordens, 2018-02-06:
What's the reason for keeping it orthogonal to Kasli?

marmeladapk, 2018-02-06:
@jordens @hartytp Greg proposed that IDC connectors, power jack and MMCX could be located on the other side than Kasli and BP connector and that the board would be perpendicular to Kasli.

hartytp, 2018-02-06:
> @jordens @hartytp Greg proposed that IDC connectors, power jack and MMCX could be located on the other side than Kasli and BP connector and that the board would be perpendicular to Kasli.

That would mean that users would need to add a blank panel next to Kasli, but I'd be okay with that.

hartytp, 2018-02-06:
> What's the reason for keeping it orthogonal to Kasli?

Forces one to use a longer rack, and the different mounting scheme which is (AFAICT from limited experience) going to be less convenient for us.

marmeladapk, 2018-02-06:
@hartytp I meant that the IDC connectors etc would be facing rear end of the crate.

jordens, 2018-02-06:
@marmeladapk you'd still have to thread 8 IDC cables in the space between the adjacent board and the right backplane edge. If you show me that that is fine in a rendering, I am happy. But from what I can see here it's not going to fly. Also, why go through the trouble of mounting the backplane orthogonal if you can just extend Kasli this way with an in-line backplane, 200 mm card guides (not even really needed), the good old angled barrel connector, an angled DIN connector and 4 IDCs in the same arrangement as on Kasli? What's the point?

@hartytp If you put the IDC connectors to the rear **or** if you want to use the space in the back for cabling, then 200mm deep card guides and the extension in-line with Kasli are just perfect for you? You seem to want to not use the space in the back and use it at the same time.

hartytp, 2018-02-06:
@jordens I'm not strongly against that just don't have any experience mounting boards in this way. Open to being convinced...

hartytp, 2018-02-06:
Fwiw thinking more about Kasli BPs in general:
- the  only way I see a Kasli  bp working out is to make all EEMs DRTIO slaves by adding a cheap artix 7 and Si5324 
- then connect then to Kasli via a tx and rx pair rather than all those DIO lines.
- have a passive   BP with fixed 4hp pitch and 1 tx  and 1 rx pair, clock and I2C to each solt.
- not worth the hassle imho but that's probably the way to go if one really wants to solve the rats nest...

jordens, 2018-02-06:
That would need much more detailed planning and a proper proposal. And there are a bunch of tricky and unresolved issues with it (development and debugging are much more difficult for one).
And that plan is not a reason to be lenient on the current design.

Finding right angle female 96 pin connectors looks more tricky than I expected. Well, make it horizontal and short and have the IDC and power connectors go out to the back. With short DIO modules mounted next to Kasli that should be OK. But you still have to resolve the problem in the case of your high density crate.

marmeladapk, 2018-02-06:
@jordens [HARTING 09 73 296 6801](http://datasheet.octopart.com/09-73-296-6801-HARTING-Elektronik-datasheet-8538369.pdf)

hartytp, 2018-02-06:
@jordens ack to all that. Not suggesting any changes just thinking out loud

hartytp, 2018-02-07:
- Okay, let's have the BP adapter as an extension of Kasli (parallel with it) with the IDCs on the same side of the PCB as the IDCs on Kasli.

- Add a couple of RA barrel connectors for power in/out.

- Let's make sure it fits inside a standard 220mm guide rail and that it fits inside a standard 255.5mm deep Schroff rack like 24572-670 (even with the back plate installed).

marmeladapk, 2018-02-07:
> Let's make sure it fits inside a standard 220mm guide rail and that it fits inside a standard 255.5mm deep Schroff rack like 24572-670 (even with the back plate installed).

This is going to be a problem.

[image]

Kasli is 162,3 mm long + 6,5 mm between boards + 63,4mm of BP adapter = 232,2 mm

12 mm shorter board wouldn't have space for J7. I could stack IDC connectors in two rows:

[image]

jordens, 2018-02-07:
Stacking is fine IMHO. Those need long cables anyway.

hartytp, 2018-02-07:
@marmeladapk The smaller board with the stacked connectors looks good to me. Let's go for that layout.

IMHO 4 barrel connectors is overkill, so I'd leave the space open for future applications (e.g. adding a clock buffer and some more MMCX connectors).

marmeladapk, 2018-02-07:
> IMHO 4 barrel connectors is overkill

@hartytp So 2 or 3 connectors?

hartytp, 2018-02-07:
Either is fine with me.

I don't think they'll be used much in practice, since not many boards have power connectors on their rear. And, anyway, by the time you need that many connectors, you're probably running out of power anyway.

jordens, 2018-02-07:
2 is fine for me.
Also we should make it the right length and add rear panel mounting holes in the PCB so that it can be mounted into the rear top and bottom rails. I couldn't verify the correct PCB length for that but it should be 60 mm minus connector gap.
Are the angled female connectors standard and aeasy to get by?

hartytp, 2018-02-07:
> Are the angled female connectors standard and aeasy to get by?

In stock from Farnell and Mouser, so seem fine. 

http://uk.farnell.com/harting/0973-296-6801/socket-din41612-r-a-r-96way/dp/1096942

marmeladapk, 2018-02-28:
[Kasli_BP_Adapter.PDF](https://github.com/m-labs/sinara/files/1766973/Kasli_BP_Adapter.PDF)

marmeladapk, 2018-03-28:
[image]

2 adapters arrived. I'll test them today or tommorow. There is right angled mmcx connector instead of straight one.

hartytp, 2018-03-28:
Looks nice!

marmeladapk, 2018-03-29:
I checked connectivity on it, everything works fine.

hartytp, 2018-03-29:
Ace! Good job.

marmeladapk, 2018-03-29:
Apart from mmcx connector (which was Technosystem's mistake) there are no issues with it, so I'll close the issue.

closed 2018-03-29 by marmeladapk

hartytp, 2018-03-29:
Cool.

Do you have one you can send me so I can test it out as well?

[3 more replies]