sinara-hw/sinara #352: Kasli/EEM softstart

hartytp, 2017-11-06:
@gkasprow In our lab, we'd like to use Kasli for a 16-channel AOM controller. Needs 4xUrukul + 2xNovo (takes up all 12 EEM connectors on Kasli, including the 4 on the BP).

Looking at the power budget, this should take up less than the 5A available from the 12V rail. However, we will be close! (Which is one reason to try to push Urukul and Kasli power consumption down in the next revision if we can). In the past, I've found that running things close to the supply's current limit can cause issues during startup, since the inrush current causes the supply to go into current limit mode, which can cause other components to lock up.

To avoid this, we should try to make sure that soft-starts are used on Kasli/EEMs to minimize the current surge when the supply is connected.

gkasprow, 2017-11-06:
@hartytp 
DDS(Urukul) has soft start
Novo, Zotino,BNC/SMA DIO don't have one mainly due to DC/DC converters but the startup current is limited by overcurrent protection of the conveters.
IMHO the easiest way is to implement soft start circuits on Kasli/VHDCI carrier board due to the number of rev1 boards already produced.

gkasprow, 2017-11-06:
The simplest circuit is PMOS + RC + D circuit

hartytp, 2017-11-07:
> DDS(Urukul) has soft start

That's good, since this is the EEM which will draw the most power. The other one I'm a bit worried about is Zotino.

> The simplest circuit is PMOS + RC + D circuit

However you think is best. I guess this will have to wait for Kasli v1.1 though...

hartytp, 2018-01-11:
@jordens @gkasprow Do we still want this feature for Kasli v2.0? My feeling is that it's an unnecessary complication that we should avoid.

High powered EEMs like Urukul should have a soft-start on them (Urukul has one). Also, most EEMs don't draw much current at startup anyway. e.g. clocks generally aren't supplied to DDSs until the FPGA boots up, DACs output 0 voltage until programmed so there is no load current.

Closing this issue for now. I'll have a look at how a fully loaded Kasli (16-channel Sampler-Kasli-Urukul servo) behaves and if there are any problems then I'll reopen it.

closed 2018-01-11 by hartytp

jordens, 2018-01-12:
Ack. Let's test this first. A scope trace of the current draw on 12 v on urukul would be nice. With 100 MHz coming into the sma.