sinara-hw/sinara #353: Kasli testing

hartytp, 2017-11-07:
Tests that need to be done on Kasli v1.0:

- [ ] Measure phase noise of clock outputs. This should be done both for the CDR clock source and using a clean 100MHz source via the SMA input.

@WeiDaZhang can you do the phase noise measurement once we have the boards set up in Oxford, please?

jordens, 2018-02-01:
Noise between two Kaslis (J2), running freely from their 114MHz XOs. Instrument floor is at -131 dBc/Hz instrument 1/f corner at ~100 Hz. The 450 kHz spurs are real. The 2 MHz spur, the roll-off above 1 MHz, and the 600 Hz spurs (AFAICT) are artefacts. Including the 3dB from the differential measurement and 3dB from the datasheet (maybe) being two-sided noise, this fits.

[image]

hartytp, 2018-02-01:
cf fig 6 from the [data sheet](https://www.silabs.com/documents/public/data-sheets/Si5324.pdf).

- I very much doubt that the data sheet uses two-sided noise -- no one ever does that, it's always SSB.
- The 3dB you mention is more or less canceled out by the 17dB to account for the carrier frequency difference!
- So, to the level of the measurement precision, both plots should be the same.

The data you have there looks to be about 10dB worse than I'd expect from the datasheet...

hartytp, 2018-02-01:
Still, not too bad.

jordens, 2018-02-01:
@hartytp 
* You'd be surprised, how many engineers do that. E.g. all of IEEE 1139-1999: "2.8 phase noise L(f): One-half of the phase instability Sφ(f), as defined in 1.7."
* The plot is at 148 MHz. There is no carrier difference to 150 MHz. Where do you get 17 dB from?

hartytp, 2018-02-01:
> The plot is at 148 MHz. There is no carrier difference to 150 MHz. Where do you get 17 dB from?

17dB was a typo (fixed in an edit before you commented), should have been 1.7dB. I misread your post, though that you were using a 114HMz output from the Si5324.

jordens, 2018-02-01:
And more specifically the E5052B they [used](https://www.silabs.com/documents/public/application-notes/AN739.pdf) in the datasheet shows phase noise as L(f) = Sφ(f)/2.

hartytp, 2018-02-01:
Okay, miscommunication: that definition of L(f) is the only thing anyone ever means by SSB phase noise in RF engineering (and, yes, IEEE 1139-1999 is the go to reference for this) which is used in basically every datasheet I've ever read (at least ones written in the last decase) and used by all instruments I've ever seen. Importantly, for white noise it means that the phase noise is 1/2 of the single-sideband NSD. 

I hadn't noticed that your axis uses Sphi. I didn't check because that's not something I ever expect to see in this context...

hartytp, 2018-02-01:
Are you measuring this with a mixer and scope or something? How are you calibrating that?

jordens, 2018-02-01:
NIST T&F uses Sφ(f) with **strong** preference. Maybe that tainted me. This is measured with a simple dual channel ADC directly. Free of any calibration.

hartytp, 2018-02-01:
Meteorologists use that, but no engineers do. In this context it's best to stick with the engineering definition so that the plots actually match the datasheet. When in Rome...

FWIW, for this kind of work, I like the fact that SSB PM + AM is the number that my spectrum analyzer gives.

hartytp, 2018-02-01:
Anyway, with that 6dB cal factor accounted for, and ignoring all spurs as well as everything below 100Hz/above 1MHz, it looks like you're within about 4dB of the datasheet noise...

jordens, 2018-02-01:
The stuff below 100 Hz is the two different XOs wobbling. The value at 100 kHz offset converted to 148 MHz and subtracting the instrument floor is L(f)=-131.5 dBc/Hz. The datasheet has about -134+-1 dBc/Hz. That's within 2.5+-1 dB.

hartytp, 2018-02-01:
Well, anyway, that's a good indicator that the Si5324 will work well as a reference source for Sayma. The real fun will be when someone measures this for a DRTIO slave getting its clock via a fibre.

jordens, 2018-02-01:
That would have been nice to do back in August 2016 with mini-DRTIO.

hartytp, 2018-02-01:
To put that number in perspective, the AD9154 phase noise is about -140dBc/Hz at 100kHz offset for a 200MHz carrier. Call it -143dBc/Hz after rescaling to 150MHz. So, the results you've measured with the Si5324 (even without an LDO or proper filtering of the SMPS) are only 9dB worse than the data sheet values for the DAC.

That backs up the conclusion that (pending measurements of close-in phase noise once this is locked up) the improvements available from using a fancy LLRFBP over DRTIO clock are not going to be huge...

jordens, 2018-02-01:
With the 10-100 Hz PLL bandwidth, at 100 kHz you are only looking at the Si5324 VCO. That should look the same even with the crappiest reference.

hartytp, 2018-02-01:
> With the 10-100 Hz PLL bandwidth, at 100 kHz you are only looking at the Si5324 VCO (or XO; I don't know how that works exactly). That should look the same even with the crappiest reference.

IIRC, the Si5324 is phase locked to the external XO on the PCB, which is phase locked to the CDR reference.

Above 1kHz, I expect this to be limited by the XO/SI5324 so it should be what the data sheet gives (as you've seen). Below that, it will follow the clock from the FPGA. I'll be curious to see what that part of the noise spectrum below 1kHz looks like.

jordens, 2018-02-01:
Yes. Once @enjoy-digital and @sbourdeauducq have the DRTIO link and CDR up we can do that with Kasli. It was done with KC705s, @jbqubit has the data, but that necessarily also included Kintex routing noise.

jbqubit, 2018-02-02:
@hartytp said
> The real fun will be when someone measures this for a DRTIO slave getting its clock via a fibre.

@jordens said
> That would have been nice to do back in August 2016 with mini-DRTIO.

Here's what I measured in August 2016 for mini-DRTIO that @sbourdeauducq and @enjoy-digital put together. This was for a fiber link between a pair of boards using DRTIO clock and data recover recovery. 

---------------

# phase noise test results 

__Aug 9 2016__
Requisition W911QX-16-P-0013. Conduct tests to confirm performance. 
- source code and instructions are at the following URL https://github.com/m-labs/drtio_transceiver_test
- Phase noise measurement tool is Symmetricom 5125A. 
- All components powered on for 2 hours prior to start of measurements.
- Using pair of KC705 boards. One is designated Tx and the other is designated Rx. 
- Gateware is github.com/m-labs/drtio_transceiver_test/artiq_ttl_rx and github.com/m-labs/drtio_transceiver_test/misoc_artiqttltx_kc705 (committed on Aug 2, 9c0e).
- Plots using TimeLab http://www.ke5fx.com/timelab/readme.htm

[image]

(above) check noise floor; 62.5 MHz sq-wave to RF power splitter to Ref and Input on Symmetricom; Symmetricom noise floor is in shaded blue [20160809_pn_4.tim]

[image]

(above) DRTIO Raw Recovered Ref = kc705_tx  usr_clk_nInput = kc705_rx usr_clk_n/p
(both at +15 dBm); lock maintained for 10 hours, test terminated by user 
- RMS jitter 1 Hz to 1000 kHz is 50 ps
- RMS jitter 1 Hz to 100 kHz is 18 ps

Comment from @sbourdeauducq about how the 62.5 MHz clock is generated. 

> It's generated from the 125MHz "SGMII" oscillator on the KC705 and then processed by the transceiver (CPLL, dividers, phase aligner).

Looking KC705 docs: "An Integrated Circuit Systems ICS844021I chip (U2) generates a high-quality, low-jitter, 125 MHz LVDS clock from a 25 MHz crystal (X3). "

[image]
(above) plot from ICS844021I data sheet

# data recovery

This design also included a deterministic 64 bit pseudo-random data generator. @sbourdeauducq demonstrated sustained data transmission at 1 Gbps. No errors were detected over a 10 hour test. 2.7e14 bits

hartytp, 2018-02-02:
Not sure I understand what's plotted there. Anyway, DRTIO should be up on Kasli soon, so let's take a new measurement on the HW we plan to use.

klickverbot, 2018-02-02:
@hartytp:
> Edit: no, that can't be right, the XO is not tunable. Actually, I can't remember how these chips work :(

IIRC, they use a nested PLL architecture, where the on-chip VCO (somewhere in the multiple GHz) is locked to the external reference using a standard frac-N PLL, and that oscillator is then used in an all digital PLL driven by the clock input. I'm pretty sure the bandwidth of the inner PLL is somewhere in the MHz, but I can't seem to find the reference for that right now.

klickverbot, 2018-02-02:
Found it: https://www.silabs.com/documents/public/white-papers/Silicon-Labs-Next-Generation-DSPLL-Technology-White-Paper---June-2015.pdf

jordens, 2018-02-03:
@klickverbot nice! Thanks for digging that up. That's an interesting architecture. I like it, good that we copied it over from the kc705. I wonder what drove the whiterabbit people to chose there CDR architecture and jitter attenuation circuitry over this.

jordens, 2018-02-03:
That also means that the xo needs a pretty stable environment down to the lock bandwidth. I hope the SFP proximity doesn't cause thermal fluctuation problems there. I tried to disturb it but it seemed to cope.

jordens, 2018-02-03:
And we need to determine the optimal loop bandwidth for typical DRTIO CDR.

hartytp, 2018-02-03:
> That also means that the xo needs a pretty stable environment down to the lock bandwidth. I hope the SFP proximity doesn't cause thermal fluctuation problems there. I tried to disturb it but it seemed to cope.

Also means that if we have noise issues we can potentially win by choosing a higher quality XO to allow us to reduce the lock BW. IIRC, the one there is pretty cheapo.

jordens, 2018-02-03:
@hartytp yes. Depending on the parameters that by itself could improve the noise from ~10 Hz=outer BW to ~1 MHz=inner BW. But I don't know what the limit is.

Note to self: that "DSPLL" is a frac-N PLL that locks a "prestabilized" VCO to the reference with many-Hz bandwidth. That "compound VCO" is itself is a many-GHz VCO frac-N-PLL-locked to a fixed XO with MHz bandwidth. It's tuned digitally through the inner frac-N divider.

hartytp, 2018-02-03:
> @hartytp yes. Depending on the parameters that by itself could improve the noise from ~10 Hz=outer BW to ~1 MHz=inner BW. But I don't know what the limit is.

AFAICT, the data sheet/app notes don't give enough info to construct a decent noise model for the Si5324, so we'd need to measure this by adding a good XO (should be easy to dead bug on). I'd probably go for one of the really good Crystek XOs for this measurement.

@marmeladapk is that something you could have a look at? Would it be interesting for your thesis? It would be really interesting to see how low we can make the noise of the Si5324 by locking it to a good XO and optimising the lock parameters. This would be particularly useful data for Sayma where the CDR clock is likely to be our main reference source, so we should take care to make it nice and clean.

The second really useful piece of data would be a differential phase noise measurement between a reference oscillator and the input clock to the Si5324 on a Kasli DRTIO slave (i.e. the clock output from the FPGA). With these two pieces of data we can predict and optimise the overall noise of the DRTIO clock.

jordens, 2018-02-03:
For DRTIO with Sayma the plan was start with using the local clock (HMC830, from SMA or LLRFBP, or directly from the SMPs/clock mezzanine), not the one from DRTIO. But yes, it is interesting for DRTIO with CDR.

hartytp, 2018-02-03:
@jordens yes, definitely for the near future that's what we'll do. I more meant "what I'd like to see happen in the long run". Anyway, it would be nice to have some more data on this, so we know how good the DRTIO clock will be.

hartytp, 2018-05-03:
Phase noise measurements have been taken and posted elsewhere. Nothing else to do on this issue.

closed 2018-05-03 by hartytp