sinara-hw/sinara #354: Urukul v1.0 testing

hartytp, 2017-11-07:
- [ ] Channel-to-channel cross-talk: both with switches opened and switches closed; also with attenuators at max/min. 
- [ ] Output phase noise/amplitude noise

Use some fixed reference frequency for all tests, such as 100MHz.

hartytp, 2017-11-07:
@gkasprow Can you ship us all but 1 of the AD9910 Urukul prototypes you've ordered as soon as basic functionality/"no smoke" testing is complete (i.e. don't complete the above detailed testing first!). 

Thanks!

hartytp, 2017-11-07:
@gkasprow @jordens Feel free to add more tests to the above list.

hartytp, 2017-11-08:
@gkasprow When you ship the AD9910 DDSs that we paid for, please can you ship one to @jordens?

jordens, 2017-11-08:
Updated it a bit. Please ship right after smoke test:

- [ ] power supplies good
- [ ] dds pll locks
- [ ] dds outputs frequency
- [ ] attenuators can be set
- [ ] rf switches work

We'll be happy to help with characterization.

gkasprow, 2017-11-12:
@hartytp OK, 2 Zotinos + BNC adapters were shipped on Friday. Once I receive DDS, I will test them ASAP.

gkasprow, 2017-11-12:
@hartytp There are no panels for DAC BNC adapters attached - I forgotten about them. Will ship them next time with some other stuff.

hartytp, 2017-11-12:
@gkasprow No problem. That can wait.

If there is any time before Urukul arrives, it'd be great to get a few of the analog test for Zotino done.

gkasprow, 2017-11-12:
@hartytp Sure, once I repair my bard, will continue with measurements.

hartytp, 2017-11-12:
Great, thanks @gkasprow!

jordens, 2017-11-19:
@hartytp could you remember to do a bit of triage on the issues you file (i.e. assignee, labels, milestone)?

hartytp, 2017-11-19:
sure

hartytp, 2017-11-20:
@gkasprow For the current source biasing, my inclination is to go for Daniel's suggestion of the LT3092. They're pretty cheap ($2 in quantity) and seem like a better/easier solution than building our own current source.

@jordens @gkasprow if you're oaky with this approach, then we should buy an eval board at some point and test this out with the v1.0 boards we have. I'm happy to do this testing if that helps -- I probably won't do this until after Christmas, but I don't think we're in a rush for it.

jordens, 2017-12-11:
@a-shafir and I will take care of the first round of noise, power, RF switch testing.

hartytp, 2017-12-11:
Cool.

Out of curiosity, do you have a proper phase noise/amplitude noise meter? Or, will you just stick this on a spectrum analyser?

gkasprow, 2017-12-11:
@jordens I have them all in my lab but was busy with Sayma Ethernet debugging. You can follow my fight here #854

gkasprow, 2017-12-11:
So the next thing is 9912 testing.

jordens, 2017-12-11:
I am following that and involved.
Thanks for the AD9912 update. Once you have smoke-tested them like the AD9910 stuff, you can send two of them to me.

a-shafir, 2017-12-11:
@gkasprow what was actually flashed into the CPLD on the 9910 board we have in Berlin?

I see 2 options
https://github.com/m-labs/sinara/tree/master/ARTIQ_ALTIUM/EEMs/Urukul/CPLD_code
or
https://github.com/quartiq/urukul

I am testing h/w but can't generate any output on the IO_SESET (P129) and some pins around.

a-shafir, 2017-12-11:
@gkasprow how you tested 3U DDS 9910?
Have you got DDS ic working?
Do you have the code that sent the spi commands, what actually sent to DDS/atten etc?

gkasprow, 2017-12-11:
@a-shafir I burned Roberts Verilog code. The only change I did was assignment of test points. Now we have:
```
assign tp = (cfg_data_rst | ((~en_9910) & tstriple7_i));
assign tp_1 = dds_cs_n;
assign tp_2 = dds_sck;
assign tp_3 = dds_sdi;
assign tp_4 = dds_io_update;

```

gkasprow, 2017-12-11:
I connected 9910 devkit via DIO-Tester module to make sure that Robert's code works. In this way I could communicate with it using LVTTL signals

gkasprow, 2017-12-11:
I also modified the verilog code so all 4 channels were active simultaneously. In this way I tested if PLLs are locked. Here is the code:
[Urukul_CPLD_all_ch.zip](https://github.com/m-labs/sinara/files/1549674/Urukul_CPLD_all_ch.zip)

gkasprow, 2017-12-11:
To do general purpose communication over SPI I use little Artix 7 devkit with USB and FORTH processor that translates UART commands to bit-bang protocol, i.e. SPI. It works really fast but it's written in FORTH, so probably not really useful for you :)

a-shafir, 2017-12-11:
@gkasprow i see io_reset commented in the code. probably some other changes.
Please also attach the pin assignment for the code.
So what was generating the DDS SPI command? 9910 devkit? I have no devkit.
Can you generate a dump (sequence) of the commands? 
Also what about the reading from DDS when all in parallel?

a-shafir, 2017-12-11:
@gkasprow i see why in_sel not working to - it commented out
Good that i got this data from you )

gkasprow, 2017-12-11:
The original DDS was working in parallel. So I didn't read the status from externally connected ones.
You can compare this code with one @jordens posted. I did not touch the pins assignment.
My CPLD project is [here](https://www.dropbox.com/sh/qj16dhavo4qxsw9/AAB5paEl0olZpSaRMjhjxSyOa?dl=0)

gkasprow, 2017-12-11:
This is how typical setup wit hdevkit looks like. I this case I connected Sampler module, but the idea is same.
[image]

gkasprow, 2017-12-11:
this is the FORTH code I use to talk to LV595 registers.
The FORTH ascii is executed in FPGA and it toggling IOs routed to IO register with a few MHz rate. CPU runs with 50MHz clock.
The function below writes 16 bits via SPI to the register.

: spi_LV595 ( a ) 
	
	1 OUT0_SET_REG io! \ SCK hi
	1 16 lshift OUT0_SET_REG io! \ OE buff active
	1 2 lshift OUT0_CLR_REG io! \ RCLK low
	
    16 0 do
        dup
        1 15 lshift and
        if
           1 1 lshift OUT0_SET_REG io!
        else
           1 1 lshift OUT0_CLR_REG io!
        then
		1 OUT0_CLR_REG io! \ SCK low
		2*
		INP0_REG io@ 4 rshift 1 and +
        1 OUT0_SET_REG io! \ SCK hi	
    loop
	
	 1 2 lshift OUT0_SET_REG io! \ RCLK high
	 1 2 lshift OUT0_CLR_REG io! \ RCLK low
	 1 16 lshift OUT0_CLR_REG io! \ OE buff inactive
	 $ffffff and  \ return only valid bits
;

gkasprow, 2017-12-11:
@a-shafir you are right - I had to modify the reset pin state to make it working.

gkasprow, 2017-12-11:
And the code you have burned is in Urukul_CPLD_1ch.v file

gkasprow, 2017-12-12:
@jordens here is your 9912 :)
[image]

jordens, 2017-12-12:
@gkasprow Nice! Other than the missing amplifier, are there any other issues? Is the solder on the underside bad as well?

hartytp, 2017-12-12:
Nice!

gkasprow, 2017-12-12:
I got this amp separately. The PN is wrong so they got ones with straight leads and had to replace them.
It seems that the ICs they got had problems with wetting because other chips are fine. Anyway, it's their responsibility to solder it right

jordens, 2017-12-12:
Are you returning the AD9912 boards to Technosystem to be fixed as well?

gkasprow, 2017-12-12:
I will fix one piece myself to test it. Rest boards will be returned. I talked with the company and they say that it is quite standard thing that some chips have poor wetting and side contacts are not always covered by the tin. The flux that is part of the soldering paste is only roughly 10..15% of it and is not always able to clean the contact side. The joint still maintains the contact because it is low side of it which does the job. Side area does not need to be covered, but if we want, they can fix it.

jordens, 2017-12-13:
IMHO it is likely that the joints are bad on the underside as well if they fail to wet that dramatically on the edge. And since they probably didn't inspect those joints there is no telling. I'd like to see them fix it.

gkasprow, 2017-12-13:
I just desoldered the chip and the bottom pads are correctly soldered. Anyway it just looks non professional and should be fixed.

a-shafir, 2017-12-13:
This is a long know for the technologists that too much solder is not helping and even can cover some problem. But in this case it is not about the professionalism but about Optical inspection standards. All the packages designed that way that there is possibility to ensure by inspection the proper electrically and mechanically soldering. For this package there must be wetting on the side edges of the pads. This indicates that there is wetting under the pads and also improves mechanical properties. 

For some of the packages we have (BGA and some of the qfn with signal pads under) the OIS will not enough. I think need to check if the factory can do x-ray inspection, they must have it in or out house at least for BGA but also for some regulator chips etc. Visual inspection of the QFN etc is the must. It shall not increase the cost of the samples production more than few %

Any way investing our time to solving the technological issues is the worst case scenario.

a-shafir, 2017-12-13:
Urukul 9910, 4th channel, 10.000MHz, variable attenuation test, no termination:
[image]

gkasprow, 2017-12-13:
@a-shafir look [here](https://github.com/m-labs/sinara/issues/377#issuecomment-351379129)

a-shafir, 2017-12-13:
@gkasprow ok, this is a news for me too. There are 3 similar looking "QFN-style" packages:
- one have sides plated to be good for wetting (some crystals for instance and old stile QFN)
- some have a etch/step around to have the wetting visible for OIS
- the newest are basicly BGA-like i.e. all wetting is under the package but just no balls.

That minds for us:
- Even we don't have a BGA on a board we need to inspect the packages are they are good for optical inspection (OIS) and if any one of them is not need to add X-ray inspection into the production list.
- Need to chose the parts for the boards more carreful. If the board normally not needs X-ray try to avoid any packages that have no OIS compatible pads (like LTM4622 etc) 
- It is good to put this as a part of the h/w and production specs checklist.

a-shafir, 2017-12-18:
Gateware, core drivers and experiment for Urukul AD9910 rev 1.0 board.

The current setup is for KC705, see the details here:
https://github.com/m-labs/artiq/commit/dd3c8a336dc12cda33cadebdf3bafc77c0bc1565

The experiment initializes FMC DIO, the Urukul board and configures the single tone for all 4 channels.
Attenuator and SW are supported.

There are 2 "interference generator loops" (for DDS3 SPI and for ATT SPI). Each can be activated in the experiment. Note: with the current RTIO gateware it is not possible to generate the NU-mode intensive SPI communication so if any interference from the SPI clock will be seen on the output it will be not the worst case scenario.

The 4-channels DDS output:
[image]

The setup:
[image]

jordens, 2017-12-18:
* RF Switch rise time is roughly 100ns to 95%
* Spectrum looks ok. A few weird close in Spurs (+- 1.2 kHz and 11kHz at -70dBc) potentially related to grounding to be debugged. They are big when the board is on the antistatic bag on an optical table and otherwise disappear.

More data and screenshots to follow.

dhslichter, 2017-12-18:
@a-shafir why are you using scope probes to look at the DDS outputs instead of SMA cables?

gkasprow, 2017-12-18:
@jordens in such setup you easily receive and induce current from every short wave radio station including CB :)

jordens, 2017-12-18:
@gkasprow in general yes, but not if you are in a room that is as insanely shielded as the one we are doing the tests in ;)

a-shafir, 2017-12-19:
@dhslichter there was only one more SMA cable in the lab and no 50R terminators at all. Any way the probes are OK to check how the driver controls.

jordens, 2017-12-19:
The screenshots are crappy and there is some analysis still to be done. I also have to parse my notes.
tl;dr: All good. The only issue are phase noise spurs at -75dBc at ~1.2 kHz offset with many harmonics, especially x11. We didn't figure out where exactly they came from. The spur offset is surprisingly low. Their offset is temperautre dependent (increases with temperature). And coupling different parts of the PCB (by finger), especially around the underside near the clock fanout, makes them worse. That spur needs to be debugged.

##  Attenuator step from 20 to 60 digital (32, 8, switch glitch)
[image]
[image]

## Attenuator step from 31 to 32 digital (major carry glitch)
[image]

## Harmonics DC to 1 GHz, clock and ref leakage below RBW
[image]

## Harmonics 2 and 3 vs power
[image]
[image]

## Close in spurs (1.2 kHz multiples, especially 11x)
[image]

## PLL bump
[image]

## Phase noise 100 mHz to 10MHz
[image]

## Phase noise 10 Hz to 10 MHz
[image]

## Phase noise 1 kHz to 1MHz
[image]

## RF switch turn on transient
[image]

## Nyquist rejection 400 MHz to 600 MHz
[image]

## Nyquist rejection 450 MHz to 550 MHz
[image]

gkasprow, 2017-12-19:
@jordens such spur may be caused by PSU instability. Then 1.2kHz ripples caused by SMPS or LDO feedback loop may cause such spurs.
So you can have a look with scope on critical power rails.

jordens, 2017-12-19:
Ack. We didn't find anything noticable on the rails. We'll do more prodding and poking.

gkasprow, 2017-12-19:
@jordens I can have a look with spectrum analyzer or when necessary with SSA. Is ADI devkit sufficient to recreate this spur?

jordens, 2017-12-19:
I don't know. But I'd be surprised if you see them on the devkit.
I'll give you details about the setup later.

gkasprow, 2017-12-19:
No, I mean Urukul controlled by ADI devkit instead of the Artiq

jordens, 2017-12-20:
I think you should see them on Urukul with yout setup. I added the description of the setup to the post above.

a-shafir, 2017-12-20:
@jordens what reference clock used for the test? KC705 output from the FPGA pll possible has the low freq modulation from the power or even fan. And the AD9910 pll most likely will unable to clean it.

jordens, 2017-12-20:
@a-shafir the reference is clean. And you hypothesis is already excluded be the other observations.

jordens, 2018-01-02:
AD9912 variant also works fine.
@gkasprow did you get a chance to hunt for the 1.2 kHz spurs?

gkasprow, 2018-01-02:
@jordens not yet. The University and my lab was closed during Xmass. I could not even cross the main gate. They open tomorrow.

jordens, 2018-01-02:
Ack. That's fine!

[39 more replies]

closed 2018-06-13 by marmeladapk