sinara-hw/sinara #355: Clocker v1.0 testing

hartytp, 2017-11-08:
- [ ] Input return loss
- [ ] Edge rise/fall time (NB this needs a very fast scope, not worth doing if the scope isn't as fast as the ADCLK948!)
- [ ] Phase noise/amplitude noise
- [ ] Output power versus frequency up to > 4GHz

Use 100MHz reference frequency for all tests, input should be ~+10dBm.

hartytp, 2017-11-08:
@gkasprow My suggestion is that you do basic functionality/no-smoke testing and then ship the boards to us for detailed testing. @WeiDaZhang can you perform the tests above once the boards arrive, please? (If you like, you can also look at the phase stability, but that's not crucial IMO).

gkasprow, 2017-11-08:
@hartytp OK, this board was meant as a quick design job, with no intensive testing.

hartytp, 2017-11-08:
> @hartytp OK, this board was meant as a quick design job, with no intensive testing.

Yes. We'll take care of any testing that needs to be done.

closed 2018-03-06 by hartytp