sinara-hw/sinara #357: panel design for Kasli, Urukul and clocker

gkasprow, 2017-11-09:
@hartytp @jordens please review [panel](https://github.com/m-labs/sinara/tree/master/ARTIQ_ALTIUM/PANELS/Project%20Outputs%20for%20PANELS) designs for the new boards.
I want to launch production ASAP.

hartytp, 2017-11-09:
@gkasprow Look good to me.

- I'd say "Kasli EEM carrier" to be consistent with the VHDCI carrier (or even just "Kasli")
- Is there a space missing in "Console& JTAG"?

jordens, 2017-11-09:
### Urukul
* The SMA connector cutouts are too narrow on the current DIO_SMA. Are they wide enough on here?
* I'd go through the DIO_SMA errata and check the other mechanical erratas.
### Clocker
* The SMA connector cutouts are too narrow on the current DIO_SMA. Are they wide enough on here?
* I'd go through the DIO_SMA errata and check the other mechanical erratas.
### Kasli
* That should be: "JTAG, UART, I2C" (those are the three important USB interfaces on that connector).
* Please don't call it "EEM carrier". It doesn't carry the EEMs (the crate does, or my English is wrong). "Kasli EEM controller" is good IMHO.
* Are you sure the USB connector including the PCB tounge fit into the FP cutout? That PCB tab with the USB receptable is much wider than the USB receptable alone, i.e. I'd expect the cutout to be much wider as well.
* And are you sure a USB connector can be inserted into the connector? Those micro-USB receptables need to be almost flush with the panel AFAIK.

hartytp, 2017-11-09:
> Please don't call it "EEM carrier". It doesn't carry the EEMs (the crate does, or my English is wrong). "Kasli EEM controller" is good IMHO.

Okay, but then we should change the VHDCI carrier to match...

gkasprow, 2017-11-09:
@jordens The SMA openings were increased to fit washer so should be fine.
USB is installed on PCB tab that touches the panel. USB connector is already inside of the panel. We use such approach in another board and one can insert USB port easily. Big cutout to fit plastic part of the plug would not be aesthetic. When necessary we will make some cavity from front side to fit the plug but IMHO this won't be necessary
We used  RA SMAs to get rid of issues found with SMA DIO.

gkasprow, 2017-11-09:
@hartytp EXT CLK on clocker means that LED indicates external clock selection. @jordens pls check it now.

jordens, 2017-11-09:
Ok. If you say that the sma are all wide enough than this looks good to me.
I would however like to transition to calling the boards 'v1.0' instead of 'rev1.0'. shorter, compliant with semantic versioning, same as the git tags, agrees with the consensus of the discussion of those version names.

gkasprow, 2017-11-09:
OK, done

marmeladapk, 2017-11-09:
@jordens Should I also change it on Kasli silkscreen?

jordens, 2017-11-10:
@marmeladapk if that doesn't delay production, yes. Thanks!

closed 2017-11-10 by gkasprow

hartytp, 2017-11-10:
> Please don't call it "EEM carrier". It doesn't carry the EEMs (the crate does, or my English is wrong). "Kasli EEM controller" is good IMHO.

Thinking about this more, the reason I don't like "Kasli EEM controller" is the old don't prescribe usage argument. I can think of plenty of uses for Kasli beyond the EEMs. e.g. controlling Sayma.

This is why I just prefer Kasli or Kasli controller or something like that. Anyway, this is obviously not that important...

jordens, 2017-11-10:
ACK