sinara-hw/sinara #360: Kasli thermal simulations

marmeladapk, 2017-11-14:
I did some thermal simulations of Kasli board.
Environment (same as in Urukul tests, only board orientation is different):
[image]
No cooling:
[image]
50 cm/s airflow:
[image]
130 cm/s airflow:
[image]

hartytp, 2017-11-14:
cf #321

hartytp, 2017-11-14:
So, max temperature for Kasli looks pretty similar to Urukul for a given air flow...

gkasprow, 2017-11-14:
This is worst case scenario with all power rails fully loaded.

jordens, 2017-11-15:
Lgtm

closed 2017-11-15 by jordens

marmeladapk, 2017-11-15:
Do you want to see power supply simulations I did while optimising the board?

jordens, 2017-11-15:
Sure! This is all very valuable information. I'd guess also for your thesis.

marmeladapk, 2017-12-06:
P1V0:
[image]
[image]

P1V2:
[image]
[image]

P1V5:
[image]
[image]

P1V8:
[image]
[image]

P2V5:
[image]
[image]

P3V3:
[image]
[image]
[image]

P12V0:
[image]
[image]