sinara-hw/sinara #362: Novo v2.0 schematic review

hartytp, 2017-11-16:
@gkasprow Schematic looks good to me! Only minor comments:

- [ ] On the current PDF there are quite a few bits of overlapping text, or text that's not located inside its box.
- [ ] Signals on J1 need to be zero-indexed
- [ ] ADC schematic symbol doesn't have a part number on it
- [ ] Can we swap the connections to ADC_IN+ and ADC_IN- so that the polarity on the ADC is correct? (This can be done in software, but we might as well do it in hardware)
- [ ] Power sequencing on ADC +5V supply. This is probably worth doing properly to avoid obscure sequencing bugs, which can damage the ADC. What about using +-12V regulators with enable inputs, and using a +5V regulator with a PG output? 5V_PG then drives +-12V ENABLE lines. This is more robust than relying on diodes.
- [ ] For consistency C17, C23 and C25 should use 2n2 notation, not 2.2n
- [ ] J13 should be EEM0, J2 should be EEM1 nets should be named correspondingly (EEM0_0P/EEM0_0N etc)

gkasprow, 2017-11-16:
@hartytp In case of LVDS repeaters we can use other ones which are 3x less expensive but they need 3.3V. And that's why I had to add 3V supply for the logics to comply with preamp requirements. Anyway we would need separate supplies for logic and ADC VDDIO.
With power sequencing it's true. But still it won't protect the ADC against too high input signal. I'd add Schottky diodes at the ADC input. This would protect the ADC against negative signal but not against positive one because 5V supply can only source the current, not sink.
For all projects I use CERN library as it is. For some reason they use different nomenclature for resistors and capacitors. This is database library and don't want to mess up with it because with next release my changes would be overwritten. So let's leave it as it is.

gkasprow, 2017-11-16:
@hartytp with FIN1101 LVDS repeaters and 3.3V I got lower power consumption than with previous ones supplied from 2.5V.

hartytp, 2017-11-16:
> @hartytp In case of LVDS repeaters we can use other ones which are 3x less expensive but they need 3.3V. And that's why I had to add 3V supply for the logics to comply with preamp requirements. Anyway we would need separate supplies for logic and ADC VDDIO.

Use whichever ones you think are best.

hartytp, 2017-11-16:
> For all projects I use CERN library as it is. For some reason they use different nomenclature for resistors and capacitors. This is database library and don't want to mess up with it because with next release my changes would be overwritten. So let's leave it as it is.

Fine.

hartytp, 2017-11-16:
> @hartytp with FIN1101 LVDS repeaters and 3.3V I got lower power consumption than with previous ones supplied from 2.5V.

Did you have to add the 3V3 SMPS just for these converters? If so, does the SMPS cost more than the money you save by using cheaper converters?

gkasprow, 2017-11-16:
No the SMPS is just to supply DAC. I can also use 6V to generate both 3.3 and 2.5V.

hartytp, 2017-11-16:
> No the SMPS is just to supply DAC. I can also use 6V to generate both 3.3 and 2.5V.

Okay, makes sense. That looks good then!

hartytp, 2017-11-16:
> With power sequencing it's true. But still it won't protect the ADC against too high input signal. I'd add Schottky diodes at the ADC input. This would protect the ADC against negative signal but not against positive one because 5V supply can only source the current, not sink.

I don't think the power sequencing is quite right yet:

- The SN65LVDS1DBVT driving the ADC CNV input can (and should) run from 2V5, not 3V3. The 3V3 rail will turn on before the 2V5 rail, which allows the CNV pin be driven beyond its supply rail at turn on, which can cause lockup. If you do that, you don't need the 220R resistor to prevent lockup (this resistor slows down the CNV rising edge, making it more noise sensitive).
- The PGIA digital inputs must not go high before the +12V supply turns on. Maybe use the SR OE pin for this?
- The reference needs to turn on before the +-12V supplies (but still after the P5V0 ADC supply) to avoid negative inputs to the ADC. If you do this then no extra diodes are needed AFAICS, since the dividers + reference prevent the ADC inputs from being driven below ground or above +5V.
- P3V3 should turn on after 5V to avoid driving the ADC inputs beyond their supply rails
- Why do we need the P3V3 supply to turn on before the +-12V supplies? i.e. what is IC15A doing?

Sigh, pain increases exponentially with number of supply rails...

hartytp, 2017-11-16:
Other than that, I'm happy with the schematic. Thanks again @gkasprow!

Since no one else seems to have any comments, you can move on to layout whenever you're ready.

hartytp, 2017-11-16:
Some of the power sequencing issues can be fixed easily with diodes rather than using shutdown inputs. e.g. a diode between the +12V supply and the 3V3 supply will ensure correct sequencing of those supplies. This might be easier/cheaper...

gkasprow, 2017-11-17:
@hartytp in case of ADC CNV I placed current limiting resistor,  but supplying it from 2.5V is better idea. Good catch.
Utilisation of SR_OEn is good idea, I implemented it.
The 12V TVS diodes are to protect against too high voltage delivered from the optional AFE connector (J1)
IC15 makes sure that +- 12V wakes up after both 3.3 and 5V are present. To enable both negative and positive LDOs I need digital signal since they have different thresholds. If we use i.e. 3.3V to enable them, the rising slope would trigger them with delays causing potential problems.
I forgotten completely that AFE depends on reference. Good catch.
I modified sequencin.
First, DC/DC + 5V LDO starts, then its PG enables 3.3 and 2.5V.
It PG enables reference.
5V reference is compared against 3.3 which is good because PG is high and enables +-12V
And this signal after ~30ms enables SR_OE

hartytp, 2017-11-17:
"Hell is other regulators"

hartytp, 2017-11-17:
SDR mode is on EEM1_2, should be on EEM0_2

hartytp, 2017-11-17:
3V3 is enabled by the 6V SMPS, not the 5V regulator. So, I assume you're relying on the fact that the 5V regulator should track the 6V regulator pretty well?

That seems okay since the load on the 5V rail is not too high. Is it worth putting pads for an rc delay on that just in case we want to give the 5V rail more time to start up (don't do this if you don't think it's needed)?

hartytp, 2017-11-17:
Why does IC15A compare a divided version of the reference to the 3V3? Wouldn't it be better to compare the reference to a divided version of the +12V?

hartytp, 2017-11-17:
Other than that, this all looks good to me! That should be very robust, hopefully no exploding ADCs ;)

gkasprow, 2017-11-17:
@hartytp I delayed wakeup of +/- 12 and 2.5V by yet another 30m. It's time to start the layout.
BTW, DDS and RFPA boards are ready.

gkasprow, 2017-11-17:
@hartytp we cannot assume that 12V is 12V. This is not stabilised voltage, dependent on number of EEMs in the crate. And we want to make sure that +/- 12V gets powered on when the reference is close to 5V.

hartytp, 2017-11-17:
> BTW, DDS and RFPA boards are ready.

Amazing! Those should definitely take priority.

> It's time to start the layout.

Agreed, go for it.

> @hartytp we cannot assume that 12V is 12V. This is not stabilised voltage, dependent on number of EEMs in the crate. And we want to make sure that +/- 12V gets powered on when the reference is close to 5V.

Makes sense.

Out of curiosity: in the schematic version currently online, what happens to the comparator IC15A before the 3V3 is powered on? It is comparing two voltages that are close to zero, but not well-defined. Is it possible for the +-12V supplies to be enabled briefly, before being disabled once the 3V3 wakes up? Is this what the extra delay you added is for?

One other question: is a diode needed across R52? What happens as the board powers off? Won't this large RC delay help to keep the SR enabled even after the +-12V supplies have been disabled?

gkasprow, 2017-11-17:
@hartytp there is no 6V that would pull the comparator output up and enable the +/- 12V. The 6V rail is enabled by 3.3V PG.
To be 100% sure, I added a diode between PG of 3.3V regulator and comparator output so this should solve any issue with power on transients.
The diodes you ask help discharge the capacitor during power off.
This is to prevent wrong sequence after you quickly disconnect and connect the supply. In such condition the capacitor won't have a time to discharge and after power on the delay circuit won't work.

gkasprow, 2017-11-17:
I added front panel diodes: supply indication and operation with connected second EEM.

hartytp, 2017-11-17:
> @hartytp there is no 6V that would pull the comparator output up and enable the +/- 12V. The 6V rail is enabled by 3.3V PG.

Right, I'd forgotten about that. In that case your design is correct and no extra diode is required.

> The diodes you ask help discharge the capacitor during power off.
This is to prevent wrong sequence after you quickly disconnect and connect the supply. In such condition the capacitor won't have a time to discharge and after power on the delay circuit won't work.

Something like that.

In practice, none of this power sequencing works if the power supply rail bounces around too much (e.g. if a user plugs the IDC in while the power is supplied to Kasli). But, that's something we can live with!

---

Let's fix the power sequencing there then.

hartytp, 2017-11-17:
> I added front panel diodes: supply indication and operation with connected second EEM.

Good idea.

Did you see the other small items on the top post (e.g. SDR/DDR is on EEM1, not EEM0)?

jordens, 2017-11-17:
SDR/DDR is on EEM0 where it should be IMO.

jordens, 2017-11-17:
Also great to hear Urukul has landed! Photos?

hartytp, 2017-11-17:
> SDR/DDR is on EEM0 where it should be IMO.

Is it? AFAICT, on the current schematics, the buffer is on the EEM0 page, but it's driven by the nets EEM1_2P and EEM1_2N...

jordens, 2017-11-17:
Right. That's a left over. But that would probably show up in ERC anyway.

gkasprow, 2017-11-17:
It is fixed.

hartytp, 2017-11-17:
> It is fixed.

Thanks!

Can you upload the completed schematic at some point, please?

gkasprow, 2017-11-18:
done

commit e7b41a2f (2017-11-18): fixed #362 fixed #363
closed 2017-11-18 by gkasprow