sinara-hw/sinara #365: Sayma_AMC shipped without uTCA mating connector

jbqubit, 2017-11-16:
@gkasprow Is the omission of the female uTCA AMC mating connector a design choice or inadvertent? 

[image]

gkasprow, 2017-11-16:
@jbqubit Digikey has problem with their stock - they ship wrong version of the connector. There was batch of  purchased ones and it was returned. So we will ship missing connectors once they arrive.

jbqubit, 2017-11-16:
Good call shipping boards without that connector as the boards seem to mate fine on the benchtop without it. Looks easy to solder-on down the line.

gkasprow, 2017-11-16:
This connector is mounted with one screw only. No soldering required.

jbqubit, 2017-11-16:
OK. Let's leave this open until the correct part arrives and is confirmed to work.

jbqubit, 2017-12-11:
@gkasprow Did these come in yet?

gkasprow, 2017-12-11:
I'll ask and let you know.

jbqubit, 2017-12-14:
@gkasprow Any word on this?

gkasprow, 2017-12-14:
We finally receive them all.
So together with other boards will ship them.
How many do you need?

jbqubit, 2017-12-14:
I need qty 2. But Issue is also about confirming that once they're installed on the AMC that AMC and RTM mate correctly in the uTCA crate.

gkasprow, 2017-12-14:
I will get them on Monday so will confirm it.

hartytp, 2018-01-07:
@gkasprow Did you have a chance to check that these connectors fit yet? Once you've shipped the connectors, please can you close this issue?

gkasprow, 2018-01-25:
yes, they fit perfectly. I shipped a few to Sebastien.

closed 2018-01-26 by jbqubit

hartytp, 2018-03-30:
@gkasprow to confirm I am still missing one if these.