sinara-hw/sinara #366: Allaki insertion-removal, SMA connectors

jbqubit, 2017-11-16:
Using Sayma_RTM v1.0 and Allaki v0.1 there are some geometric difficulties that make insertion and removal of Allaki boards difficult. These don't appear to interfere with testing but need to be addressed before production. 

- Tolerances permit zero gap between Sayma_RTM and Allaki SMA. 
[image]

- Removal of Allaki is possible but requires lots of wiggling. This appears due to three points of binding per Allaki (x4) and RTM. (1) between RTM and tall SMA (2) SMA edge-connector smooth barrel and edge of RTM rear-panel (3) threaded SMA barrel catching on apertures in RTM rear-panel (not visible in photo). 
[image]

gkasprow, 2017-11-16:
@jbqubit this was already spotted by @jordens

jbqubit, 2017-11-16:
I've found that prying the stainless frame upward slightly makes insertion/removal much easier... Is there a straightforward fix for this? Is it easier with the insulating face place?

hartytp, 2017-12-01:
Closing this issue, as these problems should be fixed by changing the connectors on Allaki to SMPs and changing the way we do the front panels on Sayma.

closed 2017-12-01 by hartytp