sinara-hw/sinara #367: Allaki insertion-removal, inter-Allaki interference

jbqubit, 2017-11-16:
Allaki can't be independently removed due to board-board interference of cutouts to accommodate SMA connectors. Sayma_RTM v1.0 and Allaki v0.1. This does not interfere with testing but should not be the case for final production. 

Here are the cutouts. 
[image]

Cutouts interfere with board removal. 
[image]

- A can be removed
- removal of D requires first removing A then B then C 

Tolerance is such that there is zero gap between Allaki at the location of the cutouts.

gkasprow, 2017-11-16:
@jbqubit this is known bug, already on the list.

gkasprow, 2017-11-16:
@jbqubit Sorry, I mixed things. This "feature" was not known before. Thanks!

hartytp, 2018-05-03:
Not relevant any more. Next version will have an FMC footprint without these cutouts.

closed 2018-05-03 by hartytp