sinara-hw/sinara #368: Allaki insertion-removal, SMP insertion-removal force

jbqubit, 2017-11-16:
I'm using 4 of 10 SMP RF couplers. Using Sayma_RTM v1.0 and Allaki v0.1. 

Using a package-weighing scale in my shipping room I measured the insertion force required to couple Allaki to Sayma_RTM on two cycles: 11 N and 13 N. To reduce risk of damage I didn't repeat on additional Allaki. So low statistics. What is expected coupling force for SMP? @gkasprow Do you see something similar? 

Separating Allaki from Sayma_RTM with just 4 SMP is fiddly. The thin Allaki PCB flexes considerably under the required force. Separation with 10 connectors seems like it would very much risk board damage.

jbqubit, 2017-11-16:
Possible solution is to modify RTM design to include solder-on threaded nuts. And modify Allaki to have holes for screws (red circles) and a mechanical support plate on the front side of Allaki to spread the load. Use trapped screws so insertion/extraction force is applied by screws. 

[image]
[image]

gkasprow, 2017-11-16:
@jbqubit I experienced the same impression and my first thought was "poor physicists who are going to use it"  The idea with 2 screws is brilliant. I already use such solderable nuts from [Wurth](http://katalog.we-online.com/en/em/WP-SMBU_THROUGH-HOLE?sid=010615a95c)

dhslichter, 2017-11-29:
SMP comes in three flavors: full detent, limited detent, and smooth bore.  Each one has a different insertion/removal force, with smooth bore having no force at all.  If we populate with smooth bore everywhere and use screws to hold down, or if we populate with mostly smooth bore and just a couple of limited detent, we can reduce the required forces dramatically.  I understand there are other connector types being looked into now, which is totally fine, just wanted to share this information.

hartytp, 2017-11-29:
@dhslichter ACK. However, the cost is bad enough with the relatively cheap connectors that @gkasprow found. I wonder if it would be worse with smooth bore connectors?

gkasprow, 2017-11-29:
@dhslichter Is the dent size a parameter of the bullet or the connector?

dhslichter, 2017-11-29:
https://www.mouser.com/search/ProductDetail.aspx?R=0virtualkey0virtualkey19S141-40ML5

dhslichter, 2017-11-29:
Detent is a connector parameter.  All bullets have the same value.  See e.g. p. 4 of this: http://www.mouser.com/pdfdocs/Emerson_SMP_Connectors.pdf

jbqubit, 2017-11-30:
Thanks @dhslichter !  This is good to know. 

DAC SMP connectors (eg J36) are 
- [MOLEX_85305-0232](http://www.literature.molex.com/SQLImages/kelmscott/Molex/PDF_Images/987650-3241.PDF) | 50 Ohms, SMP Surface Mount Plug, Limited   Detent
- force to engage (disengage) is specified as 15 N (8 N)
- $2.70/ea for 100

AFE mezzanine REF_LO (eg J15) are 
- [AMPHENOL_SMP-MSLD-PCT-10](https://www.amphenolrf.com/downloads/dl/file/id/2828/product/3489/smp_msld_pct_10_customer_drawing.pdf), 50 Ohms, SMP   Surface Mount Plug, Limited Detent
- force to engage (disengage) is specified as 45 N (9 N)
- $3.64/ea for 100

Compare with "Smooth Bore and Catcher’s Mit (SB and CM)" style SMP's in the [Emerson catalog](http://www.mouser.com/pdfdocs/Emerson_SMP_Connectors.pdf).  
- force to engage (disengage) is specified as 8 N (2 N)
- however they're not cheap; 127-3701-201 is $10.41/ea for 100 on octopart; perhaps @gkasprow can source lower cost CM's? 

If all 10 SMP's were changed to Catcher's Mit style this could reduce engagement force to 80 N. That's still a sizable force for such a small area. Some sort of mechanical support for the PCB seems prudent.

hartytp, 2017-11-30:
@jbqubit Why not go for the nice multi-pin high-isolation RF connectors that Greg found? Seem cheaper and better mechanically...

gkasprow, 2017-11-30:
@hartytp just gor confirmation that samples were shipped.
I ordered 2 pcs of IJ5-06-05.0-L-D-1-TR and 2 pcs of IP5-06-05.0-L-D-1-TR
Their stacking height is 10mm but I want to test them.
I'm still waiting for quote of 9mm ones.

hartytp, 2017-11-30:
Great!

jbqubit, 2017-11-30:
Brilliant! Samtec [IJ5/IP5](http://suddendocs.samtec.com/catalog_english/ij5.pdf) have excellent performance DC-4 GHz. Here's what I see from a mfg [test report](http://suddendocs.samtec.com/testreports/hsc-report_ij5-ip5_web.pdf). 

[image]

[image]
(above) insertion loss < 1 dB

[image]
(above) adjacent-channel cross talk < -70 dB

Cost < $15/ea for qty 100 says octopart. There exists a 12-position option. I can't find information on insertion force. 

[image]

(above) Samtec also sells a cable assembly that mates with IP5 called [IJ5H](https://www.samtec.com/products/ij5h). eg  8-position to SMA (IJ5H-08-1000-S-2-01SP1) is < $200/ea qty 10 octopart

jbqubit, 2017-11-30:
Looks like I just did a partial repeat of the discussion in https://github.com/m-labs/sinara/issues/386. That's what I get for responding to Issues without reading all pending Issues.

jbqubit, 2017-11-30:
Insertion force problem is documented. I'm closing this Issue. Discuss in https://github.com/m-labs/sinara/issues/368.

closed 2017-11-30 by jbqubit