sinara-hw/sinara #369: Allaki insertion-removal, SMP tolerances

jbqubit, 2017-11-16:
4 of 10 SMP RF couplers are installed on four Allaki on my Sayma_RTM. All four of the Allaki are successfully mated to the RTM using the SMPs. Two unconnected SMP for the analog input on two of the Allaki have large offsets relative to the corresponding SMPs on the RTM. Using Sayma_RTM v1.0 and Allaki v0.1. 

[image]
(above) This one looks good. 

[image]
(above) These don't look like they will mate. Until testing of RF output is complete I won't try to install all 10 SMPs on a board. So I'm speculating that they won't mate. 

Its not easy to inspect the other SMP-SMP interfaces so I can't comment at the moment on their relative alignment. 

What are the expected lateral tolerances for the mounting of the SMPs? How much slop can the SMP barrel connectors handle?

gkasprow, 2017-11-16:
@jbqubit this is known bug. Two upper boards have 1 mm shift. I remember I fixed this bug, then marked as fixed, but Xpedition crashed and I lost some work :(. So that's why I shifted 8 SMP connectors to make it working somehow. While DAC SMPs mate, the ADC don't. And they cannot be shifted  that easily.
I reported it and it is on the list issues to fix

gkasprow, 2017-11-16:
This bug is only with two upper Allaki boards, lower ones are fine and you can plug all SMPs easily. Extraction will be more difficult :)

jbqubit, 2017-11-16:
OK. So ADC testing using Allaki will be limited to just one of the ADCs. Understood.

closed 2017-11-16 by jbqubit