sinara-hw/sinara #370: Sayma_AMC IPMI

jbqubit, 2017-11-16:
I have a MCH-Baisc v3.5 in same uTCA.4 crate as Sayma_AMC. Sayma_AMC is installed in slot AMC1. Here's what I see. 

[image]

gkasprow, 2017-11-16:
There is no IPMI-compilant firmware in the CPU right now. Just switch on the supply manually.

jbqubit, 2017-11-16:
Do I recall correctly that Creotech is working on this? It's a variation of an open source firmware, right?

jbqubit, 2017-11-16:
I figure it would be nice to monitor RTM voltages/temperature when plugging it in for the first time. But not crucial since you've already done testing of the RTMs.

gkasprow, 2017-11-16:
Yes, they got Saymas very recently so need some time to verify it.
RTM is also supported. Yes, leave it open.

closed 2018-05-02 by gkasprow