sinara-hw/sinara #373: Sinara hardware interest as a poll

jbqubit, 2017-11-17:
Editing tables in github's wiki is annoying and error prone in my experience. I gather that @hartytp has has a similar experience. I've moved the current tabular information to a google doc which is presently open for editing. 

I propose to phrase @jordens [solicitation](https://ssl.serverraum.org/lists-archive/artiq/2017-November/001167.html) for interest in the hardware as a poll with a deadline for response. This encourages groups to respond in a timely fashion. If this is an agreeable approach when should the deadline be? 

Here's what it looks like.... 

[image]

jbqubit, 2017-11-17:
For those who'd like to see a textual representation of this I'll post a version as CSV to a wiki page once quantities settle down.

hartytp, 2017-11-19:
@jbqubit @jordens What timeline are you guys thinking for this order? Which version of the boards are you planning to ship? Current prototypes? Or, wait a few months while we finish testing and bug fixing and then ship the second revision.

Also, when you say "Urukul" do you mean the AD9910 or AD9912 variant? AFAICT, most groups would want the AD9910.

jordens, 2017-11-19:
Joe is doing his own thing here. I am doing what I outlined in the e-mail: get a number on most likely quantity for the next four months to be able to estimate prices.
I see a good level of interest in the AD9912 variant as well. Those are just variants and the variant balance is not expected to affect price much. No need for lobbying.

hartytp, 2017-11-19:
Okay, that makes sense.

From talking to people, there seems to be a bit of confusion about what hardware there is (boards/variants/revisions), when it will be available, how it all fits together, and what the current/funded level of software/gateware support is. With variants, there also seems to be confusion over what the trade-offs are.

The info is all basically there if one looks at the right place (Wiki/Artiq issues/etc), but for people not following closely it, it's easy to miss things. 

Personally, I find the spreadsheet above a bit confusing, as it doesn't distinguish the fact that some boards are ready to ship in prototype version, while the design hasn't even begun on others (and won't for a while yet), it doesn't mention the fact that there are variants of Urukul, etc...

Anyway, if anyone wants to lead the effort to getting these boards distributed properly, that's great. But, I think a bit more coordinate/documentation first would help to ensure that things move smoothly.

hartytp, 2017-11-19:
I'm also unclear about what roles the various people are playing. I thought Creotech or someone else were planning to make a batch of these boards and start selling them on the webshop, and that was being coordinated by @gkasprow. Or, is M-Labs/Quartiq planning to coordinate a large order and negotiate prices?

jbqubit, 2017-11-20:
> From talking to people, there seems to be a bit of confusion about what hardware there is (boards/variants/revisions)

Yes, this could be more clear. I'm open to whatever tells the story better. Editing the wiki tables was a headache so the google docs is my attempt at an easier interface. Would it be better if this were setup as a Google Form? I'm not a fan of the opaque approach advocated by @jordens. 

> Personally, I find the spreadsheet above a bit confusing, as it doesn't distinguish the fact that some boards are ready to ship in prototype version, while the design hasn't even begun on others (and won't for a while yet), it doesn't mention the fact that there are variants of Urukul, etc...

Indeed. Perhaps it would be better to constrain the poll to boards for which we expect a production version in the next four months /and/ ARTIQ support. AFACT on the hardware side this would include:

- 3U Kasli 
- 3U Zotino DAC 
- 3U Novo ADC 
- 3U Urukul DDS AD9910 
- 3U Urukul DDS AD9912 
- 3U VHDCI breakout 
- 3U SMA 
- 3U RJ45 
- uTCA Sayma_AMC 
- uTCA Sayma_RTM 
- AFE Allaki 
- RFPA 

It would be helpful to have a status page for ARTIQ support spanning all the Sinara PCBs.  I have trouble keeping track. https://github.com/m-labs/sinara/wiki/Status

closed 2018-01-19 by hartytp