sinara-hw/sinara #374: Novo v2.0 schematic residual issues

hartytp, 2017-11-18:
Minor residual issues on the Novo Schematic below. Electrically, it all looks good IMO, these are just cosmetics (can be dealt with later as a low priority, but should be done at some point before the final release):

- [ ] Schematic should have a version number somewhere on the title page
- What does "(500k:0.3V/50R)" mean for the OpAmp load in the power budget? Also, the Nv12 rail power is only 80mA, not 84mA (not that the difference matters!)
- Still some displaced text on the schematic, including:
  - [ ] On the ADC page: REF_SHDN, SDR_MODE, all ADC_SDO, ADC_CLKOUT
  - [ ] AFP page: term_ind, REF_2V5, REF_N2V5, R1A and RAB pins 4 & 8 overlap, ADC_INP and ADC_INN
  - [ ] LVDS interface: all LVDS buffers have the text for their inputs outside its box. Same for outputs of LVDS to LVCMOS converters.
  - [ ] The Vcc and GND text for these buffers also overlap (maybe that's not something you can easily fix if it's not your library). Similarly for the bypass and adjust/GND pins on the regulators (again, if you can't easily fix this then don't worry about it).
  - [ ] Ref shutdown text on the power supply page
  - [ ] Text for TP5 overlaps with P3V3_MP1
- [ ] Power budget: can you remove the [mA,W] box by the P3V3 rail this box doesn't make sense anymore, given the way the power budget is now laid out.
- [ ] Can you add an annotation by LDA5A saying "Power On"
- [ ] "Latchup protection" annotation needs to be removed
- [ ] Can you move IC24 under IC22 so that the buffers are in the same order as the pins on the EEM connector (EEM1_0, EEM1_1, ...) this makes the schematic a bit easier to read

hartytp, 2017-11-18:
@gkasprow One non-cosmetic comment:

- We want to keep the cross-talk low for this board (should be limited by the ADC)
- On the current schematic, the most likely area for cross-talk I can see is the N2V5 reference.
- At DC, this is well decoupled between the channels, since it connects to a virtual ground and is driven by a low impedance source.
- At AC, the virtual ground doesn't work so well due to decreasing OpAmp gain. Similarly, the driving OpAmp output impedance increases. This will lead to increasing cross-talk with frequency.
- Not sure if you modelled this to check how bad this effect will be? 
- Do you think it's worth adding pads for a filter capacitor capacitor (between 1nF and 100nF, should be NPO ideally to reduce microphonics) between R1A and R1D to help to reduce this effect? The capacitors can be DNP initially and only populated if we find a problem.

What do you think?

hartytp, 2017-11-18:
> Do you think it's worth adding pads for a filter capacitor capacitor (between 1nF and 100nF, should be NPO ideally to reduce microphonics) between R1A and R1D to help to reduce this effect? The capacitors can be DNP initially and only populated if we find a problem.

Apologies, that obviously doesn't work -- I'd forgotten how these resistors were wired up. But, I do still worry that this summing-junction could seriously degrade the AC cross-talk (the OpAmps are only 10MHz GBP, so their gain is getting pretty low by 100kHz making the bootstrapping poor).

What about this as a different suggestion that doesn't involve adding any additional components:
- We apply the reference voltages to the non-inverting terminals of both driver amps (IC2A and IC2B)
- IC2B requires a +2.5V reference, which we generate as we are currently doing, however we replace the precision 2:1 divider with a divider made from standard (cheap) 0.1% resistors.
- IC2A requires a +2V5*3/2 reference. We generate this from the +2.5V reference using the precision divider we already have in the circuit. In this way, any fluctuations in the +2V5 reference are removed by the ADC's CMRR.
- Optional: to ensure that there is no noise/cross-talk due to pickup in the traces from the reference buffer OpAmps to the ADC driver OpAmps, we can install pads for capacitors at the non-inverting terminals of IC2. These can be DNP by default.

What do you think?

Apologies for not picking this up in an earlier part of the review, these things can be a bit subtle.

hartytp, 2017-11-27:
Have the Urukul AD9912 boards arrived yet? If not, unless you're currently busy with the RF PA, could you have a look at the remaining Novo v2 schematic issues, please? It'd be great to have the prototypes made before Christmas if at all possible.

Thanks!

hartytp, 2017-11-28:
@gkasprow What do you think of the above plan for altering the reference subtraction? If we're going to do this, we should probably make the change before starting the layout...

gkasprow, 2017-11-28:
@hartytp your idea for AFE will work,  remaining  resistor values are 10k + 10k + 5k but they are in different packages so are not ideally matched.
Look at this
[image]

hartytp, 2017-11-29:
@gkasprow That's basically what I was thinking of.

The only difference is that I was thinking of distributing 2 references: 2V5 and 2V0. That saves us 1 divider per ADC channel, at the cost of having an extra reference buffer.

You can do whichever is cheaper/easier/simpler for you.

gkasprow, 2017-11-29:
@hartytp we already have these precision resistors in each channel and otherwise they would be not used.
But if we deliver externally generated 2V0 using matched resistors then tempco matching of divider resistors would not be an issue.

gkasprow, 2017-11-29:
I mean that divider is composed of 3 resistors that are parts of different ladders.

gkasprow, 2017-11-29:
@hartytp please check [panel design.](https://github.com/m-labs/sinara/tree/master/ARTIQ_ALTIUM/PANELS/Project%20Outputs%20for%20PANELS)

hartytp, 2017-11-29:
> @hartytp we already have these precision resistors in each channel and otherwise they would be not used.

In your previous design, there were two dividers per ADC channel. In the new one there are 3 dividers per ADC channel.

One of these 3 dividers produces a 2V0 reference from the 2V5 reference. My though was that it's probably cheaper (and saves PCB space) to generate a single 2V0 reference from the 2V5 reference using a precision divider + OpAmp buffer, and then distribte that to all ADC channels. That way we only need 2 precision dividers per ADC channel, as well as a single common divider to generate the 2V0.

> I mean that divider is composed of 3 resistors that are parts of different ladders.

Yes, but that won't work as well, since the absolute TCR of these resistors is 25ppm/K, so this will give us worse offset drifts.

gkasprow, 2017-11-29:
@hartytp Oh, that's just wrong enumeration, there are still 2 resistors.
But I fully agree - distribution of 2V5 and 2V0 gives us smaller drift and this is what I meant.

hartytp, 2017-11-29:
> @hartytp please check panel design.

Looks good to me!

Very minor comments:
- "EEM2" should probably be "EEM1" since that's what it says on the schematic/silk!
- If there is space, I'd leave "Novo" as a line on its own and have "8x 1MSPS 16-bit ADC" on a separate line

One question: does ADC channel 0 correspond to the BNC channel 0?

gkasprow, 2017-11-29:
@hartytp 
BNC0 = ADC CH7, 
BNC1 = ADC CH6, etc

hartytp, 2017-11-29:
> @hartytp Oh, that's just wrong enumeration, there are still 2 resistors.
But I fully agree - distribution of 2V5 and 2V0 gives us smaller drift and this is what I meant.

Aah, I see.

Well, it seems having a second OpAmp to supply the 2V0 reference is pretty cheap (we already have an OpAmp for 2V5, so we can make it a double). Unless it's going to cause problems in the routing, I'd prefer to do that use the 25ppm/K divider.

hartytp, 2017-11-29:
> @hartytp
BNC0 = ADC CH7,
BNC1 = ADC CH6, etc

Okay, well I would have preferred to have BNC0 = ADC0, but if that makes the routing a lot easier then we can do it!

Please can you make sure to add a note on the top page of the schematic that makes this mapping clear?

gkasprow, 2017-11-29:
At the moment routing looks like this: 
[image]
I can swap channels or move ADC to the bottom of the PCB
Since it is low frequency, it is not that critical to have these line nicely routed and I can keep right channel assignment.

hartytp, 2017-11-29:
No, leave it as it is and add a note on the schematic.

Components on the bottom of the PCB are a pain, and the routing would be a mess if you tried to reverse those pins.

gkasprow, 2017-11-29:
That's true - it's much easier to damage them mechanically.

hartytp, 2017-11-29:
> That's true - it's much easier to damage them mechanically.

Yes. Also, reworking is harder (or, assembling a board by hand if one ever needs to) and probing is harder as well. Generally, I try to avoid it if possible.

hartytp, 2017-11-29:
@gkasprow Can you produce an updated schematic when you have time, please?

I'd like to sign off on a version with all the functional changes made. The cosmetic changes (text in the wrong place) can wait until later if that helps...

gkasprow, 2017-11-29:
@hartytp I placed EEM2 LED on the front panel to indicate that ADC operates in fast mode with 2 IDC connected. Maybe better name would be NU ?

hartytp, 2017-11-29:
@gkasprow I don't like "NU" for this, because "fast" mode can be used for other things than the NU servo.

I'd prefer either "EEM1" to indicate that EEM1 is connected as well as EEM0, or to have "fast mode" if there is room.

hartytp, 2017-11-29:
Thanks for fixing all those issues!

hartytp, 2017-11-29:
Reopening, as there is still some displaced text that needs fixing at some point.

gkasprow, 2017-11-29:
Tom, This is Altium PDF converter bug. Below is original screenshot
[image]

hartytp, 2017-11-29:
@gkasprow  ACK.

Actually, in the new PDF it seems fixed anyway.

closed 2017-11-29 by hartytp

gkasprow, 2017-11-29:
Yep, this needs manual tuning and several iterations to make it look acceptable.

hartytp, 2017-11-29:
What a pain!

Anyway, the current schematic looks lovely, so thanks for the effort!

marmeladapk, 2017-11-29:
@gkasprow It seems that Altium 17 generates properly PDFs (in 16 all text was shifted to the left).

gkasprow, 2017-11-29:
Thanks,
I don't want to move to AD17 because it less stable than 16 and it crashed so many times that I switched back to AD16. Let's see what AD18 brings.

marmeladapk, 2017-11-29:
I used it just for pdf generation.

gkasprow, 2017-11-29:
OK, I can do so.Thanks.

hartytp, 2017-12-02:
@gkasprow Question for you: did we miss a trick here? We're using an OpAmp + precision dividers to do the single-ended to differential conversion + common-mode voltage offsetting. Is there a cheap ADC driver IC with a convenient supply range (ideally +-12V) that we could have used instead? I know there are some ICs that can do both the SE->DE conversion and offsetting in one go, but I've never used them. We would probably still have had to have a single precision divider to do the divide by 4, but maybe we could have saved the other precision resistor network?

gkasprow, 2017-12-02:
@hartytp such approach was used on original LT devkit so I simply reused it. 
There are quite a lot of precise ADC drivers like 
[LTC6362](http://cds.linear.com/docs/en/datasheet/6362fa.pdf)
[LTC6363](http://cds.linear.com/docs/en/datasheet/6363fa.pdf)
[ADA4922](http://www.analog.com/media/en/technical-documentation/data-sheets/ADA4922-1.pdf)
[ADA4941](http://www.analog.com/media/en/technical-documentation/data-sheets/ADA4941-1.pdf)
[OPA2625](http://www.ti.com/lit/ds/symlink/opa2625.pdf)

hartytp, 2017-12-02:
@gkasprow Thanks for looking into that!

AFAICT, none of these ADC driver amplifiers would simplify our circuit because:
- Since our signal is bipolar, but the ADC is unipolar, these driver amps would generally need an OpAmp before them to add the common-mode offset. e.g. ADA4922-1 non-inverting output is just a unity-gain copy of the input signal. 
- Most of them only run of +-5V so we'd need to add another (!) supply rail to this design.

So, unless there is a more suitable IC out there, it looks like our current design is the simplest/cheapest. Good to know :)

hartytp, 2017-12-04:
@gkasprow I had a bit more of a look into these chips since I was curious, and I think we did the right thing in our current design :)

tl;dr: if we could find an ADC driver with a fixed gain of 1 (or, better yet, 1/4) and a +-12V supply then it would be the perfect choice for our circuit. I haven't seen such an IC...

The reason I don't think that using an ADC driver IC would help us are:
- None of the ones I saw (admittedly on a quick look!) run from +-12V supplies. We could run from +-5V supplies in principle and use a "rail-rail" amplifier, as is currently done on BaseMod (although that means adding yet another supply rail!). However, even these amplifiers perform significantly worse close to their supply rails. I guess we might be able to fix this by using a lower reference voltage so that the ADC takes a signal less than 0->5V, but it's not totally straightforward.
- These amplifiers all require external resistors to set the gain. Currently BaseMod uses standard resistors for gain setting, which will be quite drifty (100ppm/K IIRC). Doing this properly requires a precision divider for gain setting, so using the ADC driver does not actually allow us to remove a precision divider from our circuit. Thus, it's not actually cheaper or simpler.
- I like our circuit because it uses the same OpAmp that's on Zotino. If that gets used on Zotino, Sampler, BaseMod and MixMod then we should be able to end up ordering it in 1k+ quantities, which will give us a really good price break. Having the same IC everywhere also means fewer undocumented "features" to discover.

commit 6c44a14d (2017-11-29): fixed #374