sinara-hw/sinara #375: Novo v2.0 layout/mechanical

hartytp, 2017-11-18:
Issues from v1.0:
- [ ] 3mm higher EMC shield on PSU
- [ ] add front panel handle fixing holes
- [ ] either add ~2mm plastic washers on BNC barrel **behind** front panel or move BNC connectors 2mm towards FP. This is to enable fixing the BNC connectors to the panel using the screws.
- [ ] run PI and thermal simulations
- [ ] increase PSU shield height
- [ ] Silk + schematic + panel should be "v1.x" not "rev1.x" (NB schematic should have a version number on it somewhere)

gkasprow, 2017-11-28:
@hartytp don't you mind if I swap (reverse) ADC channels? This would make routing easier.

gkasprow, 2017-11-28:
I can also place the ADC chip on bottom side of the PCB, but outside of the shielding can.

hartytp, 2017-11-28:
> @hartytp don't you mind if I swap (reverse) ADC channels? This would make routing easier.

I think it's important that all front panels should order the connectors in the same way. This makes it **much** easier for users. Currently, our convention is that channel 0 is at the top, and the last channel is at the bottom.

Novo v1.0 is about the only board we've designed so far that violates this convention, so it would be good to change it to match the other boards.

Could you update the front panel though asap though please, so I can check that I'm happy with the new connector order?

hartytp, 2017-11-28:
@gkasprow Are you planning to order things so that the top connector has channels 0 and 1, or so that it has channels 0 and 4? I'd prefer the former.

jordens, 2017-11-28:
Swapping them arbitrarily is fine, as long as the silkscreen and panel labels are still sequential.

jordens, 2017-11-28:
And please stick with
```
0 1
2 3
4 5
6 7
```

hartytp, 2017-11-28:
> Swapping them arbitrarily is fine, as long as the silkscreen and panel labels are still sequential.

I'd prefer not to do this unless it makes routing significantly easier/better; if it allows you to avoid extra layers or torturous routing that makes pickup/cross-talk worse then fine, but otherwise let's try not to do this.

Keeping the mapping between connectors and ADC channels straight forwards makes life that bit easier for users. e.g. if you use this ADC in single lane/single EEM connector mode and only want to read out two channels as fast as possible then it's nice if those correspond to connectors 0 and 1, rather than something random.

jordens, 2017-11-28:
Yeah sure. Don't do it if it is not really necessary. But swapping odd-even is fine, reversing is also fine, and doing both is also fine IMHO. There is nothing that says that when doing a readout on a single line, the channels need to be read out in that sequence. To do partial readout yourself you need to understand everything about the protocol anyway. I.e. the API will at most support all or single channel readout. The speed of the latter would always depend on the channel number.

gkasprow, 2017-11-28:
@jordens @hartytp I want to reverse channel count only. So it will be trivial to fix in software.
I confirm, channels are counted as Robert wrote.

hartytp, 2017-11-28:
> @jordens @hartytp I want to reverse channel count only. So it will be trivial to fix in software.
I confirm, channels are counted as Robert wrote.

That's fine then.

If you're reversing the channel count compared with v1 then you're actually fixing a bug anyway, which is nice ;)

gkasprow, 2017-11-30:
@hartytp @jordens The Novo layout is roughly finished. I will do PI simulations so it will change anyway. Please have a look at it. It still needs cosmetic tweaking and copper balancing.

gkasprow, 2017-11-30:
PI simulation looks OK for all rails.
3V3 rail
[image]
[image]
[image]

For 12V rail the current density is high, but within safe margin.
[image]

after optimisation
[image]

hartytp, 2017-11-30:
@gkasprow Happy to leave the layout/routing to you. (This is hard for me to review properly based on the PDFs alone, so I'd prefer to let you make the decisions).

From a quick look over it, it all looks good though! Although, are is there a layer missing from the PDF?

gkasprow, 2017-11-30:
OK, so if there are no comments, I will have look at it on Monday once again and start proto production.

hartytp, 2017-11-30:
@gkasprow Sounds good! Maybe let me have one last look at the finished layout before you send for production in case there are any trivial cosmetic issues with silk etc. We only need to produce two for now (one for @jordens to do software development and one for testing). Until @jordens finishes the software, we can use the Novo v1.0 that we already have.

Are the items at the top of this issue resolved now? can they be ticked off?

Oh, and one final thing: please can you change the name from Novo to "Sampler" (with apologies to @jordens ). This needs to be done on schematic, silk and front panel (I'll take care of the Wiki). Thanks!

gkasprow, 2017-12-06:
@hartytp Any issues to fix? Can I produce prototypes? How many, 3 pcs? Do we call it v2 or v1.2 ?

hartytp, 2017-12-06:
> @hartytp Any issues to fix?

@gkasprow Did you upload the final layout? I'd like to have a look at it before we send it off.

Did you change the name on the Silk/schematic/front-panel? This board is now Sampler (or an appropriate polish lake).

hartytp, 2017-12-06:
> Do we call it v2 or v1.2 ?

Well, IIRC, the versioning convention was that if there was a major change then we'd go from v1 to v2. This seems like a pretty major change, so I'd say v2.0.

> How many, 3 pcs? 

How many were paid for in the original WUT contract? 2?

gkasprow, 2017-12-06:
OK, let's call it v2.0.
There were 2 pcs paid.
I will change to v2.0 and upload the files.

jordens, 2017-12-07:
Could you add two more v2.0 for me?

hartytp, 2017-12-08:
@gkasprow Are you ready to upload the new design files for me to check before manufacture?

While you're at it, could you make a few quick fixes for me:
- [ ] Text around R4 is upside down
- [ ] Can you add an annotation saying something like "Maximum power into termination resistor: 0.5W"
- [ ] I've moved the files from Kasli/3U/PCU_3U_ADC (or whatever the old path was) to "EEMs/Sampler" please could you make sure you add any more design files there, and also rename all the altium files/PDFs etc Sampler.sch or whatever.

Thanks!

hartytp, 2017-12-08:
> Could you add two more v2.0 for me?

Assuming you don't need us to send you one for software development?

jordens, 2017-12-10:
I need two independently of the servo development. With a board for servo development that will the probably return to you it would be three.

hartytp, 2017-12-10:
@jordens Sounds good.

@gkasprow Low priority, but if these boards haven't gone to manufacture/it's not inconvenient, please can you change the termination resistors from 50R to 240R? Thanks!

cjbe, 2017-12-10:
@hartytp I don't think moving to a 240R termination is a wise idea. The only argument for moving to this is that a 50R termination is easy to overload. 

Overloading 50R terminations is a standard issue in equipment, and the answer is to not be an idiot! Most lab sources don't have enough ompf to drive more than 0.5W into a 50R termination anyway. My firm preference is to stay with a 50R termination. (If people do blow up the 50R termination, it is easy to replace)

hartytp, 2017-12-10:
> Overloading 50R terminations is a standard issue in equipment, and the answer is to not be an idiot! Most lab sources don't have enough ompf to drive more than 0.5W into a 50R termination anyway.

That's my feeling too.

> @hartytp I don't think moving to a 240R termination is a wise idea. The only argument for moving to this is that a 50R termination is easy to overload.

To be clear: are you pro removing the terminator entirely, or do you want to keep it? If so, what use-case do you have in mind for it?

cjbe, 2017-12-10:
I think it is useful to keep the switchable 50R terminator.

The prime use case seems to be for compatibility with something that is designed to drive 50R - for example a photodiode amp that is characterised for a 50R load.

Of course this is not necessary for the bandwidths we use here, but lab standards for termination are 'high impedance' or '50R' - this is what people expect, and it seems silly to break this expectation.

hartytp, 2017-12-10:
Okay, good, that was my original thinking too. Let's stick with 50R.

50R is also kind of nice with 4-20mA current loops since it maps 20mA to +1V, so you can use the G=10 setting to get a +1V signal.

FWIW, I really can't think of many transducers that can trash a 0.5W 50R terminator -- that's 100mA/5V, which is a hell of a current and requires something running off a >5V supply (>10V supply if the circuit is 50R source terminated). So, basically no OpAmp is going to damage this, nor any digital circuit (e.g. 5V TTL is fine). As @jordens has pointed out a few times before, users will always be able to break this hardware if they *really* try, so we have to rely on at least a small amount of common sense. 

About the only thing I can think of that we have in our lab that could damage this is (short of an actual power supply/mains electricity) the LEM Ultrastab, which can put 1W into 50R under some circumstances. But, that's a pretty contrived case.

Let's stick with 50R. We'll put a note on the schematic, Wiki and other docs reminding users of the max power rating here.

gkasprow, 2017-12-11:
@hartytp I will update all the design files names while doing second revision of them.

hartytp, 2017-12-11:
Thanks!

Have these gone to production yet? Can I see the production files?

-----

One other question: while the termination resistors are nominally rated to 0.5W, do you think they will actually be able to handle that? They might actually only take a lower power because:
1. Neither side connects to ground
2. The ambient temperature for these boards could be quite high when in a rack with a few Urukul/Kasli/etc
3. If the screening is fitted then the resistors get no direct airflow.

Very low priority, but should we consider using two resistors (series/parallel) to form the terminator in order to ensure that they can actually take 0.5W CW without damage? This can be done in the next revision. What do you think?

gkasprow, 2017-12-11:
@hartytp I replaced R4 with 1W one. To place two x100R, I'd have to move them to the bottom side.

hartytp, 2017-12-11:
> @hartytp I replaced R4 with 1W one. To place two x100R, I'd have to move them to the bottom side.

Thanks!

Out of curiosity, is that the same package? I'm surprised that you can dissipate that much heat in this configuration. Have you done the standard things like making sure there is some ground around under the resistor to act as a heat sink? Also, using thick traces into and out of the resistor to help spread heat.

gkasprow, 2017-12-11:
Nope, it's almost twice as big.

hartytp, 2017-12-11:
If that fits, then that sounds good!

So, even with limited air flow, etc, we expect the 1W package to be good to at least 0.5W, right?

gkasprow, 2017-12-11:
I added some copper around so this should work. I'll publish production files in a moment

gkasprow, 2017-12-11:
@hartytp [done](https://github.com/m-labs/sinara/blob/master/ARTIQ_ALTIUM/EEMs/Sampler/Project%20Outputs%20for%20Sampler/Sampler.PDF)

gkasprow, 2017-12-11:
So we build 4 prototypes?

hartytp, 2017-12-11:
> So we build 4 prototypes?

Yes please. Two for us (paid for by the original design contract) and 2 extras, which @jordens will pay for.

> I added some copper around so this should work. I'll publish production files in a moment

Thanks!

hartytp, 2017-12-11:
@gkasprow All looks good to me! Let's move onto production.

closed 2017-12-11 by hartytp

gkasprow, 2018-01-24:
New Sampler arrived:
[image]

hartytp, 2018-01-24:
Whoo! Looks lovely.

@gkasprow We'll handle all detailed testing, but can check the power supplies + power sequencing and check you get samples out with the DevKit please? If that checks out then please ship one to M-Labs and one to us asap.

@jordens @sbourdeauducq I'm planning to wait until you've written the general purpose driver for Sampler before doing the detailed testing. However, if you anticipate there being any significant delays getting that up and running then please let me know.

jordens, 2018-01-24:
You'll probably want to ship that one to QUARTIQ though.

gkasprow, 2018-01-24:
I have 4 pcs.

hartytp, 2018-01-24:
@gkasprow In that case, let's do this:
1 pc for @jordens 
1 pc to stay in WUT for now
2 pcs to come to oxford