sinara-hw/sinara #376: Urukul DDS v1 testing

gkasprow, 2017-11-18:
DDS boards arrived.

[image]

I plan to do following tests:
- [ ] Power supply
- [ ] CPLD programming
- [ ] DDS operation, channel by channel using ADI devkit
- [ ] gain switching using  USB-> SPI converter
- [ ] clock switching and idstribution.

jordens, 2017-11-19:
Thanks.
I'd like to see the AD9912 variant tested early, then the AD9910 variant.
We discussed and outlined initial testing [earlier](https://github.com/m-labs/sinara/issues/354#issuecomment-342750073).

jordens, 2017-11-19:
Could you also shoot a few preliminary photos od the AD9912 variant?

gkasprow, 2017-11-20:
I have 9910 in my hands, will get 9912 in 2..3 days.
So far the power supply works after minor modification of bias circuit.

gkasprow, 2017-11-20:
Clock distribution also works, CPLD is visible under ISE but it crashes just after it finds the chip. I'll install it on another computer.

gkasprow, 2017-11-20:
well, Impact crashes on another computer. Maybe WIN10 is an issue here..

jordens, 2017-11-20:
There are also xc3sprog and openocd which can program that CPLD. But YMMV.

gkasprow, 2017-11-21:
CPLD works, I managed to program it. The solution is described [here](https://wiki.nus.edu.sg/display/mpsoc/Solution+for+Xilinx+ISE+14.4+Problem+on+Windows+8+64bit)

jordens, 2017-11-24:
@gkasprow is the CPLD code working? can you talk to the DDS?

gkasprow, 2017-11-24:
I tried and it works partially. The data and clock is passed but the chip
select does not. I didnt initialise the CPLD register so mabe it is an
issue but it should work with default settings. Im at the quantum
engineering workshop in London. Im surprised that very few people heard
about Artiq :) On Monday will continue with tests.

jordens, 2017-11-24:
depends on the dip switch settings

gkasprow, 2017-11-24:
Yes. This I know. I set only one for 9910 chip. Rest are off

jordens, 2017-11-24:
Ok. Then if you set CS2=1 and CS0=CS1=0 then it should assert DDS0.CS_N. CS2=1=CS0=1 and CS1=0 for DDS1.CS_N etc.

gkasprow, 2017-11-24:
This is what I did. I will dig into this once I'm back

jordens, 2017-11-24:
Just FYI there is verilog source code and ucf in that tar ball as well if you are uncomfortable editing the python source code.

gkasprow, 2017-11-24:
Cool. Its far easier to play with it. One day in a future I wil dig into
this Migen thing but so far did not have enough time and motivation:)

gkasprow, 2017-11-27:
DDS channel 0 works. I can control it with original devkit and original @jordens  CPLD code.

gkasprow, 2017-11-27:
I connected 4 channels in parallel. With 200MHz reference, x50 multiplication it generates 200MHz and consumes 620mA@12V

gkasprow, 2017-11-27:
All 4 channels are in sync, so it means PLL is locked.
I modified CPLD code and turned on the RF switches. I observe  small signal at the output. The last thing to play with are RF switches

gkasprow, 2017-11-27:
All four outputs with different amplitudes. No termination.

[image]

jordens, 2017-11-27:
Yay!

hartytp, 2017-11-27:
> All four outputs with different amplitudes. No termination.

I assume this was done using the RF attenuators, right? Assuming that the horrible distortion there is due to the lack of termination causing the amp to saturate.

If so, it looks like the boards pass all basic functionality tests with no major issues! Well done.

hartytp, 2017-11-27:
So, I think we're ready to start shipping the prototype boards. From the AD9910 Urukul boards we paid for, please can you:

- Keep 1 in WUT for detailed testing later on
- Ship 1 to M-Labs/Quartiq (where ever @jordens wants it). Unless @jordens is in a hurry for this, and assuming the AD9912 boards will be here in the next few days, you might as well wait and ship this at the same time as the AD9912 boards to save on shipping costs
- Ship the rest to Oxford (send me the quote when you're ready).

hartytp, 2017-11-27:
Some questions/comments while this is fresh in my mind:

> All four outputs with different amplitudes. No termination.

Did you try with termination? Does the distortion go away?

> consumes 620mA@12V

This is all features active and the DDS running at 1GSPS, right? The current power budget (which has a couple of minor errors) gives 750mA as a worst-case. So, we're about 17% under, which is good! This means that a 16-channel DDS system should consume approx 2.5A, which is fine.

> All 4 channels are in sync

To confirm, this means that feeding DDS_0 SYNC_OUT into DDS_1 through DDS_3 keeps all DDS phases synchronized (this is what we've referred to as "basic synchronization"). That's good news. 

> PLL is locked.

Good news that the PLL locks. We need to test this with a 125MHz reference clock at some point, but that can be done later.

> I modified CPLD code and turned on the RF switches. I observe small signal at the output.

The isolation from the switches should be >60dB at these frequencies. So, does "small" mean that the voltage amplitude at the output decreases by a factor of roughly a thousand with the switch off? (No need to do a detailed measurement right now, just want to check that "small" means roughly 60dB rather than roughly 30dB).

gkasprow, 2017-11-27:
@hartytp I sent "random" data to attenuators to see if react at all.
Tomorrow I will connect my FPGA based USB to SPI converter and set desired values and continue tests. I want to finish them tomorrow and switch to solving Sayma Ethernet problem.

hartytp, 2017-11-27:
@gkasprow Sounds like a good plan!

> I did not synchronize the DDS chips. They run so by default because were configured using same SPI sequence.

That's what I meant. So the basic synchronisation works then, which is good news.

gkasprow, 2017-11-28:
I will get 9112 today.
I tested so far 2 pieces of 9910.
Tomorrow they will be shipped to @hartytp and @jordens .

hartytp, 2017-11-28:
Thanks @gkasprow!

Are you planning to test each board individually? Is that necessary? If it will save time, I'd be happy if you just test 1 board before shipping.

gkasprow, 2017-11-28:
@hartytp @jordens we have front panels already but stickers will be ready in 2 days. Do you prefer to have panels delivered later with sticker on it or panel now with board and sticker later?

hartytp, 2017-11-28:
Which sticker?

hartytp, 2017-11-28:
> @hartytp @jordens we have front panels already but stickers will be ready in 2 days. Do you prefer to have panels delivered later with sticker on it or panel now with board and sticker later?

We can wait a couple of days, so please can you ship them with finished panels?

jordens, 2017-11-28:
Panels later is fine. For opticlock we might want to do custom panels anyway.

gkasprow, 2017-11-28:
@hartytp the panel consists of aluminium body and self-adhesive silver sticker with graphics.
We will ship remaining boards anyway later, so later on you can attach the sticker or the panel with sticker yourself.

hartytp, 2017-11-28:
> @hartytp the panel consists of aluminium body and self-adhesive silver sticker with graphics.
We will ship remaining boards anyway later, so later on you can attach the sticker or the panel with sticker yourself.

Okay, I thought it was engraved rather than being a sticker, but I must have been mistaken.

Can you attach the sticker to the panels for us before shipping the panels, please?

> @hartytp the boards need modification. chage 2 resistor values and add piece of wire. I test basic functionality of all boards before shipping, some of them are broken do to one of many reasons - open or shorted PCB traces that appear during soldering, shorted component leads, broken components, etc.

Okay, makes sense. Thanks for doing that!

hartytp, 2017-12-11:
@gkasprow Is this done? Can we close this issue now?

gkasprow, 2017-12-11:
Yes, did you get your board? I shipped it a few days ago.

closed 2017-12-11 by gkasprow