sinara-hw/sinara #377: Urukul DDS v1 errata

gkasprow, 2017-11-20:
- [ ] connect BIAS of IC7, IC11 to Vcc >Vout+1.3V, in our case 5V. IC18 BIAS is connected OK.

jordens, 2017-12-12:
@gkasprow A lot of the pins of the LVDS-to-CMOS converters did not reflow correctly. Do you need photos or do you see that as well?

gkasprow, 2017-12-12:
sure, post it. Technosystem gives warranty for the assembly so it's their deal.

gkasprow, 2017-12-12:
@jordens I can see them on my boards too...

jordens, 2017-12-12:
We don't have the time to send the boards back for them to fix it. So we'll fix it ourselves. But I'll post the photos nonetheless.

gkasprow, 2017-12-12:
OK.
I will send them the boards I'm not using at the moment

jordens, 2017-12-12:
The photos are here: https://share.quartiq.de/s/KGZOJ60rIPNyvKU

jordens, 2017-12-13:
News to me, from TI [SLUA271](http://www.ti.com/lit/an/slua271a/slua271a.pdf#page=15): "IPC-A610D does not require a side fillet, since the side terminations are not plated. While it may appear the wetting is reduced on the side of the component, the solder joint underneath is not affected." I.e. we need to ensure proper wetting by x-ray/electrical.

jordens, 2017-12-14:
@gkasprow Regarding the patching of BIAS of IC7, IC11: I am a bit worried about the lack of decoupling now on that pin. Could you add 2x1µF on that wire somewhere close to chips when patching the other boards?

gkasprow, 2017-12-15:
I added. But this is only FET biasing, so should not cause any issues. It consumes very little power.

hartytp, 2017-12-15:
@jordens Out of curiosity, what's the objection here? AFAICT on the current schematic, there is already a 1uF ceramic on the bias pun of each regulator (C165 and C77). That's what's recommended by the data sheet. So, AFAICT, we don't need more components added to the schematic, right?

Is the objection that the placement of these capacitors isn't close enough to the BIAS pins on the ICs?

gkasprow, 2017-12-15:
Nope. But I cannot cut the trace and leave the capacitor connected. So had to add another one, close to the BIAS pin. The distance of the capacitor is not critical since almost no current is flowing.

jordens, 2017-12-15:
@gkasprow Ack. There is little current. But iirc the ldo reference is running from and thus it needs to be low noise.

a-shafir, 2017-12-15:
There shall be no problem with the BIAS pins.

I have not reviewed the board in details but i already noticed some improvements that can be done:
1. add source serial termination resistors close to CPLD on SCLK or better on all the signals. This CPLD has no internal termination in IO  and we need to keep FAST slew rate for the performance. So there can be ringing that can reduce the usable SPI clocks as well as the analog noise since the control traces are long and all around the PCB. Note: the SPI clocks can go up to 66MHz by the concept and it must be terminated.
2. add some protection to the SMA connectors, especially the clock input. Currently it can be easilly burn both by ESD and an incorrect clock source.
3. replace 100nF on the clock patchs (C196 etc, there are many). That 100nF are not really good parts for >tens MHZ signals. 10nF RF class parts shall be more appropriate.
4. Check all non-RF the traces on the RF side of the board (ATT_CLK, ATT_S_OUTs etc, basically all of them).  I think need to do more debouncing and source termination resistors everywhere since it is a obvious source of interference with the analog circuits.
5. For any "potentially user configurable" parts use bigger package than 0402 and label them.
6. Check the signal paths of CKIN1_P/N, CKIN0_P/N. Curretnly it is not well balanced and if there are 2 active clock sources (INT and SMA) it can easily interfere. Also there are some potentially noise power lines along the clocks and it certainly can be shifted closer to the board edge and shielded/separated better.

a-shafir, 2017-12-15:
@gkasprow 
1. About the termination. It needs serial termination at source. Like 22-33R serial next to the source. Ideally it can be optimized for the traces/stack used in the project. The optimal value can be tested or simulated. 33R shall be fine.

Note: There ARE in fact separate control lines for each DDS. Check DDS_SCKI0 and others on the Altium. It can be nearly ideally terminated both at source and at the end of each trace, there are nearly no multi-point. There 9910 and 9912 are near comparing to the length of the trace.

4. ATT_CLK can be fixed to be non-forked by moving the CPLD trace up. Also can be serial termination on CPLD side or even parallel next to IC3D. ATT_S_IN is single tap straight. ATT_S_OUT and the junctions between the ATTs can be easily terminated at source, there is plenty of space.

2. There is no protection at all. It is even terminated on the clock chip. I am sure it can be added. Simplest will be 2 diodes protection to GND and to a locally filtered 3.3V. Ideally need a protection and a buffer (singe ended to differential) next to the sources connectors. I think the buffers noise will not be a problem since the DDS PLL filter will not pass it. But interference from the power etc is low freq and more dangerous imho.

3. It shall work. I just saying that 100nF does not looks like RF signal path component coz of parasitic etc.

6. Of course need a ground shielding traces (with multiply vias to the ground plane along the pcb edge as well between the clocks pairs and other signal. It can be done nicely and symmetrically - there is enough space. No emi concern because it differential (at least shall be balanced differential if proper buffers used, see "2".

cjbe, 2017-12-15:
@gkasprow on the Urukul I have it is not possible to fully mate to the front panel SMA connectors as only ~5mm of the barrel is accessible when the connector fixing hardware (plastic washer, spring washer, nut) is attached.

a-shafir, 2017-12-15:
@cjbe @gkasprow yes, the SMA connector needs some improvement:
- need longer/extended thread connectors. There are some.
- need a thick (or few thin) washer between the connector and 1st plastic washer to be able to tie properly the nut.
- need to move the connectors and the LEDs bit closer to the panel, currently there is about 0.5-1mm "wasted"

gkasprow, 2017-12-15:
@a-shafir You are right.
But in DDS board SPI data and CLK are already load-terminated, so there is no issue. I'm sceptical about source-termination of a line that goes to multiple capacitive loads.
30R is not treated as source termination but rather as a kind of RC filter to make edge transitions smoother and to get rid of oscillations. 
Source termination need to be matched to the Z of the line and the end of line must be opened at the end.
So let's not talk about termination but data edge smoothing :)
Did you actually look with a scope at these lines?
What is the CPLD output resistance? Maybe it is already an order of 30Ohm so there is no issue at all.

In case of diff lines it's hard to make shielding for lines on top or bottom layer :)

gkasprow, 2017-12-15:
@cjbe I tried prior shipment and it looked fine. Maybe my SMA connector was slightly shorter.
Anyway, the mechanical 3D model is different than actually mounted connector. I already added this to the list of fixes to do.

gkasprow, 2017-12-15:
@a-shafir concerning these LEDs, are you talking about Urukul? I aligned them to make sure they touch the panel. On previous boards like ADC and DAC there was such issue.

a-shafir, 2017-12-15:
@gkasprow 
you right there is a load termination on the major DDS lines.

It is still no termination on ATT lines.

A scope is barely useful for testing it. I think the best is compare the output noise when SPI is not active with the noise when SPI busy on DDS or ATT. 

By shielding i mean Faraday-caging. And also symmetrically placed. Basically to have the perfect signal line (quite long in case of clks) it shall be all-differential and placed properly with the signals/grounds around. I don't think that it make sense to move it to internal layers in this case. I have more concern about the single-ended to differential conversion in this case.
Yes, it is not good idea to use both clock connectors simultaneously. But it needs to be warned in the board manual.

Another problem that if we use the internal 100MHz oscialltor it can't be disable so in it can be a serious interference issue with 100MHz-like input clock in this case. Btw, the OSC can be disabled via a power switch.

Yes, there are some strange gaps on Urukul we have here. Can you see it on your board?

I think for the proper assembly need to tie the SMA nuts first. Also add washers or replace the SMA connectors to have the plastic washers properly sit on the connector body. And only than tie the PCB mounting screw.

For the best result i suggest verify the SMA and LEDs connectors holes to assure it aligned strictly and the same along the line. And for the mounting holes i suggest bigger size or oval shape so it will allow precise and simple assembling (with some instructions for the factory).

a-shafir, 2017-12-15:
[image]

hartytp, 2017-12-15:
@jordens Had a closer look at the bias capacitor issue you raised:
- Datasheet p4 note (3) "If VIN and VBIAS are connected to the same supply, the recommended minimum capacitor for the supply is 4.7 µF." AFAICT, this only applies to IC18 (IC11 and IC7 are biased from 5V and supplied from 3V3). It might be worth swapping the bias capacitor to 4u7 here. (In practice, I suspect that there is so much capacitance on the 3V3 rail already that this is not necessary, but probably good practice).
- For IC11 and IC7 I don't see any reason to add more capacitors. The datasheet specifies a minimum capacitance for the BIAS as 100nF, with 1uF the preferred value. The reference and many of the specifications use a 1uF capacitor. However, we could swap to 4u7 if you'd rather...

gkasprow, 2017-12-15:
@a-shafir there are metalic washers missing. In the new boards I ordered technosystem to place them first.
To get rid of the gaps just release the screw that fix the board, push the panel to the LEDs and tighten the screws.

hartytp, 2017-12-15:
> A scope is barely useful for testing it. I think the best is compare the output noise when SPI is not active with the noise when SPI busy on DDS or ATT.

Well, I agree with half of this ;)

If you want to do noise testing on these boards then go ahead. If you can show that changing the termination makes the noise better then that would be a strong argument to change it. If you have the proper tools then this should be an easy measurement to make, and data is more helpful than speculation.

But, I don't agree that scopes aren't useful; EMI is a function of the voltages and currents on the line. Lack of termination generally causes EMI issues by increasing the voltage/current on the line. The traces are all <lambda/4 for the frequencies of interest (the CPLD outputs aren't that fast IIRC, particularly after Greg's edge filtering) so an open termination just gives a larger voltage wave. This is easy to see with a scope: if the voltage at the end of the line isn't significantly bigger than one would get for a terminated line then termination can't cause increased EMI.

As @gkasprow points out, termination can make EMI worse for the LVCMOS lines due to increased current flowing.

hartytp, 2017-12-15:
> add some protection to the SMA connectors, especially the clock input. Currently it can be easilly
burn both by ESD and an incorrect clock source.

Breaking these clock inputs isn't that easy; they already have quite robust clock buffers on them. You'd need a source that can drive more than 3V6pp AC coupled into 50R (that's +15dBm). Any user who connects something like that to a clock input shouldn't be in the lab.

Diodes aren't a good fix for this either. 
- The buffer/mux IC already has Schottky junctions at the input (this is why the max input is Vdd+0.3V). To offer proper protection, you'd need a diode that turned on comfortably before the diodes in the clock buffer. So, e.g. a diode to 3V3 would not be a good choice.
- We will often want to use signals that are only a few dB from the max allowed value to ensure lowest noise (at a 100MHz a large signal is needed to get a decently fast rising edge if the source is sinusoidal). At that point, the protection diodes would not be properly reverse biased and would be acting as a large non-linear capacitor.
- For ESD you need TVS which are also horrible as @gkasprow points out.

In the current design, the SMA outer shells are connected to circuit ground. So, as soon as one touches the SMA one grounds oneself to case ground. This provides good ESD protection (unless you manage to touch the center conductor without touching the case first). Even if we float the SMAs, we'd still leave 100k to circuit ground, which discharges ESD reasonably well.

hartytp, 2017-12-15:
> replace 100nF on the clock patchs (C196 etc, there are many). That 100nF are not really good parts for >tens MHZ signals. 10nF RF class parts shall be more appropriate.

a-shafir, 2017-12-15:
> termination can make EMI worse for the LVCMOS lines due to increased current flowing.

@hartytp that's why i was not even looking 1st time for the parallel termination on the board. In our case the main concern is EMI to the DDS signal output. And i noticed there is no source termination/debouncing resistor on the digital lines. The serial resistors on source certainly decreasing EMI and improving signal integrity for the signals over 10MHz. For most of the FPGA it is not a concern since it can be enable as part of the i/o cells. But Urukul CPLD does not have this feature.

hartytp, 2017-12-15:
> The serial resistors on source certainly decreasing EMI and improving signal integrity for the signals over 10MHz.

Not always. If the paths are short then reflections from unterminated LVDS lines aren't a problem at all. 

Anyway, as I said, you have the boards, you can easily test to see if this is a problem. Once we have the data we can have a more informed discussion about this.

a-shafir, 2017-12-15:
@hartytp it will be certainly overshot to add the resistors from the lvds receivers to cpld because it short, probably except IO_A0_SCKI.
But the DDS and especially ATT control lines are not short, up to 200mm

gkasprow, 2017-12-15:
Guys, let's estimate LVCMOS output resistance.
The [datasheet ](https://www.xilinx.com/support/documentation/data_sheets/ds093.pdf), page 4, says that V oh = Vccio - 0.4 at -8mA and Vccio-0.2 at -0.1mA.
So very rough estimate gives dV/dI = 0.2V/7.9mA which gives already 25Ohms.
For low level the value is the same. So it seems that internal buffer resistance already does the job.

a-shafir, 2017-12-15:
I will check with the scope both in slow and fast slew mode and will let you know.

a-shafir, 2017-12-15:
@gkasprow have you checked 50R traces? It is usually 17mil on a the standard stack-up but it looks narrower.

gkasprow, 2017-12-15:
@a-shafir I don't know what the standard stack-up is. It depends on manufacturer. In case of Urukul, I get 50Ohm with 10 mils traces on top/bottom.
On L3 and L4 one gets 54 Ohms with 6mils traces.

hartytp, 2017-12-15:
> @gkasprow have you checked 50R traces? It is usually 17mil on a the standard stack-up but it looks narrower.

https://en.wikipedia.org/wiki/Teaching_grandmother_to_suck_eggs

a-shafir, 2017-12-15:
http://www.bitweenie.com/listings/standard-4-and-6-layer-pcb-stackups/

17 mils has advantage to be inline with the 0402 pads that is quite important for 2.4GHZ that i am familiar with. For this specific board it shall be the same.

hartytp, 2017-12-15:
@a-shafir Please don't just unilaterally add items to the to-do list at the top of this issue. That's not how we work here. Feel free to post ideas for discussion, but @gkasprow is generally the arbiter of what gets done, as we are aiming for professionally designed hardware here, not hobbyist designs. For things like input protection on the clock lines, as @gkasprow your idea doesn't work, so shouldn't be on the to do list.

dhslichter, 2017-12-15:
>http://www.bitweenie.com/listings/standard-4-and-6-layer-pcb-stackups/

From literally the first sentence of this article: "it’s a bit of a misnomer to call the stackups presented below “standard.”  ."  

In general, @a-shafir , we are all on the same team, working toward the same goal.  You are not the leader on the hardware design, @gkasprow is; in fact, you are the most recent arrival to the effort.  Please try to understand that a lot of very thorough work has gone into all of these designs before you arrived, and that jumping to the conclusion that things were somehow done wrong anytime something doesn't work as you expect it to is not only often incorrect, it also comes across as rude.  If you have questions, comments, suggestions, please feel free to post them in a way that is respectful of others' knowledge, contributions, and time.

a-shafir, 2017-12-15:
@hartytp i was not adding anything in the to-do list.

There are some principal issues that must be fixed and that are not confirmed so. In any case every change needs to be evaluated how important it is and what are the efforts.

For pure digital boards "works" can be enough. For analog/RF boards there is no bottom-line (except the physics and the core IC specs) and it can be "ever-improved". I certainly see that there are issues that are imho not optimal. The question is how important each. Case by case. And how constrained the change.

Normally there are some design patterns to follow even it looks overkill. For instance like adding protection to any exposed pins especially for the high-cost boards.  The professional measurement equipment (and btw also most of the mainstream civil h/w like laptops of phones) has huge multi-level protection per every exposed signal. It is not s/w, adding some features later can be very problematic.

I am familiar with the ASIC design and can say that there are "ten times" more issues to avoid since the design iteration cost is enormous indeed. 

In this case i understand the constrain on the RF input to add it. Still it can be done for instance with the diodes biased not to GND/3.3V but to narrower range. It costs very little, shall not affect the RF specs and certainly more safe than internal protection diode ring inside this specific clock driver. But even if will be no changes with the PCB (assuming all of the issues i see will not be important for the intended use it still possible at least add a label/warning to the panel that the connected signals shall be handled with care. This cost nothing and can prevent damaging of the boards by users.
 
I think that getting to some issues during the board testing is bit too late. Many things can be avoided on the design review/cross check stage.

My opinion that the system concept is very powerful and there a lot of efforts done already to have it widely used. So it makes sense gradually improve the user experience as possible.

I am not a hobbyist. I am designing h/w for over 27 years professionally.

hartytp, 2017-12-15:
> For pure digital boards "works" can be enough. For analog/RF boards there is no bottom-line (except the physics and the core IC specs) and it can be "ever-improved". I certainly see that there are [...]

Right, we do get the basic idea of analog design.

But, I'm not convinced that the suggestions you've put forward will actually make anything better. e.g. using switching to RF capacitors won't reduce the VSWR of the clock paths noticeably, so why would we do it.

If you can find ways to improve things then great! But, I want data -- be it from datasheets, measurements or simulations -- to back your ideas up with.

> In this case i understand the constrain on the RF input to add it. Still it can be done for instance with the diodes biased not to GND/3.3V but to narrower range. It costs very little, shall not [...]

I disagree with that statement. For optimal noise performance these clock buffers need a high input slew rate, so one typically drives them with the highest input possible. e.g. +10dBm to +13dBm from a commercial low-noise oscillator is typical. That's 2Vpk-pk to 3Vpk-pk. Any diodes you add to that will not be reverse biased, and will cause serious issues.

IIRC, most commercial equipment does not rely on diodes for this kind of thing. Switches controlled by power detectors are quite a standard protection method. But, that's an unreasonable level of complexity in our case. There just isn't a really easy way of doing this here, which is why we haven't done it.

> But even if will be no changes with the PCB (assuming all of the issues i see will not be important for the intended use it still possible at least add a label/warning to the panel that the [...]

Sure, I think there was an issue to label max input levels on the schematic at some point. No one bothered to follow it up.

But, again, if anyone is daft enough to put >+15dBm into one of these buffers then they shouldn't be using it. If you're that daft, you've probably already broken more expensive pieces of lab equipment and have bigger problems.

gkasprow, 2017-12-15:
@a-shafir you cannot route 17mils traces because there is no way to get them out from FPGA. LVDS lines done with 17 mils traces would simply not fit between pads or balls and such design would be not practical.
Believe me, I use professional grade signal integrity and power integrity tools to validate every design, where I have even slightest doubts. RF rules very often don't have a meaning at lower frequencies because are simply impractical. Look at the Sayma RTM design. It works at 12Gbit and took some time to optimise it using Hyperlynx and RF models of critical parts.
About protection of professional equipment it is not true. High-end scopes and analyzers don't have protection at the inputs because such protection would degrade the signal. And here is that case.
I am aware of various protection mechanisms and use components when necessary or practical.
There are TVS circuits, CMCs, filters on various interfaces.
Look at the schematics of Sayma or Kasli. User-exposed interfaces are protected because it was simply practical and did not degrade the signal. 
I'm open to discussion and I don't have a problem to say I was wrong or that you are right. I simply don't like the way you do.
I designed several industry-grade devices, some of my stuff is in space, some in LHC, tokamaks, one project even reached Mars. I don't have 27 year experience as you, but I know that sometimes I'm wrong and learn sth new with almost every design.
Since you are RF guy, your experience is very valuable here. Please look at RF parts of designs, they are done in completely different way, simply because it serves its purpose.

a-shafir, 2017-12-15:
My scope is not so good so i am attaching few images to see the difference:
3U DDS rev 1.0

1. Pad R40D-2 (DDS_CLKI3 termination)
[image]

2. Pad IC3D-24 (attenuator3 serial input, 5V levels from the attenuator2)
[image]

3. Pad IC3D-21 (ATT_CLK) when CPLD output slew rate set to "SLOW"
[image]

3. Pad IC3D-21 (ATT_CLK) when CPLD output slew rate set to "FAST"
[image]

My opinion is following:
- the attenuator does not have a separate ground domain so it will pick-up some interference from the digital inputs as well from the SerialOutput (cap load, reflection)
- The best looking signal is "2" and there is a short line only between IC3C and IC3D.
- There is no major difference for the bouncing between the FAST and SLOW slew rates.
- The signal can certainly look better.
- The DDS_CLKI3 (on the terminator) still has some spikes comparing to the short line.
- I noticed that ATT_S_OUT is 5V signal but it connected to the CPLD directly. According [xapp429] (https://www.xilinx.com/support/documentation/application_notes/xapp429.pdf) it is incorrect.

Unfortunately i can't test with the current equipment the actual interference between the digital lines and the analog outputs. But i suspect it will be important in some cases, especially when the DDS output signal is not max but controlled with the amplitude modulator and also in case of the higher attenuation by the attenuator IC.

The proposed improvement to the issue:
1. Add source termination resistors near to CPLD to the long digital traces as possible in order to reduce the ringing but to keep the signal integrity on the necessary level.
2. Add resistors close to the IC3A,B,C serial outputs to "eliminate" the interference from even the short line and one input connected.
3. Remove ATT_S_OUT trace to the CPLD since there is no practical use of it. And if it needs to check the attenuator spi data path add a source termination and handle the 5V tolerance issue according the Xilinx.

Note: the return ground of the probe is as good - 3 short ground clips in parallel.

gkasprow, 2017-12-15:
@a-shafir What is BW of the scope and the probe?
Please publish the screenshot of the scope probe calibration generator.
Such reflections may be caused by the limited BW of the probe.
The CPLD has already 25Ohm series resistance. I'd rather add parallel AC resistance. It's easier to add AC termination instead of additional series resistor.
ATT_S_OUT is useful during ATE tests. It's the only way to check if attenuator is connected to the CPLD.
Good catch with 5V output. So we will keep it and simply add limiting resistor. The same applies to BaseMod design.

hartytp, 2017-12-15:
@a-shafir So far, those scope traces don't actually show a problem that should be fixed (other than the 5V signal going to the CPLD).

Before we start proposing changes to the design, I'd like to see evidence that there is a problem here, and that a different termination method will improve it. Again, we have the pro-type boards, so we can test these hypotheses.

For example, please put the DDS RF on a spectrum analyser, first withouth the attenuator, and then with the attenuator constantly updating. Let's see what happens to the SFDR/noise floor/etc. If there is a significant affect, then feel free to try different termination methods. If you can find one that improves things then great!

But, until you've done that, please stop proposing solutions to things that aren't necessarily problems.

hartytp, 2017-12-15:
This issue is generating a lot of unnecessary noise and wasting time.

With things like this, you just need data. e.g. the 100nF issue, you really could have googled the capacitor part number and looked at the S-parameters the manufacturer provided. From those, you can tell whether there is actually a problem, rather than speculating and proposing changes. AFAICT, that data shows there is no problem.

Also, this thread is getting too long. Please open a new issue for termination if you still feel strongly about it. But, again, please get some RF spectra that show a genuine noise issue first.

a-shafir, 2017-12-15:
@gkasprow the probe used is PVP2150 1:10 mode that claimed 150MHz. I checked it with the scope calibrator and it looks it RF matched correctly. I have more worry about it input capacitance that can affect this measurements.
The internal scope calibration signal is slow. I suggest you refer to the difference between the images 2 and 3. Or better to check on your board with high-quality probe/scope.
I am not sure that the CPLD has indeed 25R resistance. The specs you referred earlier for me looks more like worst case (in the production range + the temperature rtange). But it can be way lower. This is not replacement of a termination techniques imho.
I think the termination/ringing-dumping resistors can be (ideally) not small. For instance for the IC3 serial data chain it can be even 1K to pass the signal safely but minimize the captive load. ATT_S_OUT not need much bandwidth as well. Note: DDS SPI CLK palnned to be up to 66MHz for the NU-mode.

@hartytp ok, i will not post such issues anymore if it not welcome. But i still think that all the "live" digital lines on the RF-style analog boards must be ring-dumped at source just a good design pattern at least. I am unable to check it now. But i am pretty sure that there will be some noise exposed. 

Again (the _last_ time) _IMHO_ that "fixing the detected issues on the produced prototypes" is not the best design approach. There are many issues that can't be detected at all (like out-specs than needs time and conditions to pop-up) and there are many other issues that need specially planned tests and can appear in different usage scenarios etc. My opinion that doing design reviews and cross checks and adding "safe" features like extra resistors etc will practically draw no attention and will save all team time on handling issues dramatically even (and especially) on proto-level. Also for small h/w topics and ideas can be created a separate tread (not issues).

hartytp, 2017-12-16:
> @hartytp ok, i will not post such issues anymore if it not welcome.

Genuine issues welcome. But, the signal to noise ratio on your posts is not very high at the moment and it wastes time. e.g. look at capacitor datasheets before deciding that they're not good enough; etc.

> Again (the last time) IMHO that "fixing the detected issues on the produced prototypes" is not the best design approach. There are many issues that can't be detected at all (like out-specs than [...]

In most cases I'd agree with you. But:
- The signals you posted look fine for this kind of thing.
- if there is genuinely a noise issue from these digital lines then it will show up repeatably on the RF output. If you can't measure it easily with a spectrum analyzer then it's not a problem. This isn't going to manifest as a spooky intermittent issue.
- If you do measure something then we can try out different termination schemes and find the most suitable one for this design. Until you do this it's all speculation. Data makes these design decisions much quicker.

> But i am pretty sure that there will be some noise exposed.

Post once you find it and I'll be interested. We all need our intuitions re calibrated sometimes. As I said, if you can find a genuine problem and demonstrate a fix for it then great! That'd be a big help. There's just not point having long discussions without data.

> I think the termination/ringing-dumping resistors can be (ideally) not small. For instance for the IC3 serial data chain it can be even 1K to pass the signal safely but minimize the captive load. ATT_S_OUT not need much bandwidth as well. Note: DDS SPI CLK palnned to be up to 66MHz for the NU-mode.

Having said all that, if you want to add an AC load termination of a couple of hundred Ohms or more then go for it. It's unlikely to help, but won't do any harm.

hartytp, 2017-12-16:
@a-shafir One final comment about all of this: in any of these communities there is an expectation that people will do some basic legwork before posting issues.

For example, before posting issues about strange behaviour in the inputs, one is expected to look at the signals one is applying with a scope and verify that they meet the requirements of the input driver (that would have saved quite a bit of time in #422).

Similarly, before asking whether the traces labelled 50Ohm are actually 50Ohm, why not check yourself? You have the design files and can find the stackup, from which the trace width calculation is trivial. 

Before commenting that parts aren't suitable for RF, look up the part number and look for relevant data sheets.

etc etc

A lot of work went into the design of these boards and while they certainly will have bugs -- the 5V signal applied to the CPLD is one such bug that you did find, which represents a welcomed contribution to the project -- and things that can be improved, it's frankly a bit insulting posting a bunch of issues without bothering to do even that basic level of work first. The more of these spurious issues you post, the harder it is to separate your genuinely useful contributions from the noise.

a-shafir, 2017-12-16:
OK, I see the way to redefine my concerns more precisely. It is still on empirical level but more to the point:
- we are talking about the signals like the reference clock that planned to be around 120mhz.
- the sew rate and the phase noise is critical and there are already efforts to handle it by skipping the protection components etc.
- there are some DDS synchronization signals that need precise alligments. For instance there is a propagation delay compensation in the logic.
- if I assume we need 1ns edge class signal it roughly equivalent to 1Ghz bandwidth-like handling. It is certainly by a power over 100mhz scale issue.
- there are multiply sources of noise on the board. For instance many of the "cold" outputs can turn at the same time directly coupled tho the non-neglectable capacitance traces and this together with undamped reflections will generate quite unpredictable spurrious noise across the board.
- the clock and sync lines does not have self clock recovery features like the FPGA serdes multy-ghz buses so we need to ensure really good signal integrity.
- there are some imperfections on the clock and sync signal traces that will add some reflections etc.
- all that factors will be combined and affect the signal integrity and the edge quality on the traces.
- there are no easy way of testing it. So the actual problrm likely will appear very late so it will need a lot of efforts to handle.

Therefore in order to keep all this under control I suggest to follow some stricter rules. Consider the high quality signals bandwidth based on the desired edges. In this case if we need a 1ns class edge the serial capacitors shall be RF class with the proper specs for 1ghz and higher. That is not a problem - there are many cheap 2.4ghz rf class parts but of course typically not 100nf. Also try to avoid the reflections on the critical lines by handling and simulating it as GHz class or according to the desired edge quality. And since the board is very dense and there is no space typically to avoid interference try to keep the sources of the interferences low. For instance by terminating and dumping all, even humblies digital lines etc.

Note: this is all about the digital signals, the analog interference is out of the scope.

hartytp, 2017-12-16:
> In this case if we need a 1ns class edge the serial capacitors shall be RF class with the proper specs for 1ghz and higher.

Did you bother to look up the capacitors we are using and see their S-paraeters? The manufacturer specifies them up to 6GHz. Just look at the parts we're actually using and do the maths.

hartytp, 2017-12-16:
Don't give general comments like this when you can have data. e.g. you can calculate the VSWR introduced by these capacitors easily. Will that be enough to degrade the performance noticeably?

Moreover, you can compare this to the VSWR due, for example, to the capacitance of the relevant input pin. What limits the overall VSWR of this line? If the coupling capacitor isn't close to being the problem then don't use a fancy part when a cheaper one will do.

hartytp, 2017-12-16:
> the clock and sync lines does not have self clock recovery features like the FPGA serdes multy-ghz buses so we need to ensure really good signal integrity.

The sync signal just needs to meet timing with respect to the clock. You can see that with a scope. If you're worried, give us a measured eye diagram. If that's got plenty of margin, very low-level EMI won't be an issue -- and again, you can test that by measuring the effects of the various components. If we're not marginal then this just won't be an issue.

The clock does need to be very good. But, any noise on that comes directly to the output so, you can see this clearly with a spectrum analyser. Just get some data on this if you think it's an issue.

> there are no easy way of testing it. So the actual problrm likely will appear very late so it will need a lot of efforts to handle.

That's just not true. The noise due to the EMI sources you're talking about generally add linearly on the DAC's output, so you can test them one by one.

Let's not descend into the "electronics is voodoo" design methodology.

hartytp, 2017-12-16:
Sometimes the simplest design is the best one. Adding more/fancier components that do not create a noticeable performance difference is bad design.

For things like the sync signal, the question is what the eye diagram looks like and how much margin we have. This *is* digital, and if it comfortably meets timing then there is no concern. Weird intermittent issues only show up when timing is marginal or when there are several large EMI sources that can add up to give a problem. But, that's all not hard to test. If all EMI sources are measured to be small, and the eye has good margin then there won't be problems.

With regard to the clock and similar, the question isn't "does this add noise" but, "how much noise does this add?" For broadband sources, is it above the DDS's noise floor? For narrowband sources, how does this affect the SFDR of the DDS? If you can't measure the noise of something then it's just not a problem.

So, please, just give us data rather than philosophy.

a-shafir, 2017-12-16:
OK, I got your point for the capacitor specs. Replacing capacitor is easy. My major concern is the cpld outputs. When many of them are driven from the same clock domain and the outputs are switching simultaneously skipping tbe source termination resisors will lead to too sharp edges of the signals and a lot of noise across the board. I would like to be wrong indeed and don't like at all to show a poor eye diagram or the spectrum analyser to just proof my concerns.

hartytp, 2017-12-16:
> OK, I got your point for the capacitor specs. Replacing capacitor is easy. 

Not as easy as not replacing a capacitor.

These boards are designed to be price optimised (cf #347). Needlessly adding extra BOM lines and specialist capacitors inflates the price. *If you can provide any data that suggests that different capacitors will improve the board performance noticeably then we will do it. Otherwise, it's just a bad idea.*

For the last time, just do the **** maths and justify your point.

> When many of them are driven from the same clock domain and the outputs are switching simultaneously skipping tbe source termination resisors will lead to too sharp edges of the signals and a lot of noise across the board

Put this in perspective. 

- Firstly, a lot of noise across the board? Compared to the SMPSs? Are you really saying that this will be a dominant noise source?
- Secondly, there is always EMI from these outputs. The increase in EMI is basically just the overshoot, which isn't that big even for the worst signals you showed. Worst case, it's going to be a few dB increase in the EMI from these signals. However, if these signals are a few dB away from being a serious problem EMI-wise then we need a major re-think of the design. If not then we don't need to worry about it.

In any case, this conversation is a waste of everyone's time. Please get some data to back up your points or I'm just not interested.

hartytp, 2017-12-16:
> Not as easy as not replacing a capacitor.

> These boards are designed to be price optimised (cf #347). Needlessly adding extra BOM lines and specialist capacitors inflates the price. If you can provide any data that suggests that different [...]

> For the last time, just do the **** maths and justify your point.

And, hey, if you don't want/know how to do the maths then just replace the caps with fancier ones and post scope "before" and "after" scope traces which show the difference. If there is a noticeable improvement in signal integrity then we can change things. Just give us **something** to justify using fancier components.

a-shafir, 2017-12-16:
Ok. Let's forget the 100nf capacitor.
- We have on the board 4x 1 GSPS, 14-bits DACs as the signal source (integrated in the DDSs). There is >85dBc narrow band SFDR. If we would like to pass the signal of this quality to the target this alone will be a major challenge to the PCB designer.
- we have digital signals on the board. In worst (and target) case (NU-mode) there will be 66MHz QSPI bus constantly active. So there will be 66MHz clock and 4x 33MHz data lines active (also CS and other minor signals). And this including many harmonics related to the spikes is well inside of the DDS output signal filter (DC to 320MHz)
- We have single-ended outputs.

If i would design this board in order to attempt to keep the SFDR on the connectors as good as it is on the DDS DAC output i would all the put the balloon and the filter as close to the output connector as possible. Also I will make isolated ground islands there connected with the main ground only on the connectors pads. This will isolate the unbalanced signals from the parasitic ground currents etc. This is the ideal scenario.

- On this board the unbalanced signal path is longer. There is unbalanced attenuator and the 23dB gain amplifier after the shaping filter. We also have 9dB T-pad.

If i would do this board i would put that unbalanced circuits on a separate ground island each. I will try to keep this part of the circuit as compact as possible putting the RF components next to each other according the constrains. I will use lower gain amplifier and minimize the T-pad. I will inject the attenuator and amplifier power trough RF SMD instructors placed from the ground bridge side. I will use "clean" attenuator control lines (possible some buffering on the ground island entry) in order to avoid any injection of the noise from the digital domain to the output signal domain.

So the current PCB layout has done differently. I understand there are PCB size and development cycle constrains. So now it is sharing the same ground plane. Also there are some digital signals like ATTxx are traced to the sensitive area so will transfer some noise from the digital domain even when inactive to the attenuator digital pads and than, via the parasitic capacitance to the power pads and the signal path.

Of course the output signal will be not as clean as one on the DDS DAC pins directly. There will be some noise added. The question is it will be 0.1dB or 20dB peaks over the noise floor. Reaching the DDS noise floor in this case is very difficult indeed because of the dynamic range and the bandwidth. 

I can bet $1000 that if the board will be designed with all the design patterns mentioned there will be significantly better SN performance than the current setup.

Still keeping without the major changes to the current layout some improvements can be done:
1. To reduce the affect of the digital noise (including the current noise from the parallel termination) to the the outputs use serial termination at source. Data: according the stack of the board (T=0.058mm, H=0.18mm, W(digital traces) = 5mil, Er(1st layer) = 3.66) Z~=79.5R. The CPLD CMOS output impedance is certainly lower than this in scale. I am not sure if it will be safe to use 75R resistors for the 66MHz signals but we need to have the resistors placed on the board and check it. I have only one board now and not planning to cut a trace near the CPLD, add a resistor and show you with the scope the difference. There are tons of the references in the net about importance of the serial termination for the digital lines even from the view of the basic SPI signal integrity, forgetting our board SFDR specs etc.

2. Add balloon on the CLK input. Or ideally a clock buffer since the balloon itself will not prevent reflection of the clock to the source cable that can really make things bad. According my calculations there will be significant reflection on the junction between the single ended 50R connector side and the differential traces and other components on the "long" trace to the SI clock chip. 

3. Move the internal clock input connector as close to the clock distribution IC as possible (there is enough space).

All that changes are very minor will not increase the BOM/cost significantly (in fraction of a %) but will significantly increase the board analog performance.

Sorry if it sounds philosophy or rude/insulting. This is based on my background and experience in measurement h/w, rf h/w, audio h/w and DACs etc. IMHO it written to the point and technically grounded.

References:
http://www.the-signal-and-power-integrity-institute.com/files/Andrew_Norte_Source_Termination_Resistor_Location_2.pdf
http://www.ti.com/lit/an/slyt499/slyt499.pdf
etc

gkasprow, 2017-12-16:
@a-shafir please do some measurements and then prove that there is something to fix.
I asked you to post the probe calibration waveform, frequency does not matter, scope calibration output has usually very sharp edges. Zoom it to see how the signal looks like with your probe.
Do it please, then we can discuss. Otherwise further discussion is pointless.
Indeed, common ground island can be added to the RF chain and I will do it in next revision providing that the prototype performance is not sufficient. One can easily more harm than fix because return current path gets broken with such islands.
Instead of talking about whether I am right or wrong about CPLD output resistance, just measure it!
Connect 100Ohm to GND  and see how the voltage gets divided. If I'm wrong, write about your discovery.
Every circuit can be done better, every trace can be routed manually following strict RF rules. But then the board would be routed in 3 months, not  2..3 days as it was done, which would significantly increase the cost and number of layers.

a-shafir, 2017-12-16:
@gkasprow after better checking the RF of the board mu main concern now is the potential distortion in the parts after the DDS DAC. Mostly the amplifier IC and the other parts as well. I am not sure about the target application - how sensitive it for the signal quality and the noises. For a general use of the DDS with the best (limited by the DDS IC) specs approach similar to the AD reference design PCB (a balloon and the filter ) is optimal. Adding the ground islands next to the connectors is trivial.
To test all of this (distortion/harmonics of the signal, the interference) need a spectrum analyzer with good specs. Currently i have no access to it.
I think quite soon we will have some test results if not from me than from the another labs that have Urukul samples.
Right now it makes sense to focus on Kasli.
I don't think that need 3 months even for the complete redesign the board. But for 2-3 days the result is very impressive indeed! 
Any way, need to do some tests. If my concerns are valuable i can help with the layout and parts, again it is not that tricky board imho - there is no really custom design analog circuits etc.

hartytp, 2017-12-16:
> One can easily more harm than fix because return current path gets broken with such islands.

Right Greg, this is the key point that applies to many of these issues. We have prototypes and until/unless we find actual performance issues with them, we shouldn't mess with things like ground islands, differential RF amplifiers, etc. 

> I asked you to post the probe calibration waveform, frequency does not matter, scope calibration output has usually very sharp edges. Zoom it to see how the signal looks like with your probe.
Do it please, then we can discuss. Otherwise further discussion is pointless.

You're talking about those scope traces he posted? They looked fine to me. There is a bit of overshoot there, but nothing that's worse than I'd expect, and nothing that is likely to significantly degrade the performance of the design. 

There is always overshoot on these LVCMOS control lines. It can always be made lower, but there usually is no point -- and as @gkasprow pointed out, 50R terminating LVCMOS increases the current flow in the ground plane and can lead to worse noise, even if the ringing is reduced.

My view is that until I see a spectrum analyser trace showing a degradation of the RF output quality due to digital cross talk, and another one showing an improvement with different termination/grounding/etc there is nothing to discuss here.

---- 

Also @a-shafir, as @jordens has pointed out many times before, once you do have data, please open a GitHub issue for each issue you find with the circuit. These long discussion threads are hard to follow and less useful than separate issues for each item.

hartytp, 2017-12-16:
Anyway, I'm done with this conversation until some data showing an actual issue is posted. 

However, I'll leave you with my specialised RF balloon, which I think should be added to the design to reduce noise. Full S-parameters to 20GHz on request (not suitable for operation on Mars though Greg). @a-shafir since you have 27years of experience as an RF designer, I assume this is what you were talking about, and it wasn't just a typo for balun:

[image]

[37 more replies]

commit 05988c7a (2018-01-11): fixed #377
closed 2018-01-12 by gkasprow