sinara-hw/sinara #378: NAT MCH + Sayma_RTM -> MCH shutdown

jbqubit, 2017-11-21:
configuration
- chassis: NATIVE-R5
- power slot: NAT-PM-AC600
- MCH: NAT V2.18 Final
- AMC1: Sayma_AMC
- AMC2: Sayma_AMC
- RTM1: Sayma_RTM

Via serial interface restore default MCH configuration. 
```mchcfg  -> [2] reset to defaults```

Via web interface delete user-defined flash-based Startup configuration. 
```Load/Delete Configurations```

All the AMCs power up. MCH's AMC LEDs are solid red as expected. There are no other error indications on MCH. 

-----

Using the web interface Overwrite the Startup configuration with the following.
```
fw_ver = "V2.18"
mch_id = 0
mez_id = 0
ins_id = 0
amc_pwr_on = 5, 65, 3
amc_pwr_on = 6, 65, 6
wd_exception_reset = 0
wd_task_lost_reset = 0
wd_shutdown_after_reset = 0
```
Cycle the power on the crate power controller. Watch as MCH boots, powers on ACM1, pauses then tries to power on AMC2. At this point the MCH blue LED starts to flash and it no longer responds to serial or http requests. 

----

To recover do as follows. Eject the Sayma_RTM in slot RTM1. Cycle crate power.  Watch as MCH boots and AMC1 and AMC2 start up fine.  MCH's AMC LEDs flash green... OK. Can get AMC current draw using ```show_sensorinfo 50``` via MCH serial interface. OK. Via web interface delete user-defined flash-based Startup configuration:```Load/Delete Configurations``` Power crate off. Insert the Sayma_RTM in slot RTM1. Power crate on. All the AMCs power up. MCH's AMC LEDs are solid red as expected. There are no other error indications on MCH. 

----

I've reproduced the above cycle twice. 

-----

Why does Sayma_RTM interfere with MCH? 

It looks like the MCH configuration file has tons of undocumented parameters (eg related to Ethernet). This is looking fiddly enough I'd rather the developers try to reproduce this as time permits instead of giving me instructions on how to debug the problem in the field. AFACT what I observe doesn't interfere with Sayma testing but does interfere with using IPMI to readout housekeeping data.

jbqubit, 2017-11-21:
@gkasprow This isn't a high priority bug relative to the ongoing 3U related work.

gkasprow, 2017-11-21:
@jbqubit which MCH model do you have? We will try to recreate this problem, hopefully with fully working MMC it will disappear.

jbqubit, 2017-11-21:
MCH-Basic v3.5 Rel 150115

vdirksen, 2017-11-21:
Do not worry about all the Ethernet configuration. All of them are disabled. See the script file just as a template. It is only important to generate the template on that MCH, where you want to run your configuration via Script file.

closed 2018-01-29 by jbqubit