sinara-hw/sinara #379: Sayma_RTM LED LD7 P1V2 not illuminated

jbqubit, 2017-11-22:
- Sayma_AMC + Sayma_RTM on benchtop (no backplane)
- +12V bench top power supply that can source 8.1 A 
- all power LEDS on Sayma_AMC are green
- LEDs on Sayma_RTM are green except 
    - LD7 P1V2 is dark (not expected) 
    - LD15 OVTMP is dark (as expected) 

Also, current on my power supply is fluctuating between 2.79 A and 2.81 A which seems funny.

jbqubit, 2017-11-22:
P1V2 test point is 1.2 V. So looks to be a problem with LED.

gkasprow, 2017-11-22:
LED, driving transistor.
The RC components are very fragile, they go off easily i.e. if you hit the
board with some sharp edge, tool, etc.

gkasprow, 2017-11-22:
@jbqubit please check if all four components that are selected on screenshot below are mounted.
It's quite possible that one of them got damaged during Allaki removal.
If they are OK then measure the voltage across the LD7.
[image]

jbqubit, 2017-11-22:
All the ICs with thin yellow boxes in your screen shot are present on the PCB. Relative to ground the potentials on either side of LD7 are 10.4 V and 11.8 V.

gkasprow, 2017-11-23:
That's bizarre. The voltage is correct and LED should be on. Maybe light in your lab is too bright and U don't see it :)

jbqubit, 2017-11-24:
Checked with lights off. It's just not illuminated. @sbourdeauducq if you don't see this on any of your boards I'll close the issue. It may be that my LED is just broken.

sbourdeauducq, 2017-11-27:
LED is on with both boards here.

jbqubit, 2017-11-27:
OK. This may just be a bad LED on my board. Nothing to worry about. Closing.

closed 2017-11-27 by jbqubit