sinara-hw/sinara #380: Sayma_AMC: PB3 SW (front panel button)

jbqubit, 2017-11-22:
The PB3 SW is routed to the MMC (LPC1776). Currently @gkasprow reports that it does nothing. What should its default behavior be? @sbourdeauducq 
1) MMC: reset, reconfigure from flash
2) AMC FPGA: reset, reconfigure from flash
3) RTM FPGA: reset, reconfigure from AMC FPGA

sbourdeauducq, 2017-11-23:
Can't really see why we'd need a button there... maybe route to FPGA somehow and let users access it in ARTIQ kernels.

gkasprow, 2017-11-23:
Well, I placed it for historical reasons - accelerator guys wanted to have one. We can always not mount it or remove in rev2

jbqubit, 2017-11-24:
I'd like to use it for resetting the board -- something like outlined above. Seems like I use the reset button on the KC705 a couple times a day while debugging things. Its helpful to have this option without needing to attach a USB cable to the front. 

Longer term it would be nice if....
- short-press reload user-supplied .bit files
- medium-press load .bit files from DRTIO
- long-press reload golden-master configuration files -- for MMC, power chips, FPGAs

closed 2018-03-12 by jbqubit

marmeladapk, 2018-12-13:
@jbqubit Should this remain in milestone? @gkasprow does it require modifications to the AMC?

gkasprow, 2018-12-13:
no, just implementation in MMC

jbqubit, 2018-12-13:
I'd like to leave the button on the layout and connected to the MMC. Let's make the default behavior fully resetting Sayma. AFAICT this would consist of the following steps.
- soft power off of AMC+RTM LDOs
- reset MMC
- reconfigure MMC from flash
- trigger power-on of AMC+RTM LDOs