sinara-hw/sinara #382: Sayma_AMC: startup messages on UART

jbqubit, 2017-11-22:
I don't see ARTIQ startup messages on any of the four UARTs exposed via the front panel USB on Sayma_AMC. I do see some MMC messages. 

```
$ flterm /dev/ttyUSB4
Started. CPU clock = 80 MHz, SDRAM clock = 40 Mhz
LWIP HTTP Web Server FreeRTOS Demo...
Waiting for TCPIP thread to initialize...
Starting LWIP HTTP server...
Net up
```

sbourdeauducq, 2017-11-23:
The flash is also buggy (see https://github.com/m-labs/artiq/issues/846 and https://github.com/m-labs/artiq/issues/847) so 1) the bitstreams won't load 2) the BIOS cannot be executed from there. Build ARTIQ AMC with "--integrated-rom-size 36864" and load the bitstreams using JTAG direct to SRAM (you need to write your own OpenOCD script) to work around this.

jbqubit, 2017-11-24:
I don't know: how to use the proxy bit stream to target SRAM, memory address or how to trigger loading of BIOS from SRAM. Please update master to work around these gaps until a resolution is found. I'd like to be able to provide feedback on the parts that work.

sbourdeauducq, 2017-11-24:
No proxy bitstream is needed to target SRAM. Load the ARTIQ bitstream instead of the proxy bitstream using a ``pld load`` command. Here are some quick OpenOCD scripts I am using:

sayma_new.cfg:
```
interface ftdi
ftdi_device_desc "Quad RS232-HS"
ftdi_vid_pid 0x0403 0x6011
ftdi_channel 0
# EN_USB_JTAG on ADBUS7: out, high
# nTRST on ADBUS4: out, high, but R46 is DNP
ftdi_layout_init 0x0098 0x008b
reset_config none

adapter_khz 5000
transport select jtag

source [find cpld/xilinx-xc7.cfg]
set CHIP XCKU040
source [find cpld/xilinx-xcu.cfg]

init
```
load AMC:
``openocd -f sayma_new.cfg -c "pld load 1 artiq_drtio/misoc_saymaamcstandalone_sayma_amc/gateware/top.bit; exit"``

load RTM:
``openocd -f sayma_new.cfg -c "pld load 0 artiq_drtio/artiq_sayma_rtm/top.bit; exit"``

jbqubit, 2017-12-04:
```
$ openocd -f artiq_sram_load.cfg -c "pld load 0 artiq_sayma_rtm/top.bit; exit" 

Open On-Chip Debugger 0.10.0-00007-g977d9e4 (2017-10-23-23:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362e093 (mfg: 0x049 (Xilinx), part: 0x362e, ver: 0x0)
Info : JTAG tap: xcu.tap tap/device found: 0x13822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x1)
Warn : gdb services need one or more targets defined
loaded file artiq_sayma_rtm/top.bit to pld device 0 in 3s 592253us

$ openocd -f artiq_sram_load.cfg -c "pld load 1 misoc_saymaamcstandalone_sayma_amc/gateware/top.bit; exit" 

Open On-Chip Debugger 0.10.0-00007-g977d9e4 (2017-10-23-23:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362e093 (mfg: 0x049 (Xilinx), part: 0x362e, ver: 0x0)
Info : JTAG tap: xcu.tap tap/device found: 0x13822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x1)
Warn : gdb services need one or more targets defined
loaded file misoc_saymaamcstandalone_sayma_amc/gateware/top.bit to pld device 1 in 9s 680822us
```

```
$ flterm /dev/ttyUSB2

MiSoC BIOS
(c) Copyright 2007-2017 M-Labs Limited
Built Dec  1 2017 21:23:43

BIOS CRC passed (36d09528)
Initializing SDRAM...
Write leveling: 385* 27   8  29   4  20  376* 377* completed
Read bitslip: 7 1 0 
Read delays: 7:512-520  6:08-187  5:60-234  4:75-248  3:104-270  2:111-280  1:512-520  0:512-520  completed
Memtest failed: 527361/532736 words incorrect
Memory initialization failed
```
```
BIOS> memtest
Memtest failed: 527363/532736 words incorrect
```
Is this the expected result?

enjoy-digital, 2017-12-04:
@jbqubit: no sdram is failing. 
Are you generating sayma_amc bitstream yourself? If so, would you mind generating it by changing this: 
https://github.com/m-labs/misoc/blob/master/misoc/targets/sayma_amc.py#L49
to 90 or 135?
I was thinking issue was resolved for all boards but it does not seem to be the case. I'll have to look at it again. Would it be possible to have remote access to your board?

gkasprow, 2017-12-04:
@enjoy-digital which SDRAM do you use? 32bit or 64bit?

enjoy-digital, 2017-12-04:
@gkasprow: here it's the 64bit. We have tested successfully the 32bit and 64bit on the board i have and on m-labs' ones. I think @jbqubit one would work by adjusting dqs phase, but i need to have another look at that to have a single bistream working for all boards.

sbourdeauducq, 2017-12-06:
@enjoy-digital I have given you access to Joe's computer. Details by email.

jbqubit, 2017-12-06:
@enjoy-digital Any luck?

enjoy-digital, 2017-12-07:
@jbqubit: sorry not at the office. I'm back tomorrow and will test it.

enjoy-digital, 2017-12-08:
@jbqubit: i haven't been able to test on your setup, but it should be fixed. I sent you a bistream by mail for testing.

In case it's still failing, a temporary workaround is to comment the write leveling:
https://github.com/m-labs/misoc/blob/master/misoc/software/bios/sdram.c#L431-L432

jbqubit, 2017-12-08:
@enjoy-digital sent me a .bit with integrated bios for testing. Here's what I see when flashing.

```
$ openocd -f ~/artiq_sram_load.cfg -c "pld load 1 artiq-dev/top.bit; exit"
Open On-Chip Debugger 0.10.0-00007-g977d9e4 (2017-10-23-23:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362e093 (mfg: 0x049 (Xilinx), part: 0x362e, ver: 0x0)
Info : JTAG tap: xcu.tap tap/device found: 0x13822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x1)
Warn : gdb services need one or more targets defined
loaded file artiq-dev/top.bit to pld device 1 in 26s 199644us
```

```
$ flterm /dev/ttyUSB2

LiteX SoC BIOS (lm32)
(c) Copyright 2012-2017 Enjoy-Digital
(c) Copyright 2007-2017 M-Labs Limited
Built Dec  8 2017 14:26:30

BIOS CRC passed (4d1e2f11)
Initializing SDRAM...
Write leveling: 37  64  53  74  67  82  48  55  completed
Read delays: 7:08-169  6:22-192  5:71-242  4:85-253  3:110-281  2:118-300  1:143-319  0:152-327  completed
Memtest OK
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found
BIOS> 
BIOS> memtest
Memtest OK
```

enjoy-digital, 2017-12-08:
@jbqubit: ok thanks for the result. Then you can test with Artiq, the issue should be fixed.

jbqubit, 2017-12-11:
Confirmed Memtest is OK for .bit build from ARTIQ master. I'm continuing to bypass use of Flash. 

```
$ flterm /dev/ttyUSB2

MiSoC BIOS
(c) Copyright 2007-2017 M-Labs Limited
Built Dec 11 2017 12:41:26

BIOS CRC passed (7cea5d41)
Initializing SDRAM...
Write leveling: 41  74  55  76  54  71  35  40  completed
Read delays: 7:00-163  6:08-188  5:61-235  4:72-244  3:104-268  2:111-282  1:132-299  0:138-302  completed
Memtest OK
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from network...
Local IP : 192.168.1.50
Remote IP: 192.168.1.100
Unable to download boot.bin over TFTP
Network boot failed
No boot medium found
```

closed 2017-12-11 by jbqubit