sinara-hw/sinara #383: kasli pins

jordens, 2017-11-23:
There are inconsistencies between schematics and pinout. In the [pinout](https://github.com/m-labs/sinara/releases/download/Kasli%2Fv1.0/pins_Kasli_v1.0.txt) you have DDR_A0 on M5. In the [schematics](https://github.com/m-labs/sinara/releases/download/Kasli%2Fv1.0/KASLI_v1.0.PDF) you have it on L6. Other pins are also wrong. Could you fix that?

marmeladapk, 2017-11-23:
Thanks for noticing, I updated the wrong file.

>Is there no export format that blindly exports the entire netlist?

There is this format:
```
Designator	Pin Name	Net	Bus	Type	Part No.	Pin Group
A1	IO_L1N_T0_AD4N_35	DDR3_DQ13		I/O	6	ddr_dqh
A10	MGTPRXN2_216	NetIC4_A10		Input	8	
A11	GND	GND		Power	10	
A12	GND	GND		Power	10	
A13	IO_L10P_T1_16	EXT8.LVDS_2_P		I/O	4	
```
I parsed it using a script to add IO Standard, direction, if it is differential and so on to write constraint files for Vivado. I thought it would be more useful for you. Should I switch it with the one I pasted above?

jordens, 2017-11-23:
Don't bother with parsing and re-formatting. We [don't write vivado constraints directly anyway](https://github.com/m-labs/migen/blob/master/migen/build/platforms/sinara/kasli.py) and we are good at writing parsers ourselves ;)
The format above is fine. Is that directly from Altium?

marmeladapk, 2017-11-23:
Yes. I updated file in release.

closed 2017-11-23 by marmeladapk