sinara-hw/sinara #385: Nanoga schematic review

gkasprow, 2017-11-28:
@hartytp We have to produce prototypes of Nanoga board to finish the project at the University this year.
Did you have a time to look at the schematics and PCB?

hartytp, 2017-11-29:
@gkasprow I hand't realized that Nanonga had been done!

I'll have a look tomorrow.

In the mean time, please can you confirm that all relevant Allaki bug fixes have been applied to Nanonga?

gkasprow, 2017-11-29:
Well it had been done long time ago but there were more important things...
@sbouhabib Please check what @hartytp wrote about Allaki [issues](https://github.com/m-labs/sinara/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopen+allaki)

hartytp, 2017-11-29:
@gkasprow My biggest concern with producing this PCB now is the unresolved Sayma mechanical issues to do with SMPs etc. e.g. what if we decide that using this many SMPs for the AFE is too expensive/not mechanically good and we want to find a different solution for Sayma v2.

If at all possible, I'd prefer to hold off producing this PCB until we've decided what changes to make for Sayma V2. However, if that's not an option due to rules of your University we can go ahead.

gkasprow, 2017-11-29:
We can integrate @jbqubit idea of a screw-based ejection mechanism. Could be easy to implement on Nanoga right now.
Other connector idea:
http://suddendocs.samtec.com/catalog_english/grf1-j.pdf
https://www.samtec.com/products/grf1-p
they are also quite [expensive](https://www.digikey.com/product-detail/en/GRF1-P-P-02-E-ST-TH1/SAM8075-ND/1112918?utm_campaign=buynow&WT.z_cid=ref_octopart_dkc_buynow&utm_medium=aggregator&curr=usd&site=us&utm_source=octopart)

gkasprow, 2017-11-29:
@hartytp we can try to sign an annex to the agreement that prolongs this particular project.

hartytp, 2017-11-29:
Something like the IP5/IJ5 might be a better choice. If we buy a single 10-way high-isolation connector we'll probably find that it's mechanically a lot better than having many SMPs, and may not be more expensive.

gkasprow, 2017-11-29:
I looked at them but their stacking height is 10mm, we need max 9 mm.

hartytp, 2017-11-29:
Okay, that makes sense. In any case, something like that could be a better solution than the SMPs.

The more I think about it, the more I wonder whether it would also be better to make mezzanines double width (2 per Sayma rather than the current 4) so that they each take 4 DAC/ADC channels. This change might make some of the mechanical issues a bit easier (it also reduces the number of RF connections since the LO is shared between more channels).
 
Anyway, this is a conversation best had later on when we consider the next version of Sayma.

gkasprow, 2017-11-29:
@hartytp  I looked at IP5 and IJ6 connector once again and noticed that there is some undocumented feature - the IJ5 part has one vertical size while  IP5 has additional parameter in the symbol name and this number specifies stacking option.
So I downloaded 3D models and indeed, the stacking height varies.
Let's take 
IH5-06-05.0-L-D-1-TR
and   IP5-06-02.0-L-D-1-TR
With connector above combined I get 7.5mm
With connector IP5-06-03.0-L-D-1-TR I got exactly 9mm!!
So it seems this could be good solution.
There are either 2x4 or 2x6 connectors so in our case we would need to use 12 way one. We can use remaining pair for i.e. delivery of two copies of LO signal to make construction of the AFE board easier - we have free outputs in ADCLK chip anyway.

hartytp, 2017-11-29:
@sbouhabib Out of curiosity, what are your thoughts on IQ mixer choice for this design? Our top priority is phase + amplitude stability versus temperature (although, phase noise and linearity are also important). That mixer seems nice, but I haven't looked carefully for better choices. Did you look at other options? Are you happy there there isn't a significantly better alternative available?

hartytp, 2017-11-29:
Starting my review:

- [ ] Add version number (v1.0) to schematic
- [ ] On ADC channel page, "Variant" not "Wariant" (this is wrong in Allaki as well)
- [ ] Let's remove (or, at least DNP it) RF_OUT1 (this makes it easier to attach SMAs and saves some money). We can blank the front panel off (@gkasprow better suggestion than using tape?)
- [ ] Let's DNP the ADC_IN_1 SMA, and all components associated with it, including the InAmp. I don't see any need for this InAmp. If it makes things easier, feel free to remove the InAmp and associated components completely instead of DNP-ing them
- [ ] Why use the ADCLK944 instead of the two-output ADCLK925 (which has slightly better performance IIRC)? Is the 944 cheaper or something?
- [ ] Why generate the 3V3 from the 5V0, rather than directly from the 6V0?
- [ ] Why do the resistors on the ADC channel page not match the values given in the table? e.g. R1 should be 681R according to the table?
- [ ] Text is upside down for C4, R40, R42, C63 (also wrong on Allaki)
- [ ] Can you add an annotation on the demodulator page giving the max input power, please?
- Filters:
  - [ ] I think it's worth having a filter on the DAC outputs before the mixer to reject images. I'm assuming we'll use interpolation with a DAC clcok frequency >= 2GHz. Max RF output frequency is 400MHz so (@jordens correct me if I'm wrong about this) the bulk of what we're trying to filter out will be at frequencies above 1.6GHz. Pick whichever filter you want that will pass 400MHz but block above about 1.5GHz. I didn't find any nice differential filters, so we could either build our own out of discrete passives or do something like use 2 x XLF-641M+ for each DAC (please double check that there isn't a more suitable part before using!)
  - [ ] Is there any point having a filter after the mixer? If not, please remove it (both the passive and the MCL version). The current filter won't work too well as it only goes to 3GHz, and our target frequencies are up to 3.2GHz. My feeling is that if the IQ mixer is set up properly, the carrier and LO feed through should generally be low enough to not need filtering. In any case, given the range of values for the LO and RF that one can imagine using this board with, I can't see a good default value for this filter, so it's probably best for users to add an external filter if they really need it.

dhslichter, 2017-11-29:
See https://github.com/m-labs/sinara/issues/368#issuecomment-347983013, one can reduce the SMP mating/demating forces by using smooth bore SMP.  I am not saying this is the best alternative, but can be used.

jordens, 2017-11-29:
Yes. You should reconstruct after the DAC. But you need to decide which mix mode/Nyquist image you want. In the first zone you have 450 MHz bandwidth and then the first images to be suppressed from 1.55 GHz up (1 GS/s data, 2x interpolation).

hartytp, 2017-11-29:
> Yes. You should reconstruct after the DAC. But you need to decide which mix mode/Nyquist image you want. In the first zone you have 450 MHz bandwidth and then the first images to be suppressed from 1.55 GHz up (1 GS/s data, 2x interpolation).

Thanks for confirming!

My feeling is that this board is only going to be used in the first nyquist zone. I don't see so much use case for using the DAC in mixer mode in combination with an analogue mixer. In any case, so long as we chose a filter with a sensible package, users can change this later if needs be.

dhslichter, 2017-11-29:
Definitely want a reconstruction filter before the mixer, and a good one.

hartytp, 2017-11-29:
Having given this some more thought, I'd like to drop the Nanonga name and just call this the Sayma Upconverting AFE. If people make more upconverting AFEs in the future then we can call them "Upconverting AFE 2" etc. IMHO that's at least as unique and clear as giving them names like Nanonga etc.

Unless @jbqubit et al object, I'd advocate for renaming Allaki "Sayma Baseband AFE" in the next revision.

closed 2017-12-01 by hartytp