sinara-hw/sinara #386: Sayma AFE connectors

hartytp, 2017-11-29:
v1 Sayma AFEs use SMPs for RF connections (10 SMPs on Sayma, 10 on the AFE and 10 bullets joining them). This is problematic because:
- It's expensive (how expensive @gkasprow?)
- It takes an unreasonable level of insertion force to attach them
- The SMPs take up a lot of space

Can we find a multi-pin RF connector with acceptable isolation to replace the SMPs? What about IH5-06-05.0-L-D-1-TR and IP5-06-03.0-L-D-1-TR?
- [Catalog page](http://suddendocs.samtec.com/catalog_english/ip5.pdf)
- To do: ask manufacturer for S-parameters and some samples
  - Insertion loss: 3dB at 10.5GHz
  - Return loss: 15dB at 5.5GHz
  - Isolation: 80dB at 1.3GHz
- Stacking height is 9mm
- Total cost is $24USD (MOQ is about 100 pcs from DigiKey)
- Maybe go for a 2 x 6 connector, so 12 RF lines in total

gkasprow, 2017-11-29:
Each SMP barrel is roughly 5$ + SMP connector for 3$ which gives 11$ per single RF channel. With 12 channels we end up with 132$ which is 528$ per Sayma board and 4 Allaki.

hartytp, 2017-11-29:
Oh wow! I didn't realize it was that bad! Okay, let's definitely move to something like these connectors in the next iteration!

hartytp, 2017-11-29:
@gkasprow Will the 6-pin connector fit on the current Allaki? Catalog says it's 31.64mm wide. If not, that's a strong argument for moving to a 4 DAC channels per mezzanine form-factor in the next revision IMHO.

gkasprow, 2017-11-29:
Yes, it fits easily
[image]

hartytp, 2017-11-29:
Okay, nice!

Well, I don't see any reason not to do that in the next revision.

gkasprow, 2017-11-29:
Well, I remember I looked at these connectors while we were discussing a year ago about connector selection. Maybe they added other stacking options in the meantime...

dhslichter, 2017-11-29:
Stacking height is 10 mm.  I don't think we can spare the extra millimeter, can we @gkasprow?

hartytp, 2017-11-29:
> Stacking height is 10 mm.

No, see above comments. There are different versions available which give a 9mm stack height.

dhslichter, 2017-11-29:
Where do you see 9 mm?  This has to be a custom order from Samtec then, the connectors themselves are 9.09 mm tall.  

[image]

[image]

dhslichter, 2017-11-29:
The other issue that the SMPs address is placement/angular tolerances.  Since we also have another connector for the digital I/O and power rails, we need a connector which can tolerate whatever error in placement/board fabrication/etc.  One would likely want to move the RF connector a bit farther away from the power/digital IO connector for this reason.

gkasprow, 2017-11-29:
@dhslichter you can choose lead type in part number builder so I assume it is standard part number. They have 3D models which I downloaded and measured.

dhslichter, 2017-11-29:
S parameters are here for the IP5/IJ5:

http://suddendocs.samtec.com/testreports/hsc-report_ij5-ip5_web.pdf

[image]

dhslichter, 2017-11-29:
@gkasprow @hartytp I didn't see that in the other thread, now I understand.  If you can get this in a 9 mm stack height then it should be alright.  I can't see anything deeply problematic in the frequency response or the crosstalk:

[image]

gkasprow, 2017-11-29:
@dhslichter compare the two:
-03 option
[image]
-02 option
[image]

"odległość Y" means Y-axis distance

hartytp, 2017-11-29:
> "odległość Y" means Y-axis distance

Could also be a good name for our next PCB ;)

dhslichter, 2017-11-29:
If you can mate them together and get the right stack height, seems good to me.  Does the through-hole termination cause any issues with board routing on Sayma (or Allaki)?  Makes fanout on Allaki a little tighter I'd imagine...

gkasprow, 2017-11-29:
I will order samples to make sure they really exist :)

dhslichter, 2017-11-29:
FWIW if we go this way, we should ensure that the gender where multiple stack heights are available is the gender soldered to Sayma, such that there is only one height of connector available for the AFE boards (to reduce the chance of a screw-up for someone designing an AFE board).  I guess this means IJ5 on Allaki, IP5 on Sayma.

dhslichter, 2017-11-29:
AFAICT (from Octopart) nobody stocks the IP5-06-03.0-XXX anywhere, we might be subject to a minimum buy from Samtec.  Be good to understand the properties of that.  The cost savings over SMP will need to factor this in.

gkasprow, 2017-11-29:
@dhslichter I know. But we often use connector from SAMTEC that are not avaialble.
 Actually they are in so many options that it's hard to keep them on stock. On the other hand it's easy to get small quantities from Samtec directly.

dhslichter, 2017-11-29:
As long as Samtec doesn't make us buy 1,000 at a time I think it's alright :)

gkasprow, 2017-11-29:
Unfortunately they do no ship IP5-06-03.0-L-D-1-TR? They sample IP5-06-05.0-L-D-1-TR which differ in stacking height. Anyway, I will order samples to check how they look like / do some measurements.
I also asked for MOQ for IP5-06-03.0-L-D-1-TR.

dhslichter, 2017-11-29:
Thanks @gkasprow.  If the MOQ is reasonable we should proceed, otherwise I think sticking with SMP is the way to go.  Changing to smooth bore should cut the insertion/retraction forces quite substantially.

hartytp, 2017-11-29:
I think that @jordens point about risk is a good one. Given the cost of producing a batch of Saymas, we don't want to find that the AFEs don't fit because of the connectors.

AFAICT, the electrical properties of these connectors are well specified, so the risks are largely mechanical (does it fit together). How about this: we produce a pair of trivial PCBs with the RF and digital connectors for an AFE (4 DAC channels preferably). We can assemble these together and see if they fit.

We could also use this test rig to test out whether the front panels/front connectors fit to try to avoid issues there.

gkasprow, 2017-11-29:
That's why I ordered samples. In such cases I order simple 1 or 2-layer boards the size of desired boards and test if all fits.

dhslichter, 2017-11-30:
We need to do this to build statistics -- assemble ~20 dummy boards with connectors and make sure they all intermate suitably.  This comes down to board fabrication and assembly tolerances really, I think the connector dimensional tolerances themselves are generally tighter.  The big advantage of SMP is that one can tolerate substantial misalignment.  Remember that we have to consider not just lateral misalignment but also angular misalignment -- if a connector didn't fully seat down to the board across its whole width (happens all the time), what does that do?

gkasprow, 2017-11-30:
Both Samtec connectors have alignment pins so the tolerances should be tight. Anyway I will design simple AMC and mezzanine size boards to test how they mate and fit mechanically. We will also test new FMC-like front panel principle.

hartytp, 2017-11-30:
Sounds like a good plan Greg!

hartytp, 2018-03-08:
@gkasprow When you're back from your conference, can we make this the next item on the to do list, please? I want to make sure that waiting for the test PCB doesn't delay production of Sayma v2.0.

I think that what needs to be done is:
1. Run our ideas past Sammer or someone else who has experience using these kinds of connectors for low-noise RF. It's worth asking about things like:
   - What connector do they recommend? Is FMC HPC with lots of grounded pins (genders changed to ensure it's not compatible with standard FMC cards) a good choice or can they recommend something better?
   - Do they have any advice on how to lay the signals out in the connector, beyond the advice given in the app note above?
2. Design a pair of test PCBs.
   - PCBs should include 4 DAC channels, 4 ADC channels, 1 LO/reference, LVDS (for AFE ADC), TTLs, etc
   - All signals (DAC out, ADC in, LO/REF, LVDS, TLL, etc) should have CMCs on both PCBs (both sides of the connector) where possible to minimize high-frequency (>100MHz) cross-talk/radiation (go for small cheap high-frequency CMCs)
   - Pick a layout for the signals on the connector. This needs to include all relevant signals. Try to separate all signals by plenty of ground. Keep digital and analog signals on separate sides of the connector (but, still maintain a complete perimeter of grounded pins around the edge of the connector if possible).
   - All signals routed as 100Ohm differential pairs. There are several ways of routing the signals (see the app note above). Maybe try a few different options on the test PCB, so we can pick the one that works best?
   - Put a pair of SMAs on the PCB for each signal. We can use an external balun for differential-single-ended conversion where required.

Once that's done, I'm happy to make all required measurements.

gkasprow, 2018-03-08:
@hartytp we definitly need to move to Sayma 2.0 soon. The remaining issues will be solved soon and we cannot waste time waiting. There are other groups (NBI) interested in this hardware as well.
Do you think that for such measurement standard FR4 would suffice? The traces will be very short so IMHO it should do the job. I use Technoservice company to do quick prototypes on shared documentation so such 4 layer PCB cost is 50..100$ in a week. The only drawback is fixed stackup with .38mm prepreg.
Since we will try to use standard FMC panel mechanics with stuckup height of 10mm, what about testing these IP5/IJ5 connectors. I already received the samples with 10mm stacking height.
What input connector do you want to use? Up to 4 SMAs can fit on the FMC panel

hartytp, 2018-03-08:
@gkasprow Great!

FR4 sounds fine for me. We're only at roughly 1GHz, and if the SI is limited by FR4 then we're doing well! 

Maybe add a calibration trace as well (just a straight differential transmission line with a pair of SMAs at each end) for good measure?

> what about testing these IP5/IJ5 connectors

Happy to test the IP5/IJ5 connectors as well/instead, so long as they're not too much more expensive than an FMC. We'd need both the IP5/IJ5 and a digital/power connector, since there aren't as many pins as an FMC.

> The only drawback is fixed stackup with .38mm prepreg.
> Since we will try to use standard FMC panel mechanics with stuckup height of 10mm, 

I'd prefer to go for a standard 10mm stacking height, which would require switching to right-angled SMAs for the RF outputs, but I think other people (@jbqubit @dhslichter) objected  (https://github.com/m-labs/sinara/issues/402#issuecomment-350362080) because right angled SMAs didn't have good enough performance at 10GHz? Personally, I'm not too fussed about that, as I suspect that it will be better design in the long run to put the 10GHz stuff on a separate PCB outside of Sayma, but if people feel strongly about it then we need to keep the 9mm stacking height.

> What input connector do you want to use? Up to 4 SMAs can fit on the FMC panel

I think we agreed to use MMCX for the input connectors and SMA for the output connectors.

jbqubit, 2018-03-08:
AFAICT right-angled SMA for RF output is fine as is MMCX for input. 

Now that clock mezzanine is removed the AFE's can be several cm longer. The only mechanical obstructions in v1 are 2xSATA and 4-pin header. Increased board area would relax constraints on tight layout.

hartytp, 2018-03-08:
+1 to making the afe mezzanines as long as possible.

hartytp, 2018-03-08:
@gkasprow so if we allow ra smas then we can use either fmc or ip5/ij5. If the performance is okay then I prefer fmc because of cost, availability and only needing one connector for rf and digital. Can we make 2 sets of test boards so we can test both options?

Either way, let's go for a standard stack up height.

dhslichter, 2018-03-09:
@hartytp @gkasprow if we use reversed FMC for the AFE connectors we can go for an 8.5 mm stackup height (which is a standard height, stocked by distributors) and not need to use right angle SMAs.  One of the issues with right angle SMAs is not just the signal performance but also the cost (~$10 more per connector than regular edge mount).  However, if 10 mm stackup with right angle SMAs is the way things end up going, I am not opposed to doing that.  One can change the stackup height simply by changing the part number of the connectors on the Sayma board -- not something one wants to do after the fact, but if we decide to change stackup height down the line it's just changing one BOM item, no board changes required.  

Based on the test reports from Samtec it seems like a properly laid out FMC is going to perform as well as the ip5/ij5 stackup.  I think testing this using FR4 boards ought to be fine, but you just need to make sure that the stackup is controlled so you can get reliable impedances for the traces.  If you have a thru calibration to measure the loss per length that's fine.  

I would test the following sets of connectors:

FMC 8.5 mm stackup
FMC 10 mm stackup
IP5/IJ5 10 mm stackup

For the two sets of FMCs, there are enough pins that I would try several different configurations to check crosstalk -- see e.g. what Samtec has done in their SI test reports, but you could also try putting even more grounds in between signals to see if that has a beneficial effect.

dhslichter, 2018-03-09:
I also think longer AFEs would be nice.  It would be ideal if we went for two AFEs per Sayma, each one with a single reverse-polarity FMC connector to the RTM card to do analog, digital, and power.

hartytp, 2018-03-09:
Agreed @dhslichter. 

It seems that with decent CMCs, we should be able to get satisfactory performance from a standard 8.5mm stackup reversed polarity FMC.

@gkasprow can you have a go at producing the test board, please? I still think it's worth talking to someone like Sammer to see if there are any tricks we're missing.

dhslichter, 2018-03-09:
@hartytp I am a little nervous about the CMCs introducing distortion or temperature dependence into the analog signal path.  Do we really need these?  AFAICT the DAC shouldn't care about some noise coming back to its input from the connector (there are termination resistors, plus crosstalk should be low), and on the AFE side there will be a balun or diff amp anyway that will reject common mode noise.

hartytp, 2018-03-09:
Am I wrong in thinking that chokes on both ends help to reduce radiation/bear field cross talk? I was thinking of a dc-coupled 1:1 tlt/balun as a Cooke. Something like tc1-1-13m. Passes dc and microwaves, non-magnetic so no distortion.

Am I wrong here, or will that to the job. Can always add pads to test board and try shorting across it to see if it's actually useful.

dhslichter, 2018-03-09:
My point there is that you are going to have a transmission line balun on the AFE anyway, to convert from differential to single-ended, so do you really need an extra one?  And since the DAC is terminated, one shouldn't have to worry as much about common mode noise propagating backwards toward the DAC being reflected again and turning into forward crosstalk (although even then the balun on the AFE will help kill it).

hartytp, 2018-03-09:
Okay, I don't feel too strongly about this.

Might be worth adding a pad for a balun on the rx (dac) side on a couple of ljnes so we can test them out. I definitely wouldn't add them to Sayma unless we find they make a significant improvement on the test board. 

The lvcmos (switch ttl etc) should probably have cmcs in any case.

dhslichter, 2018-03-13:
@hartytp if we make some traces on the test boards with baluns and some without, we can compare the crosstalk for (nominally) identical wiring in the connectors depending on whether we have stripped common-mode crosstalk with a balun or not.  That would be instructive for this purpose...

hartytp, 2018-03-16:
@gkasprow Not sure what your workload is right now. I'd like to make getting Booster ready to ship back to Oxford your absolute top priority for now. However, let's make this adapter second priority after Booster. Would be good to get this sent off to manufacture relatively soon so that it doesn't hold up Sayma v2.0. Did you get a chance to talk to Sammer or anyone else about our ideas here?

gkasprow, 2018-03-16:
Recently I debugged the HMC issue and MMC Exar programming. And I'm involved in a thousand other projects so I'm always overloaded :)
I tried to catch someone from Samer team but it seem they are all away. They are building some super-cool RF stuff for ESS in Sweden so are away quite often.

hartytp, 2018-03-16:
:)

gkasprow, 2018-03-16:
And to answer your question, we will focus on the RF amplifier.

hartytp, 2018-03-16:
Sounds good! Let's come back to this later.

hartytp, 2018-04-23:
> I think it makes sense to do it in more or less final form. The dimension will be like FMC but longer, we will use FMC panel and screws, the connectors will be assembled at the front panel.

@gkasprow That all sounds sensible.

> The only stacking height that makes sense is 8.5mm because we can use straight SMA connectors.
Only SMA connectors with thick center pin can be used due to FMC bezel limitation.
Look at the example drawing - I placed 3 SMA connectors to check how much space would we have between the connector hole and the FMC bezel edge:

If you're confident that that will work then it's fine by me. But, looking at the images you posted, it doesn't look like there is enough clearance around the SMAs. Would it be better to go for a different stack height and use RA connectors?

gkasprow, 2018-04-23:
I'm not confident. I'd love to use RA SMA connector and 10mm stack. 10mm FMC bezels are COTS components. The same with FMC connectors. But the RF properties of RA SMA connector are worse than straight ones. @dhslichter do you think we can stay with RA SMA connectors? What about [these](https://www.digikey.com/product-detail/en/amphenol-rf-division/132203/ACX1234-ND/1011911?WT.z_slp_buy=amphenol-rf_sma-connectors)?

hartytp, 2018-04-23:
My feeling here is that we need to make sure that whatever we do actually works from a mechanical standpoint. Let's not go for something that's a bit marginal and wind up with a massive headache later on.

I'm not worried about the RF properties of RA SMAs: they should be absolutely fine to a few GHz. AFAICT, they are only likely to cause issues for work at 10GHz+. At that point, it's probably better to do the mixing on a separate, specialist board, rather than trying to cram it onto Sayma AFE. Or, one could use smaller connectors, such as SMPs, which work well at these frequencies.

The cost is a slight issue, but if we have to decide between slightly more money spent on SMAs or mechanical issues, then I'd definitely spend the money on RA SMAs (the costs are still nothing compared with the overall cost of Sayma).

dhslichter, 2018-04-23:
So is the issue here that we are trying to mash this into a standard FMC front panel?  Why are we doing this?  It seems if we make a front panel variant we could get around these issues of clearance, no?

I agree that RA SMA is probably alright for this given our frequencies, and that we definitely don't want to have mechanical issues.  However, if someone wants to do an upconverting board (or downconverting board), they might start to care about 12 GHz.  If there is a way we can modify the front panel geometry/design a bit to use the 8.5 mm stackup with straight connectors and have enough clearance, that would be my preferred solution.

dhslichter, 2018-04-23:
I am fine with RA SMA then if it fits in, with whatever FMC stackup height works best.

hartytp, 2018-04-25:
Cool, so I think we all agree that:
- The top priority is to get something that works well mechanically: no marginal clearances of odd tricks, please!
- Sticking as close to standard FMC as possible is preferred (standard stack-ups, COTS panels, etc) to doing non-standard things
- Straight SMAs are preferred to vertical ones where possible

With that in mind, I vote that we let @gkasprow pick whichever solution he thinks is best.

hartytp, 2018-05-29:
What about some traces for the digital lines to check diigtal to anlog cross talk? Digital lines will be in a separate part of the connector. For single ended io like ttls we should consider using cmcs to reduce emi issues

dhslichter, 2018-05-29:
I like it @gkasprow, makes it easier to identify the channels if they are evenly spaced out likes this.  Is there still enough room for silkscreen labeling that is legible if we add LEDs + light pipes?  

@hartytp what are you hoping the LEDs would show?  If we have status LEDs, I think it might be easier to have them live on the main Sayma RTM card, in the space in between the AFE front panels.

gkasprow, 2018-05-29:
we can place standard THT LEDs and bend prior assembly. I did not manage to find suitable light pipes.
Something like this.
[image]

Sure, we can add LEDs below but need to reserve some IO lines routed to the  AFE boards. There are no spare FPGA lines for that. I2C or SPI extender for LEDs need firmware to be written, so maybe let's stick with bi-colour LEDs on the AFE.

hartytp, 2018-05-30:
> @hartytp look here:

@gkasprow looks good! So long as those MMCXs won't interfere with getting an SMA spanner in there, it looks great!

> @hartytp what are you hoping the LEDs would show?

I quite like that Urukul displays the RF switch state via a front panel LED. I've found that useful a few times during debugging. Similarly, general purpose FP LEDs are often useful for debugging. But, it's definitely a nice-to-have and non critical. If it causes cost/mechanical issues then it can be scrapped.

>  If we have status LEDs, I think it might be easier to have them live on the main Sayma RTM card, in the space in between the AFE front panels.

I think that having some more LEDs on Sayma RTM/AMC would *also* be a good idea. e.g. for displaying PLL locked indicators.

> Sure, we can add LEDs below but need to reserve some IO lines routed to the AFE boards. There are no spare FPGA lines for that. I2C or SPI extender for LEDs need firmware to be written, so maybe let's stick with bi-colour LEDs on the AFE.

I think there will be a few spare lines on the RTM FPGA once we remove the clock mezzanine. Although, I'd like to route as much IO as possible to the AFE mezzanines for future proofing.

gkasprow, 2018-05-30:
3mm LEDs are too massive. I found 2mm ones that fit nicely. 
Look at this
[image]

@hartytp how would you like to use CMCS for LVCMOS lines?

hartytp, 2018-05-30:
> @hartytp how would you like to use CMCS for LVCMOS lines?

If you think it's redundant then don't bother. Was assuming you'd use a CMC on each end of the FMC con to convert them to differential signals (the n-signal input/output ties to ground at either end).

gkasprow, 2018-05-30:
I'm going to connect 32 IO lines to each FMC.  That would be a lot of CMCs :)
I located all IO lines on the FMC connector edge, far away from signal lines.

[97 more replies]

closed 2019-03-06 by hartytp