sinara-hw/sinara #387: Sayma AFE cost

hartytp, 2017-11-29:
@gkasprow Roughly what is the cost of Alaki now, including the board-to-board mezzanines on Sayma? What are the major drivers of this cost?

For Sayma v2.0 I'd really like to get to the point where separating the AFE onto a separate mezzanine does not contribute significantly to the overall cost of Sayma, compared with putting the AFE directly on the Sayma PCB.

Thoughts/questions:
- Are we using fancy RF substrates for this design? If so, can we scrap those and go to FR4?
- Can we make the mezzanines hold 4 DAC channels each? This has a few cost savings, including halving the number of regulators we need on the AFE. Also makes the AFE less cramped.
- Use different RF board-to-board connectors.
- Any other cost saving thoughts @gkasprow

gkasprow, 2017-11-29:
It is roughly 500EUR @ 25pcs.
The most expensive are SMA (TELEGARTNER) and SMP connectors.
Rogers material and process is also quite expensive. We can also use FR408, but here @sbouhabib opinion is needed.

hartytp, 2017-11-29:
Okay, that's a pretty high cost given how cheap the important components are.

More reason to replace the SMAs + SMPs with something cheaper and better mechanically.

I'd guess FR408 is absolutely fine for the relatively low frequencies we're working at (the trace length isn't that long IIRC). That should also reduce the lead time quite a bit, which is nice!

dhslichter, 2017-11-29:
For Allaki, I think FR408 is acceptable in terms of loss and impedance-matching quality because the frequencies involved are not aiming to be too high.  One reason we went for Rogers originally was that I had done a bunch of work with it already in testing trace widths etc., and we thought we'd try to put microwaves out directly with mix-mode (recall one can nominally generate 43Ca+ microwave frequencies using mix-mode...).  Let's save the fancy materials for another board.

jbqubit, 2017-11-30:
Cheaper, easier to manufacture sound like pluses. Specification of a thicker PCB would help with insertion force.

hartytp, 2017-11-30:
What's the insertion force of one of those 10-pin RF connectors? My guess is it's low enough that we don't need a thicker PCB (which is generally more expensive to manufacture, particularly when using nice laminates like FR408). But, Greg's test PCB will tell us this...

dhslichter, 2017-11-30:
You can probably make a top layer of FR408 with thicker layers of regular FR4 below (if we only care about the DAC side having good high-frequency loss properties).

gkasprow, 2017-11-30:
@dhslichter Yes, one can also combine FR4 with Rogers.

dhslichter, 2017-11-30:
Our NIST microwave boards are single layer Rogers on 3 layers of FR4 for cost purposes.

gkasprow, 2018-08-18:
This issue was moved to sinara-hw/BaseMod#5