sinara-hw/sinara #388: Nanonga schematic issues

hartytp, 2017-11-30:
Thanks for producing the really nice Nanonga schematic. Basically looks really good. I've completed my review, comments below, please ask if anything is unclear.

Starting my review:

- [ ] Add version number (v1.0) to schematic
- [ ] On ADC channel page, "Variant" not "Wariant" (this is wrong in Allaki as well)
- [ ] Let's remove RF_OUT1 SMA, so we just have a single SMA, labelled "OUT" on the schematic + silk.
- [ ] Let's remove the ADC_IN_1 SMA, and all components associated with it, including the InAmp. I don't see any need for this InAmp. We then have just a single input SMA labelled "IN" on the schematic + silk.
- [ ] Why use the ADCLK944 instead of the two-output ADCLK925 (which has slightly better performance IIRC)? Is the 944 cheaper or something?
- [ ] Why generate the 3V3 from the 5V0, rather than directly from the 6V0?
- [ ] Why do the resistors on the ADC channel page not match the values given in the table? e.g. R1 should be 681R according to the table?
- [ ] Text is upside down for C4, R40, R42, C63 (also wrong on Allaki)
- [ ] Can you add an annotation on the demodulator page giving the max input power, please?
- Filters:
  - [ ] I think it's worth having a filter on the DAC outputs before the mixer to reject images. I'm assuming we'll use interpolation with a DAC clcok frequency >= 2GHz. Max RF output frequency is 400MHz so (@jordens correct me if I'm wrong about this) the bulk of what we're trying to filter out will be at frequencies above 1.6GHz. Pick whichever filter you want that will pass 400MHz but block above about 1.5GHz. I didn't find any nice differential filters, so we could either build our own out of discrete passives or do something like use 2 x XLF-641M+ for each DAC (please double check that there isn't a more suitable part before using!). I don't mind whether you use the discrete filter or the MCL; do whichever is easier/cheaper/better (just make sure you annotate the schematic with the passband etc).
  - [ ] Let's put the same filter on the demodulator output (easier than using the passive filter IMO) to remove RF that will be rectified by the following OpAmp.
  - [ ] Is there any point having a filter after the mixer? If not, please remove it (both the passive and the MCL version). The current filter won't work too well as it only goes to 3GHz, and our target frequencies are up to 3.2GHz. My feeling is that if the IQ mixer is set up properly, the carrier and LO feed through should generally be low enough to not need filtering. In any case, given the range of values for the LO and RF that one can imagine using this board with, I can't see a good default value for this filter, so it's probably best for users to add an external filter if they really need it.
  - [ ] There should be a basic high-pass filter between the modulator and the amplifier to prevent any baseband leakage from reaching the amplifier. Please add an annotation for this filter specifying corner frequency etc.
- [ ] Please add an annotation with the power out of the IQ mixer when the DAC is running at max amplitude. Do we need an attenuator between the mixer and the amplifier to avoid saturating the amplifier or other components when the DAC is running at max amplitude? Should we have an attenuator like this on Allaki as well?
- [ ] What are R18 and R19 for? Either they should be DNF or IC4 + U1 should be DNF AFAICT.
- [ ] Let's make sure that we're using the cheapest substrate that will provide acceptable performance up to at least 4GHz.
- [ ] +-5V regulators must have diodes
- [ ] Is it worth putting inductors on the +-12V rails to remove high-frequency noise (these are generated by SMPSs on Sayma IIRC)? This is probably a good idea since there is no regulator on the supplies used for the InAmps. If we do this, we should also do this for Allaki.
- [ ] It would be nice to have a solder jumper to allow one to connect the analog input directly to the ADC, bypassing the InAmp, as is done on Allaki. If this is hard to do because of routing concerns then don't worry, it's not critical.
- [ ] Let's scrap the power detector and associated components . I really don't think this is very useful because:
  - The 24-bit ADC is so slow that it will take ages to read this out, making it annoying to use in most cases.
  - The dominant source of drift will generally be external power amplifiers.
- [ ] If the routing can be done easily, let's route the unused output of the RF switch (the one that currently connects to the power detector) to the demodulator via a solder jumper. In that way, one can use the demodulator as a fast power detector if wanted. If the routing for this is difficult, or likely to lead to cross-talk issues then don't bother.
- [ ] Should probably have a power budget on the schematic like the ones for the EEMs. This applies to Allaki as well.
- [ ] If we make this board a bit narrower, will that avoid some of the mechanical issues we've had with Allaki?
- [ ] Remove annotation "input considerations and Variants ...”
- [ ] Input/output SMA coupling caps should be 2*1nF on each SMA. (Why are there three caps on the input SMA?)
- [ ] Add part number to TR1 on the schematic
- [ ] Do we we need a bit of attenuation between the clock mux that supplies the and the (de)modulator? For lower LO values (below about 3GHz), the mux supplies 1.5Vpp differential signal = +8dBm. But, the (de)modulator wants a max LO input power of +6dBm. Let's aim for the max LO power (i.e. power for frequencies <2GHz) to be a little below +6dBm after the balun
- [ ] Demodulator output needs to be DC coupled, since we will generally user this with the RF at the same frequency as the LO. Note the common-mode offset on these signals.
- [ ] Demodulator load should be 450Ohm differential typically (ie not 50Ohm terminated). Currently, both demodulator outputs are terminated in 50R to GND AFAICT. These signals need to be properly matched into the differential ADC driver to ensure that the max signal amplitude corresponds to the max differential signal the ADC can accept. We should probably have three entries in the component value table: for InAmp; 1 for 50Ohm input direct to the ADC driver; and, one for the demodulator.
- [ ] IQ_DEMOD_I_P and IQ_DEMOD_Q_N inputs on ADC driver page should really just be IQ_DEMOD_P and IQ_DEMOD_N since this is used as both an I and a Q input.
-  [ ] Demodulator enable input is active low. This should be noted on the schematic (e.g. label this signal with a bar over the top). I'd like the demodulator to be enabled when this signal is not actively driven, so this pin should be pulled low, not high as in the current design. Similarly, please can we have the modulator disable input pulled low by default? That way we don't have to actively drive these pins to make the board work.
- [ ] Our LO frequency will typically be a bit below 3GHz. From figure 13 of the demodulator data sheet, the optimal Vadj is about 4.25V. From table 4, the optimal value for R77 is probably 1k (although 1k5 should be fine too). Please double check the best value for a 3GHz LO and add an annotation saying that R77 and R90 are chosen for best performance for LO at 3GHz.
- [ ] What are C45, R69, etc for? Can we scrap these? There are quite a few components on this PCB for matching, which I don't anticipate using, but which take up space. Would it be worth removing some of them?
- [ ] Modulator IQ inputs must be biased to +500mV by the DAC. If I understand correctly, the biasing resistors have been chosen so that with 20mA full-scale DAC current the common-mode voltage offset is 500mV and the swing is 1Vpp (+4dBm), while still providing a decent 100R differential termination to minimise reflections. Is that correct? This info should all be annotated on the schematic.
- [ ] In the current schematic, the jumper arrangement to connect the demodulator is not clear to me. Could you improve the annotation to make it clear how this works (maybe a little diagram like the other diagram for jumpers?). The default population option should have the demodulator connected.

hartytp, 2017-11-30:
Okay, sorry for the omission, but a big issue I missed in my review:

- The current demodulator arrangement won't work for us, since the BW of the ADC, demodulator etc is too low. Our RF frequency from the DAC will be up to 400MHz which is outside the bandwidth of these components.
- So, to fix this, the LO for the demodulator needs to be supplied by the RF output from the modulator. This can be done, for example, using a resistive splitter after the amplifier (we probably need some attenuation there anyway).
- This may allow us to scrap the LO fanout buffer. Or, you can keep it if you think it helps with signal integrity (what behaves better with a bad input clock? the buffer or the modulator?). If you think it will help, you could also add a clock buffer at the demodulator's LO input to square up the RF signal.
- The disadvantage of this arrangement is that the demodulator doesn't work when the DAC RF isn't on. But, I don't really see any way of getting around that (clever suggestions @dhslichter?)

hartytp, 2017-11-30:
Sorry, ignore the above. That's not a nice way of using the demodulator.

Provisionally, my conclusion is that we should just scrap the demodulator entirely. If we need it, it should probably be on a separate PCB.

Let me have a think about this and confirm what I'd like to do soon.

hartytp, 2017-12-01:
@gkasprow Okay, I've completed my review for this design (again!). See the top post on this issue for details.

Most of the changes involve removing components to simplify the design, so hopefully it's not too much work. Let me know if you have any questions. Otherwise, can you post the schematic once it's done so I can sign off on it.

Thanks!

gkasprow, 2018-08-18:
This issue was moved to sinara-hw/MixMod#5

closed 2018-08-18 by gkasprow