sinara-hw/sinara #390: sinara/RELEASE_NOTES.rst

jbqubit, 2017-11-30:
@gkasprow recently added tags for particular commits corresponding to releases of particular Sinara hardware. For example, https://github.com/m-labs/sinara/releases/tag/Zotlino/1.0rc1. Thank you for this! 

I propose creating a root level file RELEASE_NOTES.rst similar to [ARTIQ](https://github.com/m-labs/artiq/blob/master/RELEASE_NOTES.rst) that indicates the differences between releases. This would be an indication to users of the hardware changes they need to have in mind when switching between eg v1.0 and v1.1. I think it's fine if it's in reverse chronological order (blog style) and board types are intermixed. It would be WUT who would maintain this AFAICT so please weight in @gkasprow. 

Here are the tags so far.

3U_VHDCI/1.1rc1
3U_SMA/1.1rc1
3U_SMA/v1.1 
3U_BNC/1.1rc1
1.1rc1 
Zotlino/1.0rc1  
Kasli/v1.0
Clocker/1.0rc1
Urukul/v1.0
Urukul/1.0rc1
Grabber/1.0rc1 
rev_1.0_production_files
proto1 
production_v1
Novo_Zotlino/1.0rc1
3U_BNC_SMA_LVDS_Tester_VHDCI/1.1rc1
0.1rc1

Near the end of the list are some tags that don't indicate which board is part of the release. @gkasprow please edit/delete those tags.

gkasprow, 2017-11-30:
@jbqubit I tried to delete them but for some strange reason I cannot do it from Tortoise level...

hartytp, 2018-01-19:
Is anyone planning to do this, or should we close the issue?

jbqubit, 2018-01-19:
I advocate that we should do something like this. But as the burden is likely to fall on @gkasprow he's to have the final word.

gkasprow, 2018-01-25:
I will do that but at the moment have several critical issues to solve.

hartytp, 2018-05-03:
Not relevant now that we have separate repositories for different PCBs.

closed 2018-05-03 by hartytp