sinara-hw/sinara #394: 2018_poll: hardware naming

jbqubit, 2017-11-30:
In April 2017 a [poll](https://docs.google.com/forms/d/1SukB2WKPHiv6ipd4B4Cru3YUQyD0ptebjBNqY6-BMyU/edit#responses) was conducted of the ARTIQ and Sayma user community. Since I came up with the poll questions last year and we're approaching the one year anniversary, I'm curious what you all think should be asked in the 2018 poll. 

Naming of Sinara hardware is a hot hot hot topic. https://github.com/m-labs/sinara/issues/328#issuecomment-347619611 One aspect of the controversy is whether metadata names or unique names should be used. This is an exciting enough question I've created its own Issue to discuss if/how to include the topic in a 2018 poll. 

---------------

In your sole subjective opinion please answer the following questions. 

***Question 0***
On a scale of 0 to 4, do you care not at all (0) or very much (4) about this topic?

***Question 1***
On a scale of 0 to 4, do you prefer (0) metadata names or (4) unique names? For example, to describe a digital input-output board with a BNC form factor for the Sinara EEM peripheral standard here are example names of each class.
- metadata names:  ```EEM_DIO_BNC``` or ```EEM_BNC_8CH``` 
- unique names: ```obviouse``` or ```depeariod```

***Question 2***
On a scale of 0 to 4, do you find it easier to remember (0) metadata names or (4) unique names?

hartytp, 2017-12-01:
Let's not do this. Realistically, only a few of us are sad enough to care about this. The ones who do are being pretty vocal on GitHub, everyone else won't bother to answer the poll.

jbqubit, 2017-12-01:
>  everyone else won't bother to answer the poll.

Precisely why I'd like to include it in the poll. ;) 

Also, this Issue is meant to spark conversation about what to include in the next poll. Let's keep it open for a while.

dhslichter, 2017-12-01:
@hartytp +1

dhslichter, 2017-12-01:
@jbqubit make a separate issue "what should we include in the poll" for that.

closed 2017-12-01 by dhslichter

gkasprow, 2017-12-01:
In my lab we use metadata names anyway. That's why I put two naming conventions in documentation, otherwise I'd have to explain again and again what Novo means to all newbies and physicists.