sinara-hw/sinara #396: renaming Allaki and Nanoga

jbqubit, 2017-12-01:
@hartytp said in https://github.com/m-labs/sinara/issues/385

> Having given this some more thought, I'd like to drop the Nanonga name and just call this the Sayma Upconverting AFE. If people make more upconverting AFEs in the future then we can call them "Upconverting AFE 2" etc. IMHO that's at least as unique and clear as giving them names like Nanonga etc. Unless @jbqubit et al object, I'd advocate for renaming Allaki "Sayma Baseband AFE" in the next revision.

hartytp, 2017-12-01:
@jbqubit What do you want me to do? I'll change the Wiki once the names are changed on the schematics.

dhslichter, 2017-12-01:
Can we pick a name that uses max 3 syllables?

jbqubit, 2017-12-01:
I'm fine with ditching the name allaki in a future revision. 

I advocate for combining multi-word names with underscores. The intent is to emphasize that the name is a compound word and not an assembly of adjectives. 

If we are disciplined about this then it would be possible to search git for "sayma_afe_baseband" instead of allaki and "sayma_afe_upconverting" instead of nanoga.

jbqubit, 2017-12-01:
Let's leave this Issue open until next revision of the hardware and make the name in the wiki at that time.

dhslichter, 2017-12-01:
> I advocate for combining multi-word names with underscores. The intent is to emphasize that the name is a compound word and not an assembly of adjectives.

If you need underscores, you may well need a new name.  How about:

Basecard
Mixcard

hartytp, 2017-12-01:
Base-mez and mix-mez? To emphasize that they're mezzanine daughter cards that plug into Sayma?

hartytp, 2017-12-01:
Scratch that, dumb suggestion. I'd just stick with "baseband" and "mixer" ("full"/unique names "Sayma_AFE_baseband" and "Sayma_AFE_mixer" as @jbqubit suggests).

dhslichter, 2017-12-01:
> To emphasize that they're mezzanine daughter cards that plug into Sayma?

I don't think we need to emphasize that in the name.  I think we want a name that is suggestive enough to help associative memory, but doesn't try to say anything too precise about the thing.

> ("full"/unique names "Sayma_AFE_baseband" and "Sayma_AFE_mixer" as @jbqubit suggests).

Requiring a long name to distinguish between variants which share the same short name is a disaster.  Each thing should have one name.  I know that @hartytp has a number of properties, but I don't require them to be part of his name to know what they are.  Short names only, please!  And then if you make a new variant, come up with a new short name.

dhslichter, 2017-12-01:
> I'd just stick with "baseband" and "mixer"

You don't want to use names that are also regular words you would be using in the discussion.  I want to be able to use the word "mixer" to mean what it means without generating confusion about whether I am actually referring to a particular hardware item.  

Basecard/Mixcard, or Basemod/Mixmod, or Basehat/Mixhat, or BaseAFE/MixAFE....

dhslichter, 2017-12-01:
> If we are disciplined about this then it would be possible to search git for "sayma_afe_baseband" instead of allaki and "sayma_afe_upconverting" instead of nanoga.

This is exactly the sort of discipline that we do not have, as evidenced by all of the BNC/SMA name variants.  Make it easy!  Pick a simple name and stick to it!  Allaki and Nanoga are the right length as names, it'd just be nice to have something that helps a bit with associative memory.

jbqubit, 2017-12-01:
There's lots of things that plug into other things in the Sinara ecosystem and, in this case, variants of the thing that plugs. The idea is to have the name read like a path of sorts. 
- Sayma_AFE_baseband
- Sayma_AFE_upconverter

and

- FMC_VHDCI
- EEM_DIO_BNC
- EEM_DIO_SMA 
- EEM_DIO_RJ45
- EEM_DDS_9914
- EEM_DDS_9912
- EEM_DAC...

jbqubit, 2017-12-01:
Glad there's so much interest in naming our children! 

Can we put this decision on hold until, say Q2, 2018? For me anyway, getting the hardware to work is a higher priority than putting a bow on it.

hartytp, 2017-12-01:
> Glad there's so much interest in naming our children!

I know. We literally spent about 1% this long naming our daughter. wtf? and, where are the bike sheds again?

hartytp, 2017-12-01:
Sod it, for the up converting mezzanine v1.0 we'll go for "mixmod".

For clarity though, I'm fine with naming the directory Sayma_AFE_upconverting. That's not a name people will use in conversation, but it's instantly recognisable and people will be able to find what they want, which is the important thing!

hartytp, 2017-12-01:
Let's go for "BaseMod" and "Sayma_AFE_BaseMod" for Allaki. Please don't suggest anything else unless you **really** care about this.

jbqubit, 2017-12-01:
I like: 
-  Sayma_AFE_MixMod
 - Sayma_AFE_BaseMod

dhslichter, 2017-12-01:
Directories in repository named according to this "file path" style are OK with me, but why not just use a directory called "Sayma_AFE" containing folders called "MixMod" and "BaseMod" and just have those be the names of the boards?  It accomplishes exactly the same thing with less hassle, no double naming convention, etc.  All you have to do is be creative enough to come up with a different name for each board.

hartytp, 2017-12-01:
That's probably a better idea! Let's do that.

dhslichter, 2017-12-01:
@jbqubit the PCB silk should read Sayma_AFE/BaseMod if that's the directory you are putting it in, not with the two underscores.

hartytp, 2017-12-01:
Or just ""MixMod Upconverting AFE" as we're doing on MixMod. That's more readable IMO, and printing complete directory paths on PCB silk screen things like an odd and unnecessary thing to do IMHO.

sbourdeauducq, 2017-12-02:
Let's just keep calling it Allaki.

sbourdeauducq, 2017-12-02:
> The idea is to have the name read like a path of sorts.

No. Please stop doing that.

hartytp, 2017-12-02:
> The idea is to have the name read like a path of sorts.

Agreed, let's stop doing that.

> Let's just keep calling it Allaki.

No thanks. I've talked to quite a few users now and everyone I've spoken to prefers BaseMod.

FWIW, the up converting mezzanine will definitely be MixMod (we're making an executive decision on that as the funders of that board) so BaseMod would be more consistent than Allaki.

It's pretty common with these projects to assign a code name during prototyping, which gets changed for the production version to something more user friendly by the "marketing department".

jbqubit, 2017-12-04:
I don't like nonsensical names such as Allaki. BaseMod is good.

jbqubit, 2017-12-14:
- I've cleaned up Wiki to shift references of Allaki to BaseMod. As I'm paying for this board this choice is final. 
- I've cleaned up Wiki to shift references of Nanoga to MixMod. @hartytp is paying for this and has indicated that his choice is final.

closed 2017-12-14 by jbqubit

hartytp, 2017-12-14:
@jbqubit Still needs changes in a few other places: PCB silk, PCB schematics, PCB file names (think that's it).

hartytp, 2017-12-14:
Maybe re-open, but remove the discussion label, and close once the other changes are made?

jbqubit, 2017-12-14:
I've created new Issues for @sbouhabib.