sinara-hw/sinara #397: Sayma upconverting AFE schematic review: power detector

hartytp, 2017-12-01:
- [ ] Let's scrap the power detector. (I'd also suggest we consider doing this for the next revision of Allaki)!
- It takes up an annoyingly large amount of space on an already cramped PCB and adds extra cost and complexity.
- I don't think it's actually that useful because:
  - It is read out by the 24-bit ADC which is very slow, currently untested and doesn't have any drivers written for it.
  - In most cases, the dominant source of drift/power errors come from components external to Sayma, such as power amps etc
  - The thermal environment for this power detector isn't great, so the detector's stability won't be amazing.
  - Generally, it's probably better to use an external detector (e.g. an ADL5511) in a well controlled environment near the load and to read the detector out using the InAmp and the fast ADC.

hartytp, 2017-12-01:
Note also that on the Allaki schematic R18 and R19 seem to be populated, as well as the power detector. Either these are actually DNP, in which case that should be clearly indicated on the schematic, or this is a mistake!

gkasprow, 2018-08-18:
This issue was moved to sinara-hw/MixMod#8

closed 2018-08-18 by gkasprow