sinara-hw/sinara #401: Sayma BaseMod and MixMod ADCs

hartytp, 2017-12-01:
- [ ] Sayma BaseMod AFE and Sayma MixMod should have one 1MSPS 16-bit ADC per analog input on them to digitise the analog input.
- [ ] Probably use SPI (need to think about how this signal will be routed to Sayma RTM).
- [ ] Remove the "fast" input path where the SMA connects directly to the ADC driver OpAmp. This means removing the solder jumpers. We can also remove the matching pads at the input (this signal BW is about 1MHz so we don't need to worry about matching connector parasitics out).
- [ ] Replace driver OpAmp with a more suitable one.

hartytp, 2017-12-01:
AFAICs, there are currently 10 unused differential pairs on the AMC<->RTM connector, which could potentially be used to connect the AFE ADCs to the AMC FPGA. Additionally, since we're no longer using the fast ADCs, we can free up a further 8 differential lines. So, we potentially have 18 differential pairs or 36 single-ended lines (the SPI bus only needs to go to 50MHz) to use. That's enough for a full 4-wire SPI bus for each ADC, which is nice!

@gkasprow do you think 50MHz SPI single-ended over these connectors will be okay?

dhslichter, 2017-12-02:
Please at least translate to LVDS to run through the AMC - RTM connector -- way less crosstalk potential.  Also, can we have all the digital lines from the AFE cards go to the RTM FPGA (as is currently the case) still?  Otherwise we have a daughtercard that talks half to one of two FPGAs depending on which pin you choose.  If you decide that this is a feature and not a bug, perhaps.  It seems to me that it should be pretty simple to write gateware for the RTM FPGA that just transparently translates signals between the AFE and the AMC FPGA, without introducing meaningful latency.  You can use this to do the LVCMOS/LVDS translation as well.

hartytp, 2017-12-02:
> Please at least translate to LVDS to run through the AMC - RTM connector -- way less crosstalk potential. 

We can do that if it's necessary from a signal integrity viewpoint (don't forget that our SPI bus will only run at 50MHz or so). However, don't forget though that IO count is really limited on that connector, so we should be careful not to waste lines.

If we do decide to go differential through this connector then we'll almost certainly need to repurpose the pins currently used for the high-speed ADC JESD. That's absolutely fine IMO.

@gkasprow What do you think? Is LVDS necessary, or can we get away with LVCMOS over the AMC <-> RTM connector?

> Also, can we have all the digital lines from the AFE cards go to the RTM FPGA (as is currently the case) still?

> t seems to me that it should be pretty simple to write gateware for the RTM FPGA that just transparently translates signals between the AFE and the AMC FPGA (unidirectional translation is all that's required), without introducing meaningful latency. You can use this to do the LVCMOS/LVDS translation as well. And then if you decide to do something else later instead of using these for ADC data readback, you've just added some nice differential pairs between the AMC and RTM FPGAs which could be used for other purposes.

Potentially a really nice idea, so long as it doesn't introduce too much uncontrolled latency. Don't forget that unless we use a source-synchronous bus for the ADCs (which is a bit rare on ADCs and has its own issues), we need the latency on this to be controlled to much better than the period of the SPI clock. Deterministic latency is obviously fine, as it can be measured and compensated for, but I don't want anything that varies between gateware builds.

I'd really like to hear @jordens and @sbourdeauducq  thoughts on this, since they will probably end up writing most of the gateware in the long run! Also, what kind of ADC interface do you want to use? 

If we go differential through the connector, then we will have to run all ADCs off a common bus, which is probably okay, but does have some downsides. For a common bus, I'd imagine doing something like:

- 1 x SCLK (shared)
- 1 x CNV (shared)
- 8 x CS
- 8 x MISO (maybe run all MISO lines to the FPGA and let the FPGA mux them?)
- 1 x MOSI (shared)

However, that's 19 lines and we only have 18...

jbqubit, 2017-12-02:
> Additionally, since we're no longer using the fast ADCs, we can free up a further 8 differential lines.

@hartytp We've only recently begun discussing the ADCs on Sayma_RTM in #389. Please don't count on major modifications to Sayma_RTM since there's not budget for it right now. Using existing AFE DIO for SPI with RTM FPGA for slow ADCs sounds more approachable.

hartytp, 2017-12-02:
To put it another way: some of changes we're taking about, like changing the AFE RF connectors, involve quite a lot of work. Removing the fast ADC (clicking delete a few dozen times) and adding a few LVDS lines is almost no work to implement. The work involved is all in thinking it through, and making sure that we actually get it right this time.

hartytp, 2017-12-02:
Putting the ADC on the AFE mezzanine is also great from a risk mitigation point, since if we mess it up we only have to remake some relatively cheap mezzanines, which will also help to speed things up.

hartytp, 2017-12-02:
The ADC is an important feature which we want to get right. If we need to cut costs elsewhere to make room in the budget for that then there are things we can do. e.g. looks like the clock mezzanine won't be needed (at least in its current form) so let's cancel building that.

hartytp, 2017-12-02:
If we don't have the funding at the moment to do actually put an ADC on BaseMod, then that's fine. That board is cheap and easy to fix later. The important thing for now is to ensure that Sayma is designed so that the ADC can be added later. That means we need a decent amount of digital IO between the mezzanines and the AMC FPGA. As Dan points out, we can basically use the RTM FPGA a bit like the CPLD on Urukul, to do flexible routing. So, all we need to do is route some extra IO from the mezzanines to the RTM FPGA and route some extra lines from the RTM FPGA to the RTM<->AMC connector. That's really not a lot of work at all.

But, AFAICT, unless we repurpose the lines currently used by the ADC JSED, there just won't be enough IO to implement ADCs later on. We've all already basically agreed that the JSED ADC on Sayma was the wrong choice (far too expensive, consumes far too much power, too much latency for fast feedback applications, no funded gateware/software, too complex for users to write their own gateware, etc). So, removing it and connecting those lines directly to the RTM FPGA is a big improvement to Sayma with a very minima amount of work.

sbourdeauducq, 2017-12-02:
If we are modifying the Sayma RTM board, why not replace the JESD ADCs with SPI ADCs instead of removing the ADC capability completely from the board? This way the "analog" mezzanines remain analog.
Are there enough pins left on the RTM FPGA?

sbourdeauducq, 2017-12-02:
And let's have consistent names for those AFEs instead of BaseMod/MixMod...

sbourdeauducq, 2017-12-02:
There are also cheaper JESD ADCs, and then we can keep using the planned JESD interface and transceiver links, e.g. here is a >8-channel $120 chip:
http://www.ti.com/product/ADS52J90/description
I'm not a huge fan of transceivers and JESD latencies, but carrying the data to the AMC FPGA is going to be a problem in any case. JESD also has the advantages of scrambling and high sample rates.

hartytp, 2017-12-02:
> There are also cheaper JESD ADCs, and then we can keep using the planned JESD interface and transceiver links, e.g. here is a >8-channel "low-power" $120 chip:
http://www.ti.com/product/ADS52J90/description

Still not cheap though.

> JESD also has the advantages of scrambling and high sample rates.

Scrambling should not be required for a simple SPI bus.

High sample rates are good in principle. But, who is planning to use these boards that actually needs them? AFAICT, the ADC we have chosen is about as useless as possible: too slow for direct digitisation of the DAC's RF and to fast for the "slow" uses in most AMO experiments.

JESD also has the huge disadvantage of being very complex and requiring complex logic to interface with.

hartytp, 2017-12-02:
> And let's have consistent and opaque names for those AFEs instead of BaseMod/MixMod.
BaseMod is Allaki, and the name we had planned for the closest thing to MixMod is Nanoga/Tatysh - and I can find many similar names if you don't like those.

I don't see what BaseMod and MIxMod are inconsistent with. Sure, it's not what we originally planned, but plans change based on feedback from users. And, if the other names you can find are still based on Russian towns then we will have the same issue. AFAICT, all the users like BaseMod/MixMod, so let's use them.

hartytp, 2017-12-02:
> If we are modifying the Sayma RTM board, why not replace the JESD ADCs with SPI ADCs instead of removing the ADC capability completely from the board? This way the "analog" mezzanines remain analog.

I'd be fine with putting SPI ADCs on Sayma instead of on the AFE if that works out easier.

But, I think there is a strong argument for putting them on the AFE because it's so much easier to route digital signals between boards than analog ones. If this lets us remove a couple of SMPs then that's a big plus both in terms of cost and in terms of mechanical issues.

The other advantage of putting them on the mezzanine is risk-mitigation. It means we can press ahead with Sayma without having to design the ADCs too carefully. So long as we put enough digital IO on the PCB, we can sort out the analog side later on.

sbourdeauducq, 2017-12-03:
> ADC we have chosen is about as useless as possible

I have not chosen that ADC. It was chosen by a process similar to what you are doing with the names right now.

sbourdeauducq, 2017-12-03:
> But, I think there is a strong argument for putting them on the AFE because it's so much easier to route digital signals between boards than analog ones.

But besides cost and purported mechanical issues, SMPs solve the problem, right?

> If this lets us remove a couple of SMPs then that's a big plus both in terms of cost and in terms of mechanical issues
 
As far as I can tell, a few more SMPs do no make much of a difference mechanically, compared to what it already is. Regarding price, are there alternative manufacturers and/or discounts?

> The other advantage of putting them on the mezzanine is risk-mitigation. It means we can press ahead with Sayma without having to design the ADCs too carefully. 

The original idea of the AFE was to make them as simple as possible, so that people could mamocuske their own and focus on the purely analog/RF aspects.
Let's generally prefer well designed solutions over shortcuts. As far as I can tell the urgency comes mostly from people needing a AWG, and ignoring/DNPing the current ADC would be fine there.

hartytp, 2017-12-03:
> Let's generally prefer well designed solutions over shortcuts. As far as I can tell the urgency comes mostly from people needing a AWG, and ignoring/DNPing the current ADC would be fine there.

I don't agree with your reasoning here. Essentially all users of the AWG also need an ADC for feedback (laser/microwave intensity stabilisation). So, any well designed solution for Sayma must provide that. The alternative is users using something like the NIST external PID box and an external RF modulator (but I don't think that anyone thinks that that's an ideal solution). So simply DNPing the current ADC without providing an alternate ADC on either Sayma or the AFE mezzanine is not a good solution.

For the reasons we've discussed before, JESD ADCs are a poor choice for us, so the well-designed solution seems to be to place a simple SPI ADC on either Sayma or on the AFE mezzanine. I can live with either solution, the question is which one will be easier/cheaper.

> But besides cost and purported mechanical issues, SMPs solve the problem, right?
> As far as I can tell, a few more SMPs do no make much of a difference mechanically, compared to what it already is. Regarding price, are there alternative manufacturers and/or discounts?

> The original idea of the AFE was to make them as simple as possible, so that people could mamocuske their own and focus on the purely analog/RF aspects.

I disagree with you here as well. Compared with the complexity of these mezzanines, a MSPS ADC is totally trivial. I really don't think it's correct to say that adding a low-speed ADC makes these mezzanines significantly harder to design. Moreover, users will have BaseMod to use as a template, so they can just copy and paste the ADC design from there. But, to be honest, alone who thinks that a MSPS SPI ADC is remotely complex to design probably shouldn't be trying to build an RF front-end for Sayma.

Digitizing analog signals as early as possible is generally a good idea. Once the signal is digital one no-longer has to worry about pickup, grounding, etc. Moreover, the digitised signals can be routed through much cheaper, higher-density connectors. 

One would never put a high-speed JESD ADC on one of these mezzanines because they're far too complex and need far too much high-speed digital IO. But MSPS ADCs are not a problem because they're so much easier.

> As far as I can tell, a few more SMPs do no make much of a difference mechanically, compared to what it already is. Regarding price, are there alternative manufacturers and/or discounts?
> But besides cost and purported mechanical issues, SMPs solve the problem, right?

The SMP approach can be made to work. However, it will always be more expensive than putting the ADC on the mezzanine, and it will always be mechanically worse as well. So, yes, it can be made to work and it is *a* solution, but it doesn't seem to be a *better* solution than putting the ADC on the AFE mezzanine.

sbourdeauducq, 2017-12-03:
But what about putting something between a big JESD ADC and a 1M ADC on Sayma?

hartytp, 2017-12-03:
Like what? At this stage all suggestions are welcome.

hartytp, 2017-12-03:
Although, I'd be against anything JESD. It's just too complicated, and not necessary given the relatively low sample rates we actually need for this ADC. Anything which satisfies the following should be fine though: simple digital interface; lowish cost and power consumption; 16-bit accuracy with decent temperature coefficient, transition noise and DNL of <=2LSB; etc.

FWIW, I still think that if we go for a really simple ADC then it's probably better to put it on the mezzanine than directly on Sayma. As the complexity of getting low-noise analog signals between boards, while worrying about screening, grounding etc, and the cost/mechanical issues with the connectors we're using to do it is far worse than the increased cost of putting a simple ADC on the AFE.

Having said that, one option would be to put the same ADC we have on Sampler on Sayma RTM. That provides a nice compact digital interface for all 8 channels and has good performance. The only downside IMO is that it's unipolar so requires a somewhat complex driving arrangement (cf Sampler). Putting those dividers on Sayma would probably take up more space than we have atm.

sbourdeauducq, 2017-12-03:
I don't like the latency and complexity of JESD either, especially when combined with the abysmal design and bug density of the Xilinx transceivers, but carrying the data through a limited number of FPGA pins and a limited number of RTM connector pins isn't going to be straightforward in any case.

hartytp, 2017-12-03:
@sbourdeauducq That's fair.

Let's pause to establish some common ground. I think I'm correct in saying that we all agree on the following:
- The currently planned users of Sayma (mainly AMO physicists/ion trappers) don't need high sample rates for the ADC. Moreover, they actively don't want the expense/cost/complexity/power dissipation of a very high speed ADC.
- As a result, simple interfaces like SPI/LVDS are strongly preferred to JESD.
- However, if there aren't enough FPGA/connector pins to implement a decent SPI/LVDS interface then we may have to accept that JESD is the only option.
- We can leave the high-speed JSED ADC on the board, but DNPd, so that if a use case emerges for it in the future, people can repopulate it. (Although, note that it won't have been properly characterised, so they will need to be willing do do some testing/bug fixing).

If we all agree that that, then I think the next step should be to propose some ways of implementing the SPI/LVDS interface in detail. Once we have concrete detailed proposals we can make an informed decision.

hartytp, 2017-12-03:
Leaving the JSED pins alone, we have 10 unused differential pairs on the RTM<->AMC connector. How many pins do we have spare on the RTM/ADC FPGAs?

What if we go for the LTC2320-16 (same ADC as Sampler) on Sayma? That only needs 7 LVDS pairs: CNV, SCK, SDOA-D, CLKOUT. So, we definitely have enough pins on the AMC<->RTM connector for that. No need to route that through the RTM FPGA AFAICT (if there are pins then we can, but I'm not sure I see the benefit in this case). That ADC is well suited to the main AMO use case. Moreover, it's used on Sampler so we should have it well characterised before we start work on Sayma v2.0. That helps a lot with risk mitigation

gkasprow, 2017-12-03:
@hartytp we can go for it with existing approach I described. ADC would get config data from RTM FPGA but data would be transferred to the AMC FPGA via existing 5 LVDS pairs.
But in such situation will we preserve SMPs?

hartytp, 2017-12-03:
> @hartytp we can go for it with existing approach I described. ADC would get config data from RTM FPGA but data would be transferred to the AMC FPGA via existing 5 LVDS pairs.

What config data? IIRC, the ADC doesn't have an MOSI line or any registers to be configured. The only thing is the DDR/SDR pin, which we should probably connect to the RTM FPGA + a jumper as on Sampler.

Assuming we have enough pins on the RTM FPGA, we can follow @dhslichter's suggestion and route all 10 unused differential pairs (or, at least as many as we have FPGA pins for) to the RTM + AMC FPGAs. That gives us quite a bit of flexibility in how we set the ADC up. NB however, we do need to be careful with FPGA fin assignments: must assign clock capable pins correctly; should try to put CLK_OUT and SDOA-D on similar parts of the FPGA to simplify timings (ideally put those close to the lines going to the AMC<->RTM connector as well for the same reason).

> But in such situation will we preserve SMPs?

I'm afraid that we will have to. But, if that's the only decent way of adding a LVDS ADC to Sayma then we can live with it!

Hopefully the new RF connectors you found will make this a non-issue.

hartytp, 2017-12-03:
If the ADC is on Sayma RTM then we will probably want to route the signals from the AFE mezzanine differentially. That means that the AFE mezzanine must include the dividers + OpAmp buffer for single-ended to differential conversion + offset addition.

Is there enough space on BaseMod for this? Elsewhere, I've made a couple of proposals that should help to free up space on BaseMod. No one has objected to these yet, so I assume everyone is okay with them!
1. Only have 2 AFE mezzanines rather than the current 4, so we have 1 mezzanine per DAC. This frees up space by allowing more sharing of components like regulators/references/etc between input/output channels. It also frees up space by reducing the amount of dead space around/between the PCBs. It should also reduce the cost of the mezzanines a bit and help to reduce some of the mechanical issues we've seen.
2. Remove the power detector from the RF part. It's actually the biggest component on the PCB and IIRC caused @sbouhabib some issues fitting it on. AFACIT, it's not that useful anyway!

@gkasprow Do you think that with these changes we will have enough room to replace the BaseMod input path with one identical to Sampler (we can drop the CMCs if there isn't space for them)?

gkasprow, 2017-12-03:
@hartytp I'd keep existing mezzanine connector layout just in case somebody wants to use 4 individual modules. This does not stop us from using double width 4-channel AFE boards.
I think we will manage placing amplifier part used on Sampler module. 
Sampler module is single-sided PCB. I just did some simulation with Allaki and Samtec connector:
[image]
[image]

gkasprow, 2017-12-03:
@hartytp we have plenty of space. The CMS we use right now are too high, but low profile versions are available.

hartytp, 2017-12-04:
> @hartytp I'd keep existing mezzanine connector layout just in case somebody wants to use 4 individual modules. This does not stop us from using double width 4-channel AFE boards.
> Sampler module is single-sided PCB. I just did some simulation with Allaki and Samtec connector:

Good to know that this can be made to work! IIRC, the current BaseMod (née Allaki) form factor doesn't work because the clearance between boards is too small so they don't actually fit together on Sayma. Is that fixed in your design above?

> @hartytp I'd keep existing mezzanine connector layout just in case somebody wants to use 4 individual modules. This does not stop us from using double width 4-channel AFE boards.

Okay, if people feel strongly about this then we can keep the existing form-factor.

FWIW though, I'd still prefer to move to a 4-channel AFE form-factor (1 AFE per DAC). IMHO, the current 2-channel mezzanines provide a more fine-grained control over the AFE than any user actually needs/wants. While having more flexibility is always great in principle, it comes with cost/complexity/mechanical tradeoffs. So, I'd prefer to optimise the design for 4-channel mezzanines than to make compromises to support unnecessary flexibility.
 
- The multi-pin RF connectors are still quite expensive. If we move to a 4-channel AFE, we can use a single 20-channel RF connecter (2x10 pin) per AFE as well as a single higher-pincount digital connector. This reduces the costs, which is crucial since I suspect that the high cost of buying a fully populated Sayma v1.0 will be a big hinderance in getting new users on board.
- Even if we keep the connectors on Sayma to support 4 2-channel mezzanines, IMHO we should design BaseMod/MixMod v2.0 to be 4-channels each. Again, this reduces costs quite a bit, and will make things mechanically much better. Moreover, it will make these mezzanines less dense/cramped. This will make modifying them later/adding new features quite a bit easer, which is good (super dense boards are much harder for users to hack). Seriously, the density on these boards atm is a bit crazy.
- If we do want to make our default mezzanines 4-channels each, we should also design the Sayma front-panel to have just a pair of cutouts (rather than having the mezzanines have two front panels attached to them)!
- For building 4-channel mezzanines, I also suspect that having a single pair of RF/digital connectors will work out mechanically better than having 2 RF connectors + 2 digital connectors (fewer tolerances to worry about/things to be misaligned).

dhslichter, 2017-12-04:
I am fine with switching to two AFE boards per Sayma, each serving 4 channels.  I think the main rationale we had in the beginning was that the more channels you have per mezzanine card, the more likely you were to waste some because you'd want different combinations of output and input (baseband, up/downconverting, etc).  If we are talking about having cheap SPI ADCs on the mezzanine cards then it's not such a big deal.

Back in the dark ages, my original proposal had been to have both the DACs and the ADCs on a mezzanine (FMC basically), for the same sorts of connector signal integrity reasons @hartytp has discussed above.  We eventually came to the RTM solution as the way to separate the main FPGA from the DACs and ADCs, with a separate board for the analog front end as well.  So just to give a little reminder of how we got where we are with some of the design decisions.

> Let's generally prefer well designed solutions over shortcuts. As far as I can tell the urgency comes mostly from people needing a AWG, and ignoring/DNPing the current ADC would be fine there.

I am firmly with @sbourdeauducq here.  I think it is possible to design a system where we DNP the JESD ADCs, leave the analog connection to the AFEs for both DACs and the (DNPed) ADCs, and then the option to have an SPI ADC on the AFE is up to the user.  If @gkasprow can scrounge 4 or 5 LVDS pairs between the AMC and RTM, one can write a pretty simple gateware core that takes several SPI buses on the AMC FPGA, packages them onto the LVDS lines to the RTM FPGA (maybe 4:1 or 8:1 SERDES if desired, with one LVDS pair required for SYNC), and then unpackages them at the RTM FPGA into their respective buses to the AFE cards.  This has several very nice features:

- transparent for user hacking, they just send/receive the data they want via SPI on the AMC FPGA
- hardware remains general purpose -- we are not using up AMC-RTM pins to send specific signals for a given AFE configuration with a hacked-on chip SERDES or some such.  In the end, it's just more AMC/RTM connectivity, which is good, and allows variants to be made by reflashing FPGAs instead of respinning boards.  
- JESD ADCs (the current model or another one) can be implemented simply at a later time for those who might want them.  I point out that the Magtrap experiment is one such experiment -- all this talk of ion trappers not wanting high-latency ADCs ignores those interested in feedforward rather than feedback.  

I understand that certain customers are in an enormous rush to get boards in the door and working.  DNP the JESD ADCs, but do a reasonable (and simple) job of the AMC-AFE SPI communications that won't come back to haunt everyone when needs or wants change.

dhslichter, 2017-12-04:
> NB however, we do need to be careful with FPGA pin assignments: must assign clock capable pins correctly; should try to put CLK_OUT and SDOA-D on similar parts of the FPGA to simplify timings (ideally put those close to the lines going to the AMC<->RTM connector as well for the same reason).

@hartytp I don't think you need clock-capable pins for SPI at 50-100 MHz.  Comments @sbourdeauducq @jordens ?

hartytp, 2017-12-04:
> I am fine with switching to two AFE boards per Sayma, each serving 4 channels.

:+1: 

>  I think it is possible to design a system where we DNP the JESD ADCs, leave the analog connection to the AFEs for both DACs and the (DNPed) ADCs

DNPing the ADC is fine by me -- particularly if it really is quicker than actually removing it -- so long as it doesn't lead to compromises elsewhere.

> and then the option to have an SPI ADC on the AFE is up to the user.

AFAICT, it's going to be far easier to just add an octal ADC onto Sayma. We just provide solder jumpers to switch between the ADC and let the user decide on the population options.

dhslichter, 2017-12-04:
> AFAICT, it's going to be far easier to just add an octal ADC onto Sayma. We just provide solder jumpers to switch between the ADC and let the user decide on the population options.

This forecloses on anyone being able to use a high-speed ADC.  JESD latency is not an issue for all users, and if the problem of "I need an ADC yesterday, 1 MSPS is plenty" can be solved simply by putting it on the AFE and not foreclosing on other users who would like the JESD ADC, it seems like an overly crude step at this stage.

hartytp, 2017-12-04:
> This forecloses on anyone being able to use a high-speed ADC

Why? Just put solder jumpers to allow users to select between the two ADCs. Then users have two options which are selected between by population: fast JESD ADC; or, MSPS LVDS ADC. Isnt' that idea?

We're already using solder jumpers on BaseMod for GHz signals, so this should be fine from a signal integrity POV AFAICT.

We can route all the LVDS lines for the ADC through the FPGA, so users who don't want it can use them for something else.

Note that if we do decide to remove the clock mezzanine, that will free up some IO on the RTM FPGA.

gkasprow, 2017-12-04:
@hartytp @dhslichter We already have clock capable pins within non-used LVDS pairs in the RTM connector from AMC side. And free LVDS lines are already within same RTM FPGA bank.
I'm not sure if it is the best idea to create that many assembly variants. Testing them is a nightmare. Two variants is already a lot.
Little QFN ADC does not consume much space on AFE board, we get rid of expensive SMP connectors so it additionally lowers the cost and adds space for such ADCs :)
Moreover, we can place even high performance ADC, i.e. 10MS/s 18bit chip with LVDS outputs.
The only change is rearrangement of mezzanine lines so they are LVDS pairs instead of single ended ones.

hartytp, 2017-12-04:
> I'm not sure if it is the best idea to create that many assembly variants. Testing them is a nightmare. Two variants is already a lot.

I agree! Let's have as few variants as possible!

With the ADC, I only see the need for two variants:
1. JESD fast ADC
2. LVDS slow ADC

The slow ADC is so cheap that I don't think we need a third variant with no ADC.

hartytp, 2017-12-04:
@dhslichter I haven't fully understood your proposal yet.

- IIRC we've agreed that most users don't want the expense (close to $1k), power consumption or complexity of the JESD ADC. As a result, this will be DNP in the main variant that we produce (the one that's bought by the majority of ion trappers).
- The use-case that you highlight for this ADC is a non-feedback, right? i.e. you look at your pulse envelope and then apply a correction to flatten it. That's something you can do equally well using a fast ADC on another board (or even a scope). It doesn't need to be tightly integrated onto Sayma.
- Are you proposing that for this use-case you will produce a small batch of Sayma with the JESD ADC populated, then fund M-Labs to write a JSED Rx core? That's a lot of money for something you're only going to use in a pretty niche use-case, and will take quite a bit of time to get up and running. Wouldn't that money be better spent funding a specialized uTCA fast ADC board, which would be excellent for your use case AFAICT?

dhslichter, 2017-12-04:
@hartytp points taken....just not sure about the cost of JESD Rx core development vs starting over on a brand new ADC board.  There is also the cost/channel consideration -- there is a substantial overhead cost per RTM/AMC pair far in excess of the ADC cost.  Thus having an extra AMC/RTM card for the purpose of the ADC is going to cost a lot more than if you are already piggybacking on another card.

hartytp, 2017-12-04:
But, anyway, I don't care overly much about having an LVDS ADC on the AFE versus on Sayma. I can see arguments either way.

> If @gkasprow can scrounge 4 or 5 LVDS pairs between the AMC and RTM, one can write a pretty simple gateware core that takes several SPI buses on the AMC FPGA, packages them onto the LVDS lines to [...]

Personally, I'd call that a hack that's far better avoided!

----

If we decide to go for an ADC on the AFE, my preferred way would be this:
- Even if we leave the JESD ADC lines in place, we currently have 10 spare differential lines on the AMC<-> RTM connector.
- If we go for 4-channel AFE mezzanines, that gives us 5 LVDS lines per mezzanine to play with (if we can, we should route these to the mezzanines via the RTM FGPA).
- We can put a LTC2324-16 quad 16-bit 2MSPS ADC on the AFE mezzanine.
- This has a really flexible interface. If we route via the RTM FPGA then we have several different ways we can operate it, depending on what speed we actually want to achieve. One option is:
  - DDR, 125MHz clock (i.e. run off RTIO clock), 2 lane LVDS interface (SDOA, SDOC). This gives 2MSPS.

hartytp, 2017-12-04:
hmm...my current favorite is the LTC2325-16 on the AFE mezzanine. Objections? Let's make a decision about this and move on.

hartytp, 2017-12-04:
> @hartytp I don't think you need clock-capable pins for SPI at 50-100 MHz. Comments @sbourdeauducq @jordens ?

Oh, and NB that we want the LVDS to be good up to at least 150MHz DDR (see the ADC datasheet), so cc pins are probably a good idea.

hartytp, 2017-12-04:
Okay, I've looked over the datasheet for these 3 ADCs (1MSPS octal, 2MSPS quad and 5MSPS quad). 

They all look sweet. Analog performance is the same for all three.

I can only see one gotcha: the 5MSPS ADC does not support a 2-lane mode. Thus, to read out all channels, one has to use all 4 SDO lines. Assuming we need to use the CLKOUT from both ADCs to meet timing, we'd require 7 lines per ADC (SDOA-SDOD, CLKOUT, CNV, SCLK).

So, unless there are any other lines on the RTM<->AMC connector that we can repurpose (are there @gkasprow), we have a two options:
a. go for the 2MSPS ADC
b. go for the  5MSPS ADC and have the RTM FPGA generate the CNV and SCLK lines (program via the wb-ser bus)

My current preference is probably (b), but I'd like to hear from @sbourdeauducq and @jordens about how easy this will be in practice.

Note that the 5MSPS and 2MSPS ADCs are pin-compatible, so we can always change our mind later.

gkasprow, 2017-12-04:
@hartytp Even if we use  all Rx lines in RTM connector, we are still short with AMC FPGA pins.
So I'd prefer option (b) + some trivial IOSERDES or DDR code in the RTM FPGA and AMC FPGA.
Since we have common clock distributed to both FPGAs, we don't even need to use clock capable pins (we have one). Kintex US IOSERDES are quite powerful, comparable to early rocketIO transceivers in Virtex 2 family :) If we don't want to use IOSERDES, we can simply pack 8 data lines to 4 LVDS channels via trivial DDR buffers + 1 sync line. In this way we would be able to use new RTMs with existing Sayma AMC boards.

hartytp, 2017-12-04:
> @hartytp points taken....just not sure about the cost of JESD Rx core development vs starting over on a brand new ADC board. There is also the cost/channel consideration -- there is a substantial [...]

Point taken ;)

My feeling is that Sayma is primarily an RF generating board. It's never going to be a fast ADC->DAC feedback board because of the JSED latency. So, we're far better off trying to focus on making Sayma really good at its primary mission and keeping it as cheap as possible than trying to overload it with features that distract from what it's designed for.

In the abstract, if you need a fast digitizer then it is cheaper to add it to Sayma than to make a separate board. However, in principle it may not work out like that: if you add the fast ADC to the main Sayma variant then you're wasting money every-time you **don't** need a fast digitizer (which is most times AFAICT). That cost adds up quickly. And, conversely, if you have the fast ADC as a population option then you have to deal with all the exepense and hassle of maintaining, stocking and supporting multiple variants. This can cost you too.

So, the more I think about this, the more I'm convinced that we're best getting each board to have 1 purpose and optimizing for that. Let's just keep Sayma as an AWG.

hartytp, 2017-12-04:
> @hartytp Even if we use all Rx lines in RTM connector, we are still short with AMC FPGA pins.

How many AMC FPGA pins do we have left?

hartytp, 2017-12-04:
> we don't even need to use clock capable pins

Don't we? These fast ADCs have a source-synchronous interface, so we need to clock the FPGA input logic from the ADC CLK_OUT signal (otherwise, we don't know the phase of SDO signals relative to the FPGA clock).

> If we don't want to use IOSERDES, we can simply pack 8 data lines to 4 LVDS channels via trivial DDR buffers + 1 sync line

hmmm....that's quite a bit more complex. It also messes up the timings of these signals w.r.t. the ADC CLKOUT signal. I'd rather not get into that if we can avoid it!

hartytp, 2017-12-04:
If we're really short on IO, can we do something like share some of the DIO with one of the FMCs? (e.g. put in a mux that switches LVDS lines between the FMC and the AMC<->RTM connector?).

hartytp, 2017-12-04:
Or, can we put some solder jumpers in on either the RTM/AMC so that the ADC JSED lines can be used either as JSED lines or as LVDS?

hartytp, 2017-12-04:
There do also seem to be a few FPGA lines we can potentially re-purpose from Sayma AMC (feel free to add):
- SYSCLK1_300
- SYNCOUT21 (not used)
- SYNCOUT11 (not used)

My guess is that we can find 10 LVDS pairs to route to the AMC<->RTM connectors if we look properly.

sbourdeauducq, 2017-12-05:
> one can write a pretty simple gateware core that takes several SPI buses on the AMC FPGA, packages them onto the LVDS lines to the RTM FPGA (maybe 4:1 or 8:1 SERDES if desired, with one LVDS pair required for SYNC)

So you want a roll-your-own JESD204 ;)
@dhslichter What do you think of the cheaper and lower-power JESD ADC I was proposing?

sbourdeauducq, 2017-12-05:
> we get rid of expensive SMP connectors

A quick search turned up $7 bullets. That's still expensive, but compared to the cost of the boards not so much - and maybe there are even cheaper parts...

sbourdeauducq, 2017-12-05:
> But, anyway, I don't care overly much about having an LVDS ADC on the AFE versus on Sayma.

It's an *analog front-end*, if possible the ADC shouldn't be on it.

hartytp, 2017-12-05:
> So you want a roll-your-own JESD204 ;)

;)

Agreed though: we should try to avoid this kind of trickery if possible. However, we don't need to do this at all. Either putting an octal ADC on Sayma or putting a quad ADC on each AFE gives us a very compact and simple LVDS interface, which easily fits on the pins available on the AMC<->RTM connector.

> @dhslichter What do you think of the cheaper and lower-power JESD ADC I was proposing?

Thanks for suggesting alternatives @sbourdeauducq! Personally, I don't think this is well suited to our designs:
- It's still a complex JSED interface, with all the drawbacks that entails, such as being not easily hackable by users.
- It's sample rate still doesn't really match our use-case IMO. It's far faster than we want for basic servo-loops (while still having huge latency) and it's not that fast for use as a high-speed digitiser.
- As I said before, I think that putting this kind of ADC on Sayma is a mistake. Since it can't be used for feedback there is no real advantage of having on Sayma compared to on another board. Let's stick to making Sayma an RF generating board and make a separate fast digitiser board.
- It's still not cheap or low power (if we get into the mindset that $100 doesn't matter then Sayma will always be overprices for most use cases)
- It's much higher risk than either sticking with our current JESD ADC or adding an LVDS ADC (particularly if we're using the same LVDS ADC that's used on Sampler and hence already tested). This is a complex part that's unlikely to work perfectly on the first part and could lead to major delays with Sayma v2.0
- Its analog performance is not good enough. e.g. it's only 14-bit max, the DNL/INL aren't great, neither is gain accuracy. I didn't see an offset temperature coefficient on the spec sheet, the offset is pretty large (I don't think this is really designed to be a DC precise component). There is a reason that the other ADC is so expensive -- if you want speed **and** DC performance, that always costs you!

Why are you so keen on the JESD ADC when there is a nice LVDS alternative that meets our needs? We can always keep the JESD ADC as a DNPd population option in case it's needed in the future

hartytp, 2017-12-05:
> A quick search turned up $7 bullets. That's still expensive, but compared to the cost of the boards not so much - and maybe there are even cheaper parts...

There are two different questions being addressed in this thread:
1. Should we DNP the current JESD ADC and add an additional LVDS ADC somewhere, or should we replace the current ADC with a different JESD ADC. My preference (and, I think, the current consensus) is that we should go for the former.
2. If we do go for the LVDS ADC, should we put an octal ADC on Sayma main board, or a quad ADC on each AFE mezzanine. I don't feel too strongly about this.

> It's an analog front-end, if possible the ADC shouldn't be on it.

Let's not get too hung up on terminology. It's still an AFE even if it has a simple LVDS ADC on it -- all the fast digital stuff is on Sayma main board.

I don't see any major disadvantage of putting the ADC on the AFE board. It doesn't make the AFE less hackable (the major limit there is the fancy RF connectors and RF substrate).

For all the reasons we've talked about before, routing DC precise analog signals between PCBs is much harder than digital signals and generally best avoided. As well as the cost and mechanical issues, and the added flexibility gained by putting the ADC on a user-configurable mezzanine, there are issues like having to ESD-proof sensitive analog nodes (ever seen an analog part get popcorn noise due to ESD? It's not pretty and it's not easy to debug).

Seems to me that putting this on the mezzanine makes far more sense.

dtcallcock, 2017-12-05:
> Let's stick to making Sayma an RF generating board and make a separate fast digitiser board.

Could someone perhaps comment on pre-existing FMC solutions (eg. [FMC ADC 250M 16B 4CHA](https://www.ohwr.org/projects/fmc-adc-250m-16b-4cha/wiki/wiki)).  Specifically what is the per-channel cost once you've added a uTCA carrier and how much work is it to write the gateware to make it Sinara/Artiq compatible (if that's possible without hardware redesign)?

hartytp, 2017-12-05:
Doesn't Sayma AMC have an FMC that you could plug that into? Or does that not work because of a lack of transceivers on Sayma AMC?

hartytp, 2017-12-05:
Scratch that, it's LVDS. Is there any reason that can't work with Sayma AMC, because that would be a really sweet addition.

hartytp, 2017-12-05:
One other thing I like about @dtcallcock suggestion: most users **either** need a DC-precise ADC or a fast ADC, but generally not both. By having a single ADC on Sayma, we have to try to cater to both, which makes things expensive. Putting a slow ADC on the AFE mezzanine and a fast one on an FMC allows everyone to find an ADC for their needs without overcomplicating Sayma, which is not primarily designed as an ADC.

sbourdeauducq, 2017-12-05:
@hartytp Again, please stop renaming boards and especially not Novo since there was a vote.

dtcallcock, 2017-12-05:
https://en.wikipedia.org/wiki/The_customer_is_always_right

[84 more replies]

closed 2019-03-06 by hartytp