sinara-hw/sinara #402: BaseMod RF connectors

hartytp, 2017-12-01:
- Not enough space on front panel for SMAs
- High density connectors don't really make sense, since: the load almost always has an SMA on it, so breakout is always needed; they're either very expensive, or have surprisingly poor RF performance; they're not trivial to get hold of.
- Best option is SMP or MMCX. Probably leaning towards SMP, but final decision required. Deciding factors are likely mechanical robustness and cost.
- Should ship short (say, 10cm) pigtails to convert our chosen connector to SMA with each BaseMod

cf #7

gkasprow, 2017-12-01:
What about SMB connectors? They can be densely packed.

dtcallcock, 2017-12-01:
If we switch the ADC inputs to a small flush-mounting connector like MMCX there will presumably be enough room for SMAs.  

I want to use this board to generate 1.7GHz with good long term channel-channel phase stability.  Therefore I would like a connector that doesn't rotate or 'float'. 

Will also prevent accidentally mixing up inputs/output.  Something that'll be easy to do on a cramped panel.

dhslichter, 2017-12-01:
SMA for DAC outputs, MMCX for ADC inputs (only using 1 MSPS ADCs on the BaseMod board) sounds like a good plan to me.  Will they fit @gkasprow?

gkasprow, 2017-12-02:
@dhslichter MMCX connectors are much smaller so they will fit easily. Moreover they don't need any additional space for a wrench so once you unplug them, access to SMAs will be far easier.

gkasprow, 2017-12-08:
I just got replay from Samtec: 
> I’m sorry for my delayed response.
> There are no plans to release the -03.0 lead style for our IsoRate product line.
> The NRE for tooling would be greater than $50K.

It seems there is no way to use these connectors.
But, we use FMC modules with same AMC front panels and it fits. The onlly difference is that surface mounted SMA connectors cannot be used.
So if we adopt FMC or PMC front panels idea we could use these samtec connectors as well.

dhslichter, 2017-12-08:
Ouch!  That's not what you like to hear from Samtec...

IIRC once of the issues was component height on the bottom side of the BaseMod board, which restricted the stack height of the connectors.  Can we get away with 1 mm less clearance on the bottom side?

hartytp, 2017-12-08:
> But we use FMC modules with same AMC front panels and it fits. The stacking height is 10mm. The only difference is that surface mounted SMA connectors cannot be used. Right angled versions fit well.
So if we adopt FMC or PMC (slightly wider) front panels idea we could use these Samtec connectors as well.

So, to check I understand you:
- The current stacking height on the AFE mezzanines is 9mm
- 9mm stacking RF connectors aren't available from Samtec
- 10mm are connectors readily available 
- 10mm is a nice standard height anyway -- e.g. it's used by FMC

As @dhslichter says, the only potential issue I see with the 10mm height is that it gives us less room on the side of the AFE mezzanines opposite the connectors. This is the side with the input components on. AFAICT, these are all low profile components (with the possible exception of the CMCs, but those are low priority components which can easily be scrapped) apart from the SMAs. So, as long as you can find decently priced SMAs that fit, I'd say let's go for the 10mm stackup.

gkasprow, 2017-12-08:
@hartytp standard, low cost right-angled SMAs work just fine.
But their matching is far worse then SMT ones we use right now.
[This](https://www.ohwr.org/projects/fmc-adc-130m-16b-4cha/wiki/wiki) is example of a card I designed some time ago.
At that time I used SMT SMA connectors, but they are too fragile so were replaced by THT version.
[There](https://www.ohwr.org/attachments/3388/bot_r.jpg) is a place on the bottom side of the board. 2mm components fit easily

hartytp, 2017-12-08:
> @hartytp standard, low cost right-angled SMAs work just fine.
But their matching is far worse then SMT ones we use right now.

Anyway, this isn't a problem right, as we've already decided to replace those SMA connectors with MMCX, right (the bottom-side connectors are the input ones, right)?

Even with the JSED ADCs, this is only a 100MHz input, so cheaper SMAs should be fine here if one really wants to go that way. Or, SMPS would fit as well.

So, my vote is to go for the 10mm stackup. It's more standard than 9mm which is nice, and it leaves more room for screening on the critical RF parts on the top side of the board.

hartytp, 2017-12-08:
Unless there are any objections to the 10mm stackup, I'll update the issue summary to reflect this decision.

dhslichter, 2017-12-08:
> @hartytp standard, low cost right-angled SMAs work just fine.
But their matching is far worse then SMT ones we use right now.

I think the point is that with a 10 mm stackup, we have to use these angled SMAs for the DAC outputs -- cannot use the current surface-mount/cutout SMAs for the DAC outputs.  And since we care about the connector performance, this is less desirable, especially if we'd like to use the same kind of component on MixMod.  

@gkasprow how about something like https://www.digikey.com/product-detail/en/cinch-connectivity-solutions-johnson/142-0771-831/J808-ND/726115 or https://www.digikey.com/product-detail/en/cinch-connectivity-solutions-johnson/142-0761-801/J797-ND/726105?  This would stick out another 3 mm past the RTM-facing surface of the mezzanine card, thus a total of 10 mm (stackup) + 3 mm + 0.8 mm (half of RTM thickness) is 13.8 mm from the RTM centerline.  According to the spec it looks like we can do this mechanically if we are prepared to notch the faceplate in the areas where mezzanine cards would come through.  This is probably very nice to do anyway to help with ease of installation.

[image]

gkasprow, 2017-12-08:
@dhslichter we use this kind of SMA connector already on the RTM board.
Anyway, almost all SMA THT connectors have same position of the hot signal pin with respect to the PCB surface. Good matching is achieved when one does not bend the signal path. Main difference is how they are fixed to the PCB edge.
Maybe there are some high performance RA SMT SMA connectors?
What about [this](https://www.pasternack.com/sma-female-standard-thru-hole-pcb-connector-pe4545-p.aspx)?  Would such performance be OK?

dhslichter, 2017-12-08:
@gkasprow are the dimensions/plan I outlined above not workable?  The connectors you mention could potentially work, but they are considerably more expensive (~$18/channel more than the regular edge mount SMA) and they probably won't have as good performance -- although they are spec'ed up to 12.4 GHz.

gkasprow, 2017-12-08:
@dhslichter let me do some quick simulation. It will be hard to install insulation washers due to the AMC edge.
[image]
[image]

gkasprow, 2017-12-08:
And we need bulkhead connector to attach them to the front panel.

dhslichter, 2017-12-08:
Another consideration: if instead of using Samtec RF connectors plus a separate pin header, we went to a single connector per mezzanine using some high-speed connector?  If we used an FMC connector with 7 mm stackup, the test reports can be found here:
http://suddendocs.samtec.com/testreports/hsc-report-seam-seaf-07mm_web.pdf
http://suddendocs.samtec.com/testreports/hsc-report-sma_seamp-seaf-07mm_web.pdf

We'd be running the DAC signals as differential, could surround them with ground pins, and you can then achieve very low crosstalk (70-80 dB below 1 GHz even for nearby pairs, could isolate more since there are so many pins you can move the other signals farther away).  This connector has enough pins to send in all 4 DAC signals as differential pairs, plus a bunch of nice differential pairs for LVDS signaling from an ADC, plus lots of leftover pins for power, slow single-ended logic, etc.  We would just need to take care to place these other pins physically farther away, and ground the various intervening pins to create an effective crosstalk shield around the DAC traces.  

If you go this way, though, you have a single connector to attach the AFE to the RTM board.  Then you can make the stackup 7 mm or 8 mm instead of 10 mm and all these issues with the SMA are resolved as well.

dhslichter, 2017-12-08:
> And we need bulkhead connector to attach them to the front panel.

@gkasprow I was envisioning that the front panel would be held on to the board by other means (screwed in to a little right-angle block, perhaps), and that you'd just use something fairly thin.  Then you just poke the SMA through the middle of the front panel but it doesn't touch or have mechanical attachment to the fron panel.  The entire front panel is a small part that then contacts the RTM front panel with an EMI gasket around the edges.  No need for insulating washers then as the SMA outer shield is not touching the front panel.

gkasprow, 2017-12-08:
@dhslichter I remember our discussion a year ago and this type of connector was disqualified. Maybe it was done too early. We don't need FMC connector but smaller from Searay family. 

It would be nice to reuse standard PMC front panels which are slightly more narrow than FMC.

gkasprow, 2017-12-08:
@dhslichter I saw today that our RF guys are using FMC connectors for real RF stuff. I just saw some random PCB, need to ask for details. Maybe @sbouhabib know more details..

dhslichter, 2017-12-08:
> @dhslichter I remember our discussion a year ago and this type of connector was disqualified. Maybe it was done too early. We don't need FMC connector but smaller from Searay family.

I remember that too -- I think the issues were that we wanted crosstalk as low as possible, and to have the AFE boards be simple/user-solderable (which SEARAY is not really).  I think that given our other options we are debating, the crosstalk is in the same ballpark.  The user-solderable AFE was a nice idea but SEARAY is easy to have a boardhouse assemble for you reliably, so I don't think it's a big issue.

dhslichter, 2017-12-08:
> @dhslichter I saw today that our RF guys are using FMC connectors for real RF stuff. I just saw some random PCB, need to ask for details. Maybe @sbouhabib know more details..

That would be awesome, I am sure they will have some useful opinions.  We only need to have signals running through up to ~3.5 GHz at the absolute maximum.

hartytp, 2017-12-08:
I'd still be a bit worried putting about putting the LVDS bus and DAC signals on the same connector. Might be fine depending on how they are distributed in the connector. Would probably need some data/prototyping in any case. But, yes, would be really interested to hear from @sbouhabib about this.

hartytp, 2017-12-08:
> I think the point is that with a 10 mm stackup, we have to use these angled SMAs for the DAC outputs -- cannot use the current surface-mount/cutout SMAs for the DAC outputs. And since we care about [...]

Use SMPs on MixMod?

hartytp, 2017-12-08:
> We'd be running the DAC signals as differential, could surround them with ground pins, and you can then achieve very low crosstalk (70-80 dB below 1 GHz even for nearby pairs, could isolate more [...]

> If you go this way, though, you have a single connector to attach the AFE to the RTM board. Then you can make the stackup 7 mm or 8 mm instead of 10 mm and all these issues with the SMA are resolved as well.

This does sound quite appealing on many fronts: cost, insertion force, availability of components, etc. And, you're right, we can add a lot of ground pins to really stamp out the cross-talk. The test PCB you'd need to characterize this would be pretty trivial to make -- an FMC and a few SMAs.

dhslichter, 2017-12-08:
The standard FMC connectors would be nice @gkasprow from an availability standpoint.  We can do an 8.5 mm stackup, which fixes the SMA issue on the front panel.  

However, I can see value in choosing a connector which does not intermate with FMC, though (e.g. slightly fewer pins) so that people don't try to mate with an FMC (potentially blowing it up in the process).

gkasprow, 2017-12-08:
It's enough to change the gender of connectors.
Would there be any benefit to adopt FMC mechanical form factor? PMC would be too wide, but FMC should be fine both in terms of panel size. Board would be same length as current mezzanine is.
One can  fit there 4 SMAs and 4 MMCXs

gkasprow, 2017-12-08:
FMC panel requires 10mm distance, anyway, we can go for something similar. The advantage of FMC mechanical fixing is its rigidity.

hartytp, 2017-12-08:
@gkasprow Sticking to something as close to FMC sounds good to me.

How does the width of FMC compare to 2x the width of the current Allaki/BaseMod?

hartytp, 2017-12-08:
So long as the cross-talk/RF performance is good enough, adopting an FMC form factor and mounting sounds like a really good idea IMO.

gkasprow, 2017-12-08:
@hartytp the FMC connector will fit easily. FMC is 2 mm narrower than doubled Allaki/BaseMod.
Anyway we cannot use FMC connector as it is, so will have dedicated version machined  that is perfectly adopted to what we need.

hartytp, 2017-12-09:
> @hartytp the FMC connector will fit easily. FMC is 2 mm narrower than doubled Allaki/BaseMod.

Sorry, I meant how does the width of a FMC card (not connector) compare with a doubled Allaki/Base mod? Sounds like a FMC card is almost exactly the right size for us, which is good!

> Anyway we cannot use FMC connector as it is, so will have dedicated version machined that is perfectly adopted to what we need.

Not sure I follow you here. Do you mean having a custom FMC connector made? Or, do you mean having a PCB made that has a slightly different shape to a standard FMC card?

hartytp, 2017-12-09:
@gkasprow Are you okay to make a simple test PCB to see if the FMC or similar connector will work for us.

- [ ] Choose a sensible pin assignment for this, including
  - 4 differential DAC channels
  - 4 differential JSED ADC channels
  - LVDS bus for AFE ADC
  - Power, ground
  - The misc digital lines we have on the current AFE (SPI for attenuators, TTL for switches, I2C, etc)
  - Make sure connector pinout is designed for minimum cross-talk. e.g. space LVDS far from DAC and surround each DAC with as many grounds as possible.
  - How do you make FMC 50Ohm? If you surround the DAC channels by ground, does that create a 50Ohm transmission line?
- [ ] Make simple test PCB with our chosen connector (e.g. FMC) and SMAs for the DAC/ADC/LVDS lines. Then we can measure insertion loss/return loss/cross-talk between all channels.

With that information, we can decide if this is good enough.

dhslichter, 2017-12-11:
@gkasprow @hartytp I think we should use a connector pair from the SEAM/SEAF series, but not with the same pin count as FMC, so that people do not mistakenly populate a Sayma AFE into a FMC slot or vice versa.

hartytp, 2017-12-11:
@dhslichter Sure, anything like that is fine by me.

I'd love it if these connectors worked out, since they'd be great from a mechanics, cost and availability stand point.

FWIW though, I'm still not convinced by the RF performance of those connectors. The data on the links you sent didn't look great. e.g. cross-talk was -30dBc at 12GHz, so (do we assume 20dB/dec?) something like -50dBc at 1GHz. We'd probably want more than 80dBc at 1GHz, and it's not clear to me that spacing the pins we use out a bit more + adding some more ground will really do that. Simillaryly, the impedance control looked a bit poor.

Still, will be interested to hear from @sbouhabib or to see some measurement results.

dhslichter, 2017-12-11:
>  The data on the links you sent didn't look great. e.g. cross-talk was -30dBc at 12GHz, so (do we assume 20dB/dec?) something like -50dBc at 1GHz. We'd probably want more than 80dBc at 1GHz, 

Just look at the plots here rather than extrapolating.  We are more concerned with FEXT than NEXT because the outputs of the DAC should be reasonably well matched and able to handle low-level interference.

>  it's not clear to me that spacing the pins we use out a bit more + adding some more ground will really do that.

Again, below shows how things have worked with several different combinations of pin pairs, so you can see that ground/distance is not everything, but there are some combinations that work better than others.  Making a test board would be the thing to do.

[image]

[image]

>  Simillaryly, the impedance control looked a bit poor.

If you're trying to send random digital signals at 12 GHz, sure.  However, we're talking about signals at max 3.5 GHz, with slow envelopes on them (tens of ns timescales at the very fastest).  I'd argue here that the connector return loss is a relevant spec here as well as impulse responses.  Your point is taken, but if you see how this stacks up against the IJ5/IP5 we are considering as the alternative, it all looks pretty similar: http://suddendocs.samtec.com/testreports/hsc-report_ij5-ip5_web.pdf .  The crosstalk is a bit better with IJ5/IP5 but the return loss is same/worse and the impulse response is similar.  

In the end, if "cheapo" performance won't cut it, we're stuck with SMP.  Furthermore, if we do end up going with IJ5/IP5, we have to use right-angle SMA connectors at the panel and those definitely have worse impedance properties, so you have to choose your poison a bit.  

[image]

gkasprow, 2017-12-11:
@hartytp Let's talk about FMC connector testing. To make it happen and test with SMAs one need to add some form of matching. I'm not sure what impedance such FMC has. Is it 50Ohm or let's say 60Ohm and it is optimised for diff operation where 2x60 Ohm gives 100Ohm.
What exact scenarios do you want to test?
I can use low cost FR4 prototyping service where boards are printed using shared documentation.
@dhslichter I'd like to use identical connectors as used for FMC but reverse the gender. I'm pretty sure that no one will try to plug Sayma AMC to Sayma RTM then :)
If we want to connect all lines to the SMAs, there are plenty of them.
So I will take 400pin connector, one side will reserve for LVCMOS and LVDS signals, then array of GNDs, later on DAC , array of GNDs, and  JESD pins.
Who wants to measure it?

dhslichter, 2017-12-11:
> @dhslichter I'd like to use identical connectors as used for FMC but reverse the gender. I'm pretty sure that no one will try to plug Sayma AMC to Sayma RTM then :)

Sounds good to me!

> Who wants to measure it?

Probably we'd need to look over some schematics first :)  Manufacture with the same stackup (at least for the top layer thickness) as the RTM card.  Would need to have some baluns on the board to convert between differential and single-ended (network analyzer excitation will be single-ended).  Looking at the Samtec test board layout/design might inform some of this.

hartytp, 2017-12-12:
@dhslichter Okay, looking at that data more closely, I agree it looks pretty good in terms of RL and Xtalk for the differential lines.

One thing that wasn't clear to me: what are all the pins labelled T in the test setup? 50R terminated? Or just grounded.

> I'm not sure what impedance such FMC has. Is it 50Ohm or let's say 60Ohm and it is optimised for diff operation where 2x60 Ohm gives 100Ohm.
What exact scenarios do you want to test?

From the polts in the link @dhslichter shared, it looks like a pair of adjacent pins (either horizontal or vertical) surrounded by ground should give a good 100Ohm transmission line. So, matching from a 50Ohm feed in just needs a 2:1 balun. Say TCM2-43X+.

> I can use low cost FR4 prototyping service where boards are printed using shared documentation. 

Worth going to FR408 or similar to have good controlled impedance traces? Also, not to self, let's have a calibration signal on the PCB as well (SMA, trace, balun, trace, balun, trace, SMA) for good measure.

> So I will take 400pin connector, one side will reserve for LVCMOS and LVDS signals, then array of GNDs, later on DAC , array of GNDs, and JESD pins.

Sounds good. although what do you mean by "JSED pins"?
- Make sure to space the DAC and ADC lines out with as much ground as possible. 
- The plots in the files that @dhslichter uploaded suggest that diagonal spacings work best (see the difference between teh 136/126 and 117/107 cross-talk figures)
- Not sure if it will help to alternate "horizontal" and "vertical" routing stratergies?

As @dhslichter Says, I think the first thing to do is to is to propose a pinout for the connector.

> Who wants to measure it? 

If you design the board, I'm happy to do the tests.

--- 

Before we go too far down this line, it would be good to hear from Sammer on things like:
- Have they routed high-frequency analog signals on FMCs before?
- If so, how well did it work?
- Do they have any advice for how to do this? (e.g. how to space pins out to minimize cross talk)
- Is FMC the best connector to use for this, or is there a similar one they've found to work better?

hartytp, 2017-12-12:
> Not sure if it will help to alternate "horizontal" and "vertical" routing stratergies?

dhslichter, 2017-12-12:
> One thing that wasn't clear to me: what are all the pins labelled T in the test setup? 50R terminated? Or just grounded.

50 ohm terminated.

> Worth going to FR408 or similar to have good controlled impedance traces? Also, not to self, let's have a calibration signal on the PCB as well (SMA, trace, balun, trace, balun, trace, SMA) for good measure.

I think it's also worth having two such calibration traces running near each other for a bit (~few cm) so we can also measure the crosstalk from two open traces on the PCB -- I imagine the crosstalk there will be non-negligible as well.  

> The plots in the files that @dhslichter uploaded suggest that diagonal spacings work best (see the difference between teh 136/126 and 117/107 cross-talk figures)

That is true for FEXT but not for NEXT, where this is definitely the worst configuration; one has the issue that NEXT which is reflected by an imperfect match at the DAC effectively turns into FEXT, and at these levels we'd need the DACs to be matched to give >30 dB return loss.  I think we should make test boards with a couple of different signal traces which can be tested, and I would space them out maybe with a full ground surround and more than one "pair" away.

dhslichter, 2017-12-12:
> FWIW though, given how many other things are going on at the moment, making this test PCB should not be top priority right now!

Sayma 2.0 is currently on hold until this connector decision is resolved, so if lead times are an issue here we shouldn't neglect it too much...

hartytp, 2017-12-12:
@dhslichter ACK to all of that.

Would be good to build this test board whenever @gkasprow has time to do it (and to chat to @sbouhabib about it).

gkasprow, 2018-06-06:
Since we decided that will use SMA + MMCX, further discussion and tests are [here](Sayma RTM to AFE board-to-board connectors #386)

closed 2018-08-18 by gkasprow