sinara-hw/sinara #405: Sayma RTM clocking

hartytp, 2017-12-01:
There have been a lot of developments since we originally designed the clocking scheme for Sayma. Are there any changes we want to make for the v2.0 release?

**DAC clock**
- Is the HMC830 sufficiently good that we don't need to bother distributing a DAC clock via the BP/FP/ClockMod? That would be a big simplification.
- To do: look at HMC830 phase noise, both measured and specified. If this is close (say 6dB) to the DAC's noise floor then maybe we should rely on this as our sole DAC clock source.
- To do: look at HMC830 phase drift/temp co (needs measurement)

**LO**
- It's really nice to have a single LO input to Sayma, which is then supplied to the AFEs for upconverting etc. Having each upconverting AFE take its own LO input via the FP takes up a lot of FP space, which we don't really have, and relies on having a lot of external wiring and LO distribution.
- Is it enough to have this supplied via an FP SMA or the BP (use mux). Do we also need the option of a ClockMod mezzanine as in the current design?

**ClockMod**
- Does it still make sense to have this as a plug-in board to Sayma? Would it make more sense to have this as a Eurocard? Then, it can either supply the uTCA rack via Baikal, or an individual Sayma via a FP SMA or similar.

**Muxes**
- Summarize current clocking options, and think about what we might want to do differently in light of the above points.

hartytp, 2017-12-01:
Can someone remind me what REFSYNCIN is for on Sayma RTM, and whether we still need it? IIRC it's a fallback plan for synchronization, in case the M-Labs scheme doesn't work, so can be scrapped once that's working. Is that correct?

hartytp, 2017-12-01:
@gkasprow It's been a while since I looked at the RF BP schematics, but can you remind me how the two connectors work. AFACIT:
- We have the ERNI connector (J74), which is currently only used for the 100MHz clock, u_RFB_REF_CLK_IN, as well as the synchronisation signal, REFSYNCIN.
- Additionally, we have a Radial R694252107 RF connector, J73, which supports 3 differential signals up to 6GHz. 
- Can we use the Radial connector to carry REF_CLK, DAC_CLK and LO, and scrap the ERNI connector? AFACIT, that would be sufficient for all the signals we want to use, and would save some cost + routing.

hartytp, 2017-12-01:
@gkasprow Do we have any data on the performance (cross-talk and return-loss) of the ERNI connector as a function of frequency? If not, could you measure it for us, please?

This would be really helpful information, as it could potentially allow us to simplify things quite a bit:
- Currently the REC_CLK is distributed over both the ERNI and Radial connector. IIRC, this is because we were worried about reflections from the ERNI connector. If this is not a problem, then only distributing REF_CLK over the ERNI connector would allow us to:
  - uRFB_REF_IN signal (simplifies schematic + routing)
  - C667 C604 C605 R238 R239 R58 R59 TR2 
  - This allows us to remove a solder jumper + another clocking option, which would otherwise have to be explained properly in the documentation.
- If the ERNI connector is decent at RF frequencies up to a few GHz, we can consider using it to distribute the LO instead of the radial connector. 
  - This would allow us to DNP the radial connector, which is 5 times the price of the ERNI connector.
  -

gkasprow, 2017-12-02:
@hartytp the difference between these connectors is how the traces are routed.
The ERNI_973028 is for differential, digital LVDS signals .
The RADIALL_R694252107 is for single ended, high isolation clock routing.  That's why DAC, reference and the LO signals are routed there. The idea was to use common clock distribution mezzanine in the Baikal module instead of in every AMC. The clock distribution module in the Baikal is the same as in Sayma.
So it is not the connector who is limiting factor here but the RF backplane design.

hartytp, 2017-12-02:
> The ERNI_973028 is for differential, digital LVDS signals .
> The RADIALL_R694252107 is for single ended, high isolation clock routing.

Differential traces can be excellent for high-isolation clocks too (in fact, they're often preferred to single ended traces). Is the concern here that the routing of these differential traces is not done in a low-crosstalk fashion, so that they will provide worse performance than the Radial connector?

If the ERNI connector is bad for routing clock traces, then why do we have a clock trace routed on it? Can we just scrap that trace as well as the R58 R59 jumper?

Do we actually need the ERNI connector at all (we generate all the voltages we need locally using SMSPs AFACT), so can we just DNP it?

gkasprow, 2017-12-02:
@hartytp sure, we can DNP them by default.
All questions about RF backplane should go to @sbouhabib who actually was involved in the design.

hartytp, 2017-12-02:
Okay, I'd be interested to hear from @sbouhabib about the possibility of routing low-noise LOs through the ERNI connector.

But, assuming that we can't use the ERNI for clocks/LOs, I don't see any reason not to DNP it to save some money. We might as well also remove the REF_CLK line that connects to it as well as the solder jumpers to simplify the schematic a bit (at this point there are so many solder jumpers etc that the schematic is quite hard to follow, if we can simplify the design a bit, it makes the schematic much more understandable).

gkasprow, 2018-12-30:
Most of the discussion here is not relevant. RTM backplane was removed.

closed 2018-12-30 by gkasprow