sinara-hw/sinara #406: Sayma Regulators

hartytp, 2017-12-01:
@gkasprow Why do we have so many LDOs on Sayma*?? e.g. on the clocking page why is there a 3V3 regulator per component? Wouldn't it be sufficient to just decouple the components with LC filters? That would simplify the schematic, save some money and board space.

* I suspect that the answer to this question is that we ran out of steam on the design review before thinking too much about this, but maybe there really is a reason?

gkasprow, 2017-12-02:
@hartytp we have so many because they were used on the devkit and there were sufficient number of risks to mitigate. Some of them can be removed. Take into account that decoupling works well only beyond certain frequency, it also increases supply line impedance significantly. So if you need to keep the phase noise low, especially at low frequency offsets, it's better to keep these LDOs.

hartytp, 2017-12-02:
Okay.

Out of all the Sayma v2.0 issues raised so far, this is probably the least important, so if you'd prefer we can keep the regulators. In any case, some of them will disappear once we simplify the clock distribution network, so will become less off an issue.

FWIW though, I still think that many of these regulators are unnecessary. I can see the argument for having the DACs on their own regulators, since they can draw quite a bit of current with a very broad spectrum. However, I don't see that having a separate regulator for every clock buffer is required. These only really draw current at the clock frequency (plus some fixed bias current, which is very low noise), so LC decoupling is absolutely fine. Clock buffers just don't draw large low-frequency noisy currents.

If we can time, we can always test this using the Sayma v1.0 prototypes and then we'll know for sure.

marmeladapk, 2019-01-14:
I feel that this is no longer applicable:
- Clk distribution has separate LDOs for HMC830, ADF4356 and ADCLK948, which is fine by me,
- @hartytp said that DACs should (or at least can) their own regulators
- There are no other "duplicate" regulators and CLK distribution was vastly simplified

closed 2019-01-14 by marmeladapk