sinara-hw/sinara #408: Sayma v2.0

hartytp, 2017-12-04:
@jordens @sbourdeauducq if there are any other things you'd like to change from the Sayma v1.0 design, then please post them!

jbqubit, 2017-12-06:
Response to https://github.com/m-labs/sinara/issues/411#issuecomment-349134846. 

Thanks to everyone for working on Sayma and BaseMod design idea changes. And to @hartytp for organizing all the Issues. How to proceed will depend on cost (WUT, M-Labs) and risk of proposed changes. I don't expect there to be funding on the near term for another full prototype round of Sayma_RTM so this next round should be viewed as a production board. Risk should be weighed with that in mind. 

Delegation:
- @hartytp  Since you've been champion of many of these changes please do the following.
    - Confirm that proposed Sayma diffs to date are part of milestone [Sayma v2.0](https://github.com/m-labs/sinara/milestone/22)
    - Review each Issue that is beyond bug fixing and distill down to one or two paths as controversy dictates. Post that distillation to tail of each Issue. We will discuss in existing Issue threads. 
- @sbourdeauducq please comment on cost of M-Labs support (if any) required for each Issue. 
- @gkasprow  please comment on WUT cost and perceived risk for each Issue.

From my perspective it's not possible to make a final decision on hardware changes as significant as those on the current "wish list" until the existing hardware debugging is complete. Available budget will be allocated first for critical bug fixes required for analog output. Nailing down expected costs needs to happen soon (prior to final decision) due to long procurement lead time. 

If we're in rough agreement as to this approach, let's do the same for BaseMod changes. But with greater tolerance for high risk changes.

hartytp, 2017-12-07:
@jbqubit 

> How to proceed will depend on cost (WUT, M-Labs) and risk of proposed changes. 
> @sbourdeauducq please comment on cost of M-Labs support (if any) required for each Issue.

I don't think that any of the proposed changes directly affect M-Labs, so there should be no costs on that front.

e.g. if we add more LVDS lines on the RTM<->AMC connector then they are free to ignore it! Obviously actually using this IO for a servo etc. could require funding, but that's a separate matter.

> Confirm that proposed Sayma diffs to date are part of milestone Sayma v2.0

Confirmed. All proposed changes/fixes are part of Sayma v2.0 milestone.

> Review each Issue that is beyond bug fixing and distill down to one or two paths as controversy dictates. Post that distillation to tail of each Issue. We will discuss in existing Issue threads.

Largely done already. I'll double check this this evening and have finished by tomorrow AM. 

> If we're in rough agreement as to this approach, let's do the same for BaseMod changes. But with greater tolerance for high risk changes.

Will do. Situation is essentially the same wrt BaseMod AFAICT: none of the proposed changes directly affect gateware/software required to get RF out of Sayma so should be cost-neutral as far as M-Labs are concerned; all suggestions are posted as part of the BaseMod v2.0 milestone; I'll make sure that everything is summarized on the top post of each issue this evening.

One thing to bear in mind: most of these changes are very low risk and should be quick to implement. e.g. no risk/little work involved in DNPing an ADC; (almost) no risk/little work involved in adding some extra LVDS lines -- if EMI is an issue then users can choose not to use them.

The most risky and time consuming issues are the connector/mechanical issues with the AFE mezzanines.

hartytp, 2017-12-07:
> From my perspective it's not possible to make a final decision on hardware changes as significant as those on the current "wish list" until the existing hardware debugging is complete. Available [...]

ACK.

No decisions need to be made on any of this right away. If possible, I would like to get the bulk of the discussions out of the way relatively soon. That way, once the testing is finished, we can make a final decision relatively quickly.

hartytp, 2017-12-07:
@jbqubit Done!

I've tried to supply a summary of the conversations on each issue, as well as the risks involved etc. @gkasprow should comment on these risks/time taken to implement the issues before we make final decisions.

sbourdeauducq, 2017-12-07:
> no risk/little work involved in adding some extra LVDS lines -- if EMI is an issue then users can choose not to use them.

I don't like that.

hartytp, 2017-12-07:
Elaborate?

sbourdeauducq, 2017-12-07:
It's inconsistent design, if scrambling is not necessary then it shouldn't be used anywhere, if it is necessary it should be everywhere.

sbourdeauducq, 2017-12-07:
Or more exactly: where scrambling is used should be motivated by technical reasons, not "we tacked on this ADC later on".

hartytp, 2017-12-07:
"A foolish consistency is the hobgoblin of little minds, adored by little statesmen and philosophers and divines...."

Thoughts:
- Just because it's not necessary for these LVDS links to be scrambled doesn't mean it's not necessary for anything else to be scrambled. e.g. scrambling might be needed for the JSED DACs (although, I haven't tested/simulated this, so I'm not certain one way or the other).
- If EMI from the DACs causes problems then Sayma as a whole is useless for all users. That's a big issue, requiring a major design change. This is a good motivation to take all precautions to get the DACs working as well as possible. With cheap LVDS ADCs on the AFE, it's easy for individual users to decide how much they care about low-level EMI from the LVDS links. If they deem the EMI from the LVDS links to be acceptable for their case then they can use them. If not then they don't have to. They have other options which don't require a Sayma design change and can be easily adopted according to their individual needs. For example, they can use one of the existing FMC LVDS ADCs on Sayma AMC(the JSED link latency is so high that there is little benefit from having the ADC directly on Sayma RTM in most cases) or repopulate the JSED ADC, or even an external PID controller. It's their choice.
- In the unlikely case that the LVDS links do cause noise issues, users can gate the LVDS clocks so that the ADCs only run during certain "noise eating" pulses. This ensures that there is no noise during the critical part of the experiment. That's not an option for the DACs (ie sometimes we need low-noise RF from the DACs, but we don't need the ADCs to be running at that moment)
- Where scrambling can be implemented easily there is no reason not to. It doesn't hurt and in some cases it does some good. This isn't a reason to rule out anything that can't be scrambled. 
- Your argument would mean that we can't use any LVDS DACs/ADCs anywhere, which is obviously not reasonable.
- Scrambling often makes routing easier (we don't have to be so careful about where the lines go), so it's a good thing. The more stuff that's scrambled the better from this perspective. But, that's only an argument to maximise the amount of scrambling we do, it doesn't forbid using scrambling anywhere.

> Or more exactly: where scrambling is used should be motivated by technical reasons, not "we tacked on this ADC later on".

Ultimately, this design decision is something that should be informed by an electrical engineer like @gkasprow on the basis of what they expect to cause issues in the design, not by your desire for consistency.

My experience of these things is that, if routed properly, (internal layers etc) the LVDS links won't cause any EMI issues. Remember, we have all kinds of other EMI sources such as SMPSs, so the LVDS links shouldn't be the worst offender on the RTM!

sbourdeauducq, 2017-12-07:
> Ultimately, this design decision is something that should be informed by an electrical engineer like @gkasprow on the basis of what they expect to cause issues in the design, not by your desire for consistency.

Sure. I said "motivated by technical reasons".

hartytp, 2017-12-07:
Also, for what it's worth, the AFE ADCs aren't being "tacked on at the end". They are being proposed as the best solution to a concrete problem faced by physicists in the lab. If I had carte blanch to design Sayma from scratch, I would still put these ADCs on the AFE mezzanine and not bother with a JESD ADC. It's just the best approach for what we want to do in the lab.

Also, as far as "tacking things on at the end goes" this is actually something a few of us discussed months ago at Oxford. It was pretty clear to us that the JESD on Sayma ADC was never going to be well suited to our needs, so we talked about modifying BaseMod (then AllakI) to have an ADC. At that point, a lack of IO meant that this wasn't possible, so we decided that we would probably go for a fallback plan of hacking on a Sampler to Sayma via the FMC (this is another fallback option still open to users if the LVDS ADCs cause noise issues).

hartytp, 2017-12-07:
> Sure. I said "motivated by technical reasons".

Well, your objections are noted. Ultimately, these design decisions will all have to be rewarded by @gkasprow, who can flag up any issues.

Anyway, as I've pointed out before, I just don't understand your objection to this (other than an aesthetic one):
- Adding LVDS lines from the AMC to the AFE mezzanines is inherently low-risk and useful independent of the LVDS ADCs. There are many other things AFE designers could choose to use these for. So, there is no good reason *not* to do this.
- It's become pretty clear that most physicists who have contributed so far don't intend to use the JESD ADC. It's just too expensive and has all the other drawbacks we've been over before. The alternatives are better, whether adding an FMC LVDS ADC or adding Sampler via the FMC. So, making the JESD a population option also makes sense independent of the LVDS ADCs. If users want it then they are free to populate it, although AFAICT few will choose to do this.
- Adding an LVDS ADC onto the is low-cost and low risk. It's easy to strip this off later if people don't want it. And several of us have said we definitely plan to use it. These AFEs are designed to be easily user modifiable, so if one doesn't want the ADC then it's easy to remove it (they're trivial LVDS parts).

hartytp, 2017-12-07:
> Sure. I said "motivated by technical reasons".

Anyway, tl;dr: the above is a long list of carefully thought through technical reasons that motivate this decision. It's not something we're sticking on with no thought at all.

hartytp, 2017-12-07:
One final comment about the scrambling thing, since you've brought it up several times now.

It just feels like a case of "to the man with a hammer everything is a nail". Yes, scrambling is an important tool in our kit to use against EMI. But, it's far from the only one, we have screening (including using internal planes for routing), filtering, differential signalling, routing optimisations, gating of signals, etc. Good engineering solutions are not about picking your favourite tool and slavishly applying it to every situation -- they're certainly not about rejecting a good solution because it doesn't work with your current favourite approach to EMI -- they are about considering each situation individually and trying to find a solution. Consistency just isn't the right thing to demand here.

sbourdeauducq, 2017-12-07:
Not my favorite thing, just pointing out that we've been careful to use it everywhere before, so this has to be taken into consideration. If you read my previous messages, I said: "whether [no scrambling on LVDS ADC] is a problem or not is debatable".

hartytp, 2017-12-07:
>  just pointing out that we've been careful to use it everywhere before, so this has to be taken into consideration. 

Well, 👍 to being careful about EMI from the ADC LVDS links.

In practice, we will do the following:
- ADC is on the opposite side of BaseMod to the RF. Internal layers including a solid ground plane will provide excellent screening.
- We use the ADC LVDS interface rather than the LVCMOS. Differential signalling significantly reduces radiation
- We place the ADC very close to the digital connector, minimising the lengths of trace that can radiate
- The RF comes through its connector differentially and then into a balun giving excellent rejection of common-mode pickup
- On Sayma main board, we route the LVDS links directly from the digital connector to a layer of the pcb that's separated from the RF lines by a ground plane. Again, this gives excellent screening
- From there, we route the LVDS lines away from the RF lines to minimise cross-talk
- RF signals are routed differentially on Sayma, again to minimise cross-talk
- Furthermore, users can gate the LVDS signals to avoid digital communication during sensitive parts of their experiment (though I highly doubt that the noise induced by the LVDS will be large enough to warrant this).
- For more discussion on this kind of thing, I highly recommend the Ott EMC classic text if you haven't already read it.

Additionally, we note that the level of noise that users can tolerate from the LVDS lanes varies from user to user. For many users, a tiny bit of cross-talk won't be an issue. For those users, for whom it is an issue (although, again I do not expect the noise from the ADC LVDS to be worse than the noise from near-by SMPSs)  they don't have to use the ADC. It's cheap and can easily be removed from the AFE board, so the costs and risks are extremely low. Moreover, *having more IO from the AMC to the AFE mezzanine is never a bad thing, even if people don't use the ADC.* 

So, my point is that even if the LVDS ADC were to create noise that was problematic in some cases, that's still not a reason not to put it on the AFE for those of us for whom it is useful.

Anyway, I think we've all made our points, so let's leave this here for now.

jordens, 2017-12-14:
Sidenote: In a crate where Saymas (especially full with mezzanines) or other boards are adjacent, there is no appreciable "opposite side".

dhslichter, 2017-12-14:
@jordens this is true, although if the ADC is on the top side of the AFE mezzanine, it will face the bottom of the neighboring Sayma RTM, which doesn't contain analog signals.  Thus hopefully this slot-to-slot crosstalk potential is somewhat mitigated, at least with the current design, but this is fragile and we should consider having a low-profile shielding can option that can go over the ADC section of the AFE.  If we are using FMC connectors with an 8.5 mm stack height (see #402), then there should be enough room to add a can over the top.

jordens, 2017-12-20:
* I'd like to see the following items thoroughly tested and integrated on Sayma 1.0 before starting Sayma 2.0. I expect changes (removal, wiring changes, bugs) from them:
  * openMMC
  * SCANSTA
  * RF backplane operation
  * high density full scale operation (at least 5 Sayma in a crate)
  * the current ADC (if kept)
  * DRTIO instead of SERWB over the RTM-AMC link

sbourdeauducq, 2017-12-20:
> DRTIO instead of SERWB

It's not "instead", both need to be running at the same time, the HMC clock chips are needed for DRTIO.

gkasprow, 2017-12-21:
Are there any issues with SCANSTA chip?
Yesterday I tested it by connecting JTAG probe to the backplane and it works.

jordens, 2017-12-21:
We don't even know since we haven't tested all relevant combinations of ports, chain lengths. Currently it is an unknown area since we don't know what the behavior is (with multiple JTAG masters FTDI/backplane and some or all devices in the chain). We don't know how the backplane access behaves with openocd. And AFAICT we don't share a plan of the ultimate functionality level of that chip on Sayma.

hartytp, 2018-01-12:
@gkasprow More test points on Sayma v2.0, please! Particularly on all the SPI buses.

gkasprow, 2018-08-10:
added testpoints on SPI and other control lines, closing since there is nothing else to do here.

closed 2018-08-10 by gkasprow