sinara-hw/sinara #409: BaseMod input path

hartytp, 2017-12-04:
- [ ]  BaseMod input path should be a direct copy of Sampler.

To think about: do we put the capacitors and resistors on Sayma or on BaseMod?

gkasprow, 2017-12-04:
The capacitors should be not far from the ADC, trace inductance can add some error. But they also shape the bandwidth so it may limit some use cases.

hartytp, 2017-12-04:
> But they also shape the bandwidth so it may limit some use cases.

True. But in this case, the capacitors are big enough and the frequency is low enough that I don't think it will be a problem.

hartytp, 2017-12-04:
I was thinking of the following layout for the ADC inputs on Sayma:
- 25R resistor
- Protection diodes (to +5V and GND)
- 25R resistor
- Capacitors

My reasoning is:
- I don't think it's going to be easy to sort the power sequencing out properly for these ADCs, so probably better to assume that the wrong voltages will be applied to the ADC inputs and make the whole thing robust with diodes.
- The idea of splitting the resistors is that the first resistor limits the current through the diodes, while the second resistor helps to ensure that the diodes turn on before the diodes in the ADC (what do you think about this approach).

Since there is no resistor on the AFE, the OpAmps will see the capacitive load presented by the traces and connectors. Do you think this will be enough to cause ringing? If so, should we put something like 20R on the AFE right by the OpAmp outputs?

gkasprow, 2017-12-04:
@hartytp this approach should work.

gkasprow, 2018-08-18:
This issue was moved to sinara-hw/BaseMod#3

closed 2018-08-18 by gkasprow